ros/meta-ros #313: Roslaunch issue

bmoraine, 2015-01-30:
Hello,

I am facing an issue running  roslaunch whereas I have no issue when executing:
roscore &
rosrun gate_zigbee gatezigbee_pub _name:=ZigbeeNode _publisher_type:=capID _gateway_id:=A8003LQr _init_period:=1hour

when I tried to execute roslaunch, I have the following behaviour:

root@k_sama5d3:~# roslaunch gate_zigbee zigbee.laanch
[zigbee.laanch] is neither a launch file in package [gate_zigbee] nor is [gate_zigbee] a launch file name
root@k_sama5d3:~# cd /opt/ros/hydro/share/gate_zigbee/
root@k_sama5d3:/opt/ros/hydro/share/gate_zigbee# roslaunch zigbee.launch
... logging to /home/root/.ros/log/1a9cfd6c-a878-11e4-aa45-ce2f883abf9a/roslaunch-k_sama5d3-1067.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://k_sama5d3:45118/

SUMMARY
========

PARAMETERS
 * /rosdistro
 * /rosversion
 * /zigbeenode/gateway_id
 * /zigbeenode/init_period
 * /zigbeenode/name
 * /zigbeenode/publisher_type

NODES
  /zigbeenode/
    zigbeenode (gate_zigbee/gatezigbee_pub)

ROS_MASTER_URI=http://k_sama5d3:11311

core service [/rosout] found
ERROR: cannot launch node of type [gate_zigbee/gatezigbee_pub]: gate_zigbee
ROS path [0]=/opt/ros/hydro
ROS path [1]=/opt/ros/hydro/share
No processes to monitor
shutting down processing monitor...
... shutting down processing monitor complete
Unhandled exception in thread started by 
sys.excepthook is missing
lost sys.stderr

It seems I have a path issue, please find below my environment:

root@k_sama5d3:/opt/ros/hydro/share/gate_zigbee# printenv
HZ=100
SHELL=/bin/sh
TERM=linux
ROS_ROOT=/opt/ros/hydro
ROS_PACKAGE_PATH=/opt/ros/hydro/share
ROS_MASTER_URI=http://k_sama5d3:11311
HUSHLOGIN=FALSE
USER=root
LD_LIBRARY_PATH=/opt/ros/hydro/lib
QWS_MOUSE_PROTO=Tslib:/dev/input/touchscreen0
PATH=/usr/local/bin:/usr/bin:/bin:/usr/local/sbin:/usr/sbin:/sbin:/opt/ros/hydro/bin:opt/ros/hydro/lib
PWD=/opt/ros/hydro/share/gate_zigbee
EDITOR=/bin/vi
TZ=UTC
PS1=\u@\h:\w\$ 
SHLVL=1
HOME=/home/root
PYTHONPATH=/opt/ros/hydro/lib/python2.7/site-packages
LOGNAME=root
CMAKE_PREFIX_PATH=/opt/ros/hydro
TSLIB_TSDEVICE=/dev/input/touchscreen0
_=/usr/bin/printenv
OLDPWD=/home/root

Did I miss something?

Regards

Baptiste

bulwahn, 2015-02-04:
To allow others to reproduce the reported issue:
- Could you check if the issue also occurs with the qemu(x86/arm) machines?
- Could you provide a pointer to the gate_zigbee recipes?

bulwahn, 2015-05-27:
I close this issue because I cannot reproduce it and there has not been any further response or request from the issue reporter.

closed 2015-05-27 by bulwahn