ros/meta-ros #322: ros-realtime-tests: timestamp_msg.h not found

KristofRobot, 2015-05-14:
Using the latest master, I have an error bitbaking ros-realtime-tests:

Config:
```
Build Configuration:
BB_VERSION        = "1.27.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-14.04"
TARGET_SYS        = "arm-oe-linux-gnueabi"
MACHINE           = "cubieboard2"
DISTRO            = "frobo"
DISTRO_VERSION    = "0.2.0"
TUNE_FEATURES     = "arm armv7a vfp neon callconvention-hard vfpv4 cortexa7"
TARGET_FPU        = "vfp-vfpv4-neon"
meta              = "master:1702f52b5860336d36fb912fbe2851437e74a7d5"
meta-oe           
meta-networking   
meta-python       
meta-perl         = "master:53d55216c8c721d3b66ec8f968737bf081def870"
meta-sunxi        = "master:59437d24b143d8c9ca6aa16268d7f7a3bc824d60"
meta-ros          = "master:fd9394c2871213ca3c151f7678a6c32cddffe736"
meta-frobo        = "master:8d2e30b3ad9833710ec9a3f03f3289796008e9c2"
```

Error:
```
Building CXX object CMakeFiles/communication_tests_publisher.dir/src/Publisher.cpp.o
/home/kristof/OE/build/tmp-glibc/sysroots/x86_64-linux/usr/bin/arm-oe-linux-gnueabi/arm-oe-linux-gnueabi-g++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"communication_tests\" -march=armv7-a -mfloat-abi=hard -mfpu=neon-vfpv4 -mtune=cortex-a7  --sysroot=/home/kristof/OE/build/tmp-glibc/sysroots/cubieboard2  -O2 -pipe -g -feliminate-unused-debug-types -fvisibility-inlines-hidden -I/home/kristof/OE/build/tmp-glibc/work/cortexa7hf-vfp-vfpv4-neon-oe-linux-gnueabi/communication-tests/0.1.3-r0/build/devel/include -I/home/kristof/OE/build/tmp-glibc/sysroots/cubieboard2/opt/ros/hydro/include -I/home/kristof/OE/build/tmp-glibc/sysroots/cubieboard2/usr/include -I/home/kristof/OE/build/tmp-glibc/work/cortexa7hf-vfp-vfpv4-neon-oe-linux-gnueabi/communication-tests/0.1.3-r0/ros_realtime_tests-0.1.3/communication_tests/../rt_tests_support/include    -o CMakeFiles/communication_tests_publisher.dir/src/Publisher.cpp.o -c /home/kristof/OE/build/tmp-glibc/work/cortexa7hf-vfp-vfpv4-neon-oe-linux-gnueabi/communication-tests/0.1.3-r0/ros_realtime_tests-0.1.3/communication_tests/src/Publisher.cpp
In file included from /home/kristof/OE/build/tmp-glibc/work/cortexa7hf-vfp-vfpv4-neon-oe-linux-gnueabi/communication-tests/0.1.3-r0/ros_realtime_tests-0.1.3/communication_tests/src/Subscriber.cpp:9:0:
/home/kristof/OE/build/tmp-glibc/work/cortexa7hf-vfp-vfpv4-neon-oe-linux-gnueabi/communication-tests/0.1.3-r0/ros_realtime_tests-0.1.3/communication_tests/src/Subscriber.h:15:47: fatal error: communication_tests/timestamp_msg.h: No such file or directory
 #include <communication_tests/timestamp_msg.h>
                                               ^
compilation terminated.
/home/kristof/OE/build/tmp-glibc/work/cortexa7hf-vfp-vfpv4-neon-oe-linux-gnueabi/communication-tests/0.1.3-r0/ros_realtime_tests-0.1.3/communication_tests/src/Publisher.cpp:12:47: fatal error: communication_tests/timestamp_msg.h: No such file or directory
 #include <communication_tests/timestamp_msg.h>
                                               ^
compilation terminated.
```

If I do a full search for timestamp_msg.h, it is nowhere to be found on the system:
```
kristof@test:~/OE/build/tmp-glibc/log/cooker/cubieboard2$ sudo updatedb
kristof@test:~/OE/build/tmp-glibc/log/cooker/cubieboard2$ locate timestamp_msg.h
kristof@test:~/OE/build/tmp-glibc/log/cooker/cubieboard2$ 
```

bulwahn, 2015-05-15:
Dear @KristofRobot , I just re-tried compiling with the current master of oe-core and meta-openembedded on a clean build directory and could not reproduce the issue you reported:
```(shell)
$ rm bitbake.lock  cache/ sstate-cache/ tmp-glibc/ -rf
$ bitbake communication-tests
...
Build Configuration:
BB_VERSION        = "1.27.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Fedora-20"
TARGET_SYS        = "arm-oe-linux-gnueabi"
MACHINE           = "qemuarm"
DISTRO            = "nodistro"
DISTRO_VERSION    = "nodistro.0"
TUNE_FEATURES     = "arm armv5 thumb dsp"
TARGET_FPU        = "soft"
meta              = "master:90861b8908d254154f4d1d613471070df8013da8"
meta-oe
meta-python       = "master:53d55216c8c721d3b66ec8f968737bf081def870"
meta-ros          = "master:fd9394c2871213ca3c151f7678a6c32cddffe736"

NOTE: Preparing RunQueue
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
NOTE: Tasks Summary: Attempted 1415 tasks of which 0 didn't need to be rerun and all succeeded.
```

I am now repeating it again with exactly the versions of oe-core and meta-openembedded you used. At first glance, it seems to be due to your build environment (some dirty build directory, some temporary issue with oe-core, or some issue with your machine configuration or distribution setup) and not an issue that is persistently reproducible in many settings.

bulwahn, 2015-05-15:
I just got the results with the build configuration you provided, and I cannot reproduce the issue with those versions as well:
```(shell)
$ bitbake communication-tests
...
Build Configuration:
BB_VERSION        = "1.27.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Fedora-20"
TARGET_SYS        = "arm-oe-linux-gnueabi"
MACHINE           = "qemuarm"
DISTRO            = "nodistro"
DISTRO_VERSION    = "nodistro.0"
TUNE_FEATURES     = "arm armv5 thumb dsp"
TARGET_FPU        = "soft"
meta              = "(detachedfrom1702f52):1702f52b5860336d36fb912fbe2851437e74a7d5"
meta-oe
meta-python       = "(detachedfrom53d5521):53d55216c8c721d3b66ec8f968737bf081def870"
meta-ros          = "master:fd9394c2871213ca3c151f7678a6c32cddffe736"

NOTE: Preparing RunQueue
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
NOTE: Tasks Summary: Attempted 1415 tasks of which 0 didn't need to be rerun and all succeeded.
```
@KristofRobot , maybe you can re-run it with a clean build?

KristofRobot, 2015-05-16:
@bulwahn 

Thanks for testing, I retried with a clean build, but same problem.

Do you find ``timestamp_msg.h`` on your system?

I will try again without the 'frobo' distro, perhaps there is some interference there. 
More likely there is some problem combining this with my machine layer (``cubieboard2``), but want to exclude that first.

KristofRobot, 2015-05-17:
I even have the problem with the same build configuration as you used:

```
Build Configuration:
BB_VERSION        = "1.27.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-14.04"
TARGET_SYS        = "arm-oe-linux-gnueabi"
MACHINE           = "qemuarm"
DISTRO            = "nodistro"
DISTRO_VERSION    = "nodistro.0"
TUNE_FEATURES     = "arm armv5 thumb dsp"
TARGET_FPU        = "soft"
meta              = "master:1702f52b5860336d36fb912fbe2851437e74a7d5"
meta-oe           
meta-python       = "master:53d55216c8c721d3b66ec8f968737bf081def870"
meta-ros          = "master:fd9394c2871213ca3c151f7678a6c32cddffe736"
```

And note that this was a completely fresh build -  using a new build directory.

Nevertheless, this suggests some build problem specific to my machine/configuration - will look further.

KristofRobot, 2015-05-18:
The issue seems due to my custom ``local.conf```; I am still tracking down the exact configuration entry.

KristofRobot, 2015-05-18:
Apparently it is due to:
```
BB_NUMBER_THREADS ?= "2"
PARALLEL_MAKE ?= "-j 2"
```
This suggests some missing dependency.

bulwahn, 2015-05-19:
Interesting---thanks to your pointer, I could finally reproduce it. It is probably due to an issue in the communication_tests' CMakeLists.txt. I will further investigate that.

commit c4ff23ac (2015-05-28): ros-realtime-tests: updating to 0.1.4 (resolves #322)
closed 2015-05-28 by bulwahn