ros/meta-ros #331: Buiding ROS package - catkin_make doesn't work

amoghskulkarni, 2015-06-19:
I wrote a new recipe for bare-bone ros containing ros-comm package and its dependencies. The recipe is as follows -

```bash
DESCRIPTION = "A small image with bare-bone ROS."

IMAGE_INSTALL = "packagegroup-core-boot ${ROOTFS_PKGMANAGE_BOOTSTRAP} ${CORE_IMAGE_EXTRA_INSTALL}"

IMAGE_LINGUAS = " "

LICENSE = "MIT"

inherit core-image

IMAGE_ROOTFS_SIZE = "8192"

# remove not needed ipkg informations
ROOTFS_POSTPROCESS_COMMAND += "remove_packaging_data_files"

IMAGE_INSTALL += "packagegroup-ros-comm packagegroup-core-buildessential openssh opkg"

```

I can `bitbake` the recipe without getting any error, and run the image on the target architecture (qemu in this case). `roscore` runs fine (although for that to work, I have to tweak `/etc/hosts` and `/opt/ros/indigo/share/roslaunch/resources/roscore.xml`).

Although, `catkin_init_workspace` and `catkin_create_pkg` work; `catkin_make` fails with following output -
```bash
root@qemuarm:~/workspace# ls
src
root@qemuarm:~/workspace# cd src/
root@qemuarm:~/workspace/src# ls
CMakeLists.txt  hello_world
root@qemuarm:~/workspace/src# cd hello_world/
root@qemuarm:~/workspace/src/hello_world# ls
CMakeLists.txt  package.xml
root@qemuarm:~/workspace/src/hello_world# cd ../..
root@qemuarm:~/workspace# catkin_make
Base path: /home/root/workspace
Source space: /home/root/workspace/src
Build space: /home/root/workspace/build
Devel space: /home/root/workspace/devel
Install space: /home/root/workspace/install
####
#### Running command: "cmake /home/root/workspace/src -DCATKIN_DEVEL_PREFIX=/home/root/workspace/devel -DCMAKE_INSTALL_PREFIX=/home/root/workspace/install -G Unix Makefiles" in "/home/root/workspace/build"
####
-- The C compiler identification is GNU 4.9.2
-- The CXX compiler identification is GNU 4.9.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
CMake Error at CMakeLists.txt:59 (message):
  find_package(catkin) failed.  catkin was neither found in the workspace nor
  in the CMAKE_PREFIX_PATH.  One reason may be that no ROS setup.sh was
  sourced before.

-- Configuring incomplete, errors occurred!
See also "/home/root/workspace/build/CMakeFiles/CMakeOutput.log".
Invoking "cmake" failed
```

`CMAKE_PREFIX_PATH` is set to `/opt/ros/indigo` and there's no `setup.sh` to source in `/opt/ros/indigo`. Other environment variables are set to appropriate values as suggested in [here](https://github.com/bmwcarit/meta-ros/blob/master/README.md).

Am I doing something wrong here?

KristofRobot, 2015-06-20:
Hi  @amoghskulkarni - building on the target is currently not supported, see also #103

amoghskulkarni, 2015-06-20:
Thank you for the information, @KristofRobot .

I successfully cross-compiled my own ROS packages writing bitbake recipe for them (used [diving-meta-ros](https://github.com/vmayoral/diving-meta-ros) as a guide, thanks @vmayoral ). I was looking for some alternative to this (such as `bitbake -c populate_sdk`), where I can just use `catkin_make` or `cmake` to cross-compile packages, which is why I first tried to compile on target.

The reason for searching an alternative is, large amount of disk space (**~25GB**) required when using former method to cross-compile. Is there a way to shrink down `build` directory and reduce overall footprint of `meta-ros`?

KristofRobot, 2015-06-21:
@amoghskulkarni 
> Is there a way to shrink down build directory

Yes, add following line to your ``local.conf`` file:

```
INHERIT += "rm_work"
```

Adding this statement deletes the work directory used for building a package once the package is built. I personally use this all the time.

Minipada, 2017-03-15:
Hey,
Deploying to the target during development is a pain since we have to compile on the Yocto machine and send to the development one. I'm interested in solving this issue. Any idea where I could start?
I thought about preparing the setup.sh, _setup_util.py manually in /opt/ros/<ros version> etc. I miss knowledge in ROS, is there a way to generate by themselves?

bulwahn, 2017-03-16:
@Minipada I think it would be great if you could pick up the work by @dominiquehunziker. His/her work  was to enable to use the bitbake `sdk functionality` with meta-ros. Essentially, it allows you to compile a binary on any kind of development machine (without need of the full bitbake & openembedded setup), and then simply copy that binary onto a running target. We never merged it in the main branch, as I never could test it due to time constraint and insufficient instructions how to set it and what to test. The pull request #386 should get you to some latest version. I would love to see that functionality merged into the current master. The main difficulty is to go through and checked how master and this branch has diverged and document how and what we must test to make sure that it works in the future   (when new pull requests are merged).

Minipada, 2017-04-25:
I am continuing work of @dominiquehunziker and I just managed to compile on the target with some basic ROS packages. You can find the branch here: [https://github.com/sncn-private/meta-ros/tree/feature_compile_on_target](https://github.com/sncn-private/meta-ros/tree/feature_compile_on_target).
I'll make a PR tomorrow. Next step will be to test more complex packages and work on the SDK.

bulwahn, 2017-04-25:
@Minipada Could you successfully rebase it to the current master? @rojkov did quite some changes for the recipe-specific sysroot adjustments; hopefully, those do not contradict with the changes needed to compile on the target.

dominiquehunziker, 2017-04-25:
As far as I can tell they are conflicting. I have an updated branch (WIP) [here](https://github.com/rapyuta/meta-ros/tree/merge-upstream), but this one more or less replaces the recent changes with me changes which might not be desired.

Minipada, 2017-04-25:
@bulwahn They are conflicting, I am sure about it. I will work on this, it's so much needed to have this and will improve everyone's work speed.

bulwahn, 2017-04-25:
@Minipada If we can support you in any way, just let us know. In the past, we did not merge @dominiquehunziker's work because we did not have any basic description/documentation what we would have to continuously run and check with every pull request and with changes on openembedded-core to make sure that we don't break compiling on the target when merging pull requests or moving forward on the openembedded-core repository. Please keep this in mind for your work.

Minipada, 2017-04-25:
I'll keep it in mind, I'm using your work since almost 2 years, I'm happy to finally contribute. Thanks for all the support :)

Minipada, 2017-04-26:
I have made a test to merge with the HEAD, it's still a WIP. Now I get errors on most packages, being
`The compile log indicates that host include and/or library paths were used.` during QA.

Can it be because I use Jethro, meta-openembedded@131255d, or should it work with it?

Minipada, 2017-04-26:
By the way, should I make a PR and we talk there or should we just continue here?

TSC21, 2017-04-26:
@Minipada should we also consider support `catkin tools`?

Minipada, 2017-04-26:
I will definitely work on it, but this is another topic and will be a different PR.

TSC21, 2017-04-26:
> I will definitely work on it, but this is another topic and will be a different PR.

Sure thanks!

bulwahn, 2017-04-26:
@Minipada The current master should always at least work with the current master of openembedded-core; I would not even expect it to work with Jethro, as we add backwards incompatible changes from time to time, as Jethro is already an older yocto release. So, could you update to the current master of openembedded-core for testing your work?
You can open a pull request and we discuss the changes there directly; the noise for a pull request not being ready to merge immediately is no issue at all. Reviewing and rebasing works quite nicely with pull  requests, so give it a try.

herb-kuta-lge, 2019-11-17:
@amoghskulkarni < Can this issue be closed?

amoghskulkarni, 2019-11-21:
@herb-kuta-lge 
Yes, thanks for asking.

closed 2019-11-21 by herb-kuta-lge