ros/meta-ros #338: roslaunch: unbuildable dependency catkin-runtime-native

KristofRobot, 2015-07-24:
I have an error with the latest OE master, building roslaunch:

```
bitbake roslaunch
NOTE: Started PRServer with DBfile: /home/kristof/OE/build/cache/prserv.sqlite3, IP: 127.0.0.1, PORT: 60381, PID: 9223
Loading cache: 100% |###########################################| ETA:  00:00:00
Loaded 2506 entries from dependency cache.
Parsing recipes: 100% |#########################################| Time: 00:00:11
Parsing of 2002 .bb files complete (1992 cached, 10 parsed). 2515 targets, 107 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing PROVIDES 'catkin-runtime-native' (but virtual:native:/home/kristof/OE/meta-ros/recipes-ros/catkin/catkin_0.6.14.bb DEPENDS on or otherwise requires it). Close matches:
  catkin-runtime
  catkin-native
  gtk-icon-utils-native
ERROR: Required build target 'roslaunch' has no buildable providers.
Missing or unbuildable dependency chain was: ['roslaunch', 'catkin-native', 'catkin-runtime-native']

Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

```
Build Configuration:
BB_VERSION        = "1.27.1"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-14.04"
TARGET_SYS        = "arm-oe-linux-gnueabi"
MACHINE           = "cubieboard2"
DISTRO            = "frobo"
DISTRO_VERSION    = "0.2.0"
TUNE_FEATURES     = "arm armv7a vfp neon callconvention-hard vfpv4 cortexa7"
TARGET_FPU        = "vfp-vfpv4-neon"
meta              = "master:3143920c541b55b543b9dcc12b18af4e0e4b7ae1"
meta-oe           
meta-networking   
meta-python       
meta-multimedia   = "master:c12b9596afd236116b25ce26dbe0d793de9dc7ce"
meta-ros          = "various-fixes:1e1249be4531de92fb945b51eb45e0c32e81508d"
meta-sunxi        = "master:f4ebd29a5075e94d78b5c31dcf9d856962430e67"
meta-frobo        = "master:f3ed66fe5117c706be8e2c35cbd90bd8564b7330"
```

KristofRobot, 2015-07-24:
`catkin` indeed depends on `catkin-runtime-native`:
```
DEPENDS_class-native += "catkin-runtime"
```

Moreover, a naieve fix is to add following to the `catkin-runtime` recipe [1]
```
BBCLASSEXTEND += "native"
```

This in turn leads to:
```
ERROR: Nothing RPROVIDES 'python-argparse-native' (but virtual:native:/home/kristof/OE/meta-ros/recipes-ros/catkin/catkin-runtime_0.6.14.bb RDEPENDS on or otherwise requires it)
```

This can be solved as explained in #101 by adding:
```
RDEPENDS_${PN}_class-native = ""
```

These two changes seem to solve the problem, but might have other effects I am not aware of.

Also, I dont understand why this now suddenly is an issue - this seems to have worked with previous OE versions - and still works with the fido release.

Thoughts?

[1] https://github.com/bmwcarit/meta-ros/blob/master/recipes-ros/catkin/catkin-runtime_0.6.14.bb

bulwahn, 2015-07-25:
I can confirm this problem. On our company's internal continuous integration system, I first observed this error message twelve days ago (on July 13th); I then tried to reproduce it on one of my build system (Fedora), but there I could not reproduce it, so I assumed the error is related to the CI system. After you reported the error, I now just took another machine (Ubuntu 14.04) and setup OE and meta-ros. On that machine, I could reproduce it.
I just also tried to bisect the problem, but it seems that it is not related to a specific oe-core commit. On the CI, I see that some build configurations work, but on the other machine the same build configuration does not. Today, I will not spend more time to bisect it.

I suggest to commit your solution, and we will observe what happens. I also happy to report when my CI system reports errors; in the past, all of the errors (except this one) were related to the CI configuration and not to oe-core, meta-oe or meta-ros changes (the CI configuration is still quite young and experimental.).

commit 73180844 (2015-07-25): catkin-runtime: adding support for native version (resolves #338)
commit a5f04ee8 (2015-07-25): Merge pull request #340 from KristofRobot/catkin-runtime-fix

catkin-runtime: adding support for native version (resolves #338)
closed 2015-07-25 by KristofRobot