ros/meta-ros #349: rosrun: requires full catkin package

KristofRobot, 2015-07-26:
``rosrun`` seems not to be working without the full ``catkin`` package (i.e. only with the standard ``catkin-runtime`` package, as explained in [1]):

I first detected this when trying to ``rosrun`` xv_11_laser_driver (after setting ``ROS_PACKAGE_PATH`` as explained in #348):
```
$ rosrun --debug xv_11_laser_driver neato_laser_publisher 
[rosrun] Looking in catkin libexec dir: 
[rosrun] Looking in rospack dir: /opt/ros/indigo/share/xv_11_laser_driver
[rosrun] Searching for neato_laser_publisher with permissions /111
[rosrun] Couldn't find executable named neato_laser_publisher below /opt/ros/indigo/share/xv_11_laser_driver
```

It should find ``neato_laser_publisher`` in the lib dir though:
```
$locate  neato_laser_publisher
/opt/ros/indigo/lib/xv_11_laser_driver/neato_laser_publisher
```

It does not find it, however, as the ``catkin libexec dir`` variable is empty. Looking into ``rosrun``, this variable is filled as follows:
```
if [[ -n $CMAKE_PREFIX_PATH ]]; then
  catkin_package_libexec_dir=`catkin_find --without-underlays --libexec --share $1
  debug "Looking in catkin libexec dir: $catkin_package_libexec_dir"
fi
```

The reason it does not find it, is because ``catkin_find`` does not exist:
```
$catkin_find --without-underlays --libexec --share xv_11_laser_driver
-sh: catkin_find: command not found
```

After installing ``catkin``, the ``catkin_find`` command is present, and everything works fine:
```
$ catkin_find --without-underlays --libexec --share xv_11_laser_driver
/opt/ros/indigo/lib/xv_11_laser_driver
/opt/ros/indigo/share/xv_11_laser_driver
```

IMO, ``rosrun`` is a basic command, that should be working in a standard install of ``meta-ros``.
Not sure if there is a way to add this to ``catkin_runtime``, without pulling the gcc dependencies and such; but if not, I would suggest we re-include the full ``catkin`` as default.

Alternatively, we should add a clear warning in the documentation/README on this.

[1] https://github.com/bmwcarit/meta-ros/commit/185882428c346c0f262b14b62b05cc9f690e2a36

bulwahn, 2015-08-18:
As far as I understand the issue, a proper solution would be to add `catkin_find` to the `catkin_runtime` recipe/package if `catkin_find` does not require any further runtime dependencies with large footprint, such as gcc, cmake and make.

KristofRobot, 2015-08-18:
Yups, that seems to work, opened a pull request to do exactly that.

commit 38854bd2 (2015-08-19): Merge pull request #350 from KristofRobot/catkin_find

catkin-runtime: including catkin_find (resolves #349)
commit 5244c5eb (2015-08-19): catkin-runtime: including catkin_find (resolves #349)
closed 2015-08-19 by KristofRobot