ros/meta-ros #380: Workflow for cross-development catkin packages with meta-ros

graugans, 2016-03-10:
This issue is a follow up of the discussion with @sephalon and @dominiquehunziker  pull request #376 

I have the following scenario in mind. We develop a embedded embedded system and provide the *customers* a toolchain for this target. The toolchain is generated by the ``populate_sdk`` task. This task generates a SDK suited for the image deployed on the target. It would be really cool when the user could do the following and compile packages for the target in this case the [O3D3xx ToF Camera ROS Package](https://github.com/lovepark/o3d3xx-ros) from @tpanzarella

```
$ source /opt/poky/1.8.1/environment-setup-armv7ahf-vfp-neon-poky-linux-gnueabi
$ mkdir o3d3xx-catkin
$ cd o3d3xx-catkin
$ mkdir src
$ cd src
$ catkin_init_workspace
$ ln -s ~/dev/o3d3xx-ros o3d3xx
$ catkin_make
```
In my naive way I tried to build a ``nativesdk`` version of catkin. To be able to build the nativesdk packages  the following patch was needed. This patch is against my fork for integration of the [libo3d3xx](https://github.com/graugans/meta-ros/tree/fido) into meta-ros

```patch
 diff --git a/recipes-devtools/python/python-catkin-pkg_1.2.6.bb b/recipes-devtools/python/python-catkin-pkg_0.2.6.bb
index fe54e25..259c388 100644
--- a/recipes-devtools/python/python-catkin-pkg_0.2.6.bb
+++ b/recipes-devtools/python/python-catkin-pkg_0.2.6.bb
@@ -12,4 +12,4 @@ S = "${WORKDIR}/${SRCNAME}-${PV}"

 inherit distutils

-BBCLASSEXTEND = "native"
+BBCLASSEXTEND = "native nativesdk"
diff --git a/recipes-devtools/python/python-empy_3.3.2.bb b/recipes-devtools/python/python-empy_3.3.2.bb
index bfe11e4..b6a8a44 100644
--- a/recipes-devtools/python/python-empy_3.3.2.bb
+++ b/recipes-devtools/python/python-empy_3.3.2.bb
@@ -12,4 +12,4 @@ S = "${WORKDIR}/${SRCNAME}-${PV}"

 inherit distutils

-BBCLASSEXTEND = "native"
+BBCLASSEXTEND = "native nativesdk"
diff --git a/recipes-extended/libo3d3xx/libo3d3xx-camera_0.4.0.bb b/recipes-extended/libo3d3xx/libo3d3xx-camera_0.4.0.bb
index 556106b..a41a6c6 100644
--- a/recipes-extended/libo3d3xx/libo3d3xx-camera_0.4.0.bb
+++ b/recipes-extended/libo3d3xx/libo3d3xx-camera_0.4.0.bb
@@ -6,4 +6,4 @@ DEPENDS = "boost glog libxmlrpc"
 S = "${WORKDIR}/libo3d3xx-${PV}/modules/camera"
 FILES_${PN}-dev += "${prefix}/lib/o3d3xx_camera"

-BBCLASSEXTEND += "native"
+BBCLASSEXTEND += "native nativesdk"
diff --git a/recipes-extended/libxmlrpc/libxmlrpc_1.33.17.bb b/recipes-extended/libxmlrpc/libxmlrpc_1.33.17.bb
index 475423d..0c1490c 100644
--- a/recipes-extended/libxmlrpc/libxmlrpc_1.33.17.bb
+++ b/recipes-extended/libxmlrpc/libxmlrpc_1.33.17.bb
@@ -41,4 +41,4 @@ B = "${S}"
 SRC_URI[md5sum] = "5c1b4514ffc8a7e0e30aeddd3bb5c83e"
 SRC_URI[sha256sum] = "50118a3ca1114828f7cae27b7253cc25045849e487f5d0bb91d962777fc05355"

-BBCLASSEXTEND = "native"
\ No newline at end of file
+BBCLASSEXTEND = "native nativesdk"
diff --git a/recipes-ros/catkin/catkin-runtime_0.6.16.bb b/recipes-ros/catkin/catkin-runtime_0.6.16.bb
index a63ec53..712d824 100644
--- a/recipes-ros/catkin/catkin-runtime_0.6.16.bb
+++ b/recipes-ros/catkin/catkin-runtime_0.6.16.bb
@@ -23,4 +23,4 @@ do_install_append() {
     rm -rf ${D}${ros_libdir}/pkgconfig
 }
-BBCLASSEXTEND += "native"
+BBCLASSEXTEND += "native nativesdk"
diff --git a/recipes-ros/catkin/catkin_0.6.16.bb b/recipes-ros/catkin/catkin_0.6.16.bb
index b09e07c..f0be7bb 100644
--- a/recipes-ros/catkin/catkin_0.6.16.bb
+++ b/recipes-ros/catkin/catkin_0.6.16.bb
@@ -1,8 +1,10 @@
 require catkin.inc

 DEPENDS_class-native += "catkin-runtime"
+DEPENDS_class-nativesdk += "catkin-runtime"

 RDEPENDS_${PN}_class-native = ""
+RDEPENDS_${PN}_class-nativesdk = ""
 RDEPENDS_${PN} = "cmake make binutils binutils-symlinks gcc gcc-symlinks g++ g++-symlinks \
     catkin-runtime"

@@ -17,4 +19,4 @@ do_install_append() {
     rmdir ${D}${ros_libdir}/python2.7
 }

-BBCLASSEXTEND += "native"
+BBCLASSEXTEND += "native nativesdk"
diff --git a/recipes-support/boost/boost_%.bbappend b/recipes-support/boost/boost_%.bbappend
index ec17f55..78ee78f 100644
--- a/recipes-support/boost/boost_%.bbappend
+++ b/recipes-support/boost/boost_%.bbappend
@@ -1 +1,2 @@
 PACKAGECONFIG_append = " python"
+BBCLASSEXTEND += "nativesdk"
diff --git a/recipes-support/glog/glog_%.bbappend b/recipes-support/glog/glog_%.bbappend
index d6f5869..5174650 100644
--- a/recipes-support/glog/glog_%.bbappend
+++ b/recipes-support/glog/glog_%.bbappend
@@ -1 +1 @@
-BBCLASSEXTEND += "native"
+BBCLASSEXTEND += "native nativesdk"
```

The **nativesdk-catkin** builds fine and after adding the following to ``TOOLCHAIN_HOST_TASK`` I do have catkin in my SDK.

```
TOOLCHAIN_HOST_TASK +=  \"
    nativesdk-cmake \
    nativesdk-catkin \
    nativesdk-catkin-dev \
    nativesdk-python-catkin-pkg \
    nativesdk-python-catkin-pkg-dev \
    nativesdk-catkin-runtime \
    nativesdk-catkin-runtime-dev \
"
```

All of this is a bit hacks and need some polishing. But I want to ask if this is a possible way to go or if I will run in more headache then I do expect at the moment.

dominiquehunziker, 2016-03-10:
This is along the same lines as what I have.

The main issue that has to be addressed (as with getting catkin working on the embedded target) are all the files auto-generated by catkin (such as <PKG>Config.cmake) containing hard-coded absolute paths.

I will try to clean up what I have and will commit it to our fork.

graugans, 2016-03-10:
That would be really helpful. Thanks in advance.

dominiquehunziker, 2016-03-18:
So, I finally was able to clean up my current status a bit which can be found [here](https://github.com/rapyuta/meta-ros/tree/catkin-fix-v2)

The current status contains a `test-image` which I was using as a basis for the command `populate_sdk` and as a similar image on the target.

I still have several issues; foremost, I have now an issue that for some reason the `boost_thread` library is no longer properly linked and has to be manually added (i.e., by patching the CMakeLists.txt) to quite a few packages. Also, the generated `setup.<shell>` scripts as well as the `_setup_util.py` script is not yet correct.

Finally, here the commands to build stuff using the SDK:
```
cd <catkin_ws>
catkin_make --cmake-args -Wno-dev -DCATKIN_ENABLE_TESTING=OFF -DCATKIN_BUILD_BINARY_PACKAGE=ON
```
* `-Wno-dev` to suppress some annoying warnings (I haven't looked into them yet) with regard to the include directories which are now prefixed with `=` to better support cross compilation. This offloads the responsibility of finding the proper includes to the compiler (I have tested this only with gcc, so no guarantees that this will work with other compilers)
* `-DCATKIN_ENABLE_TESTING=OFF` to disable testing just to void huge dependencies
* `-DCATKIN_BUILD_BINARY_PACKAGE=ON` to suppress the generation of of the (broken) `setup.<shell>` and `_setup_util.py` utilities

and to install it:
```
cd <catkin_ws>
DESTDIR=<target sysroot> catkin_make install --cmake-args -DCMAKE_INSTALL_PREFIX=/opt/ros/indigo
```

Notes:
* there are backport recipes in `recipes-backport` which are stored in directories named `<commit date>-<commit hash>`; so depending on which commit you are using you will have to remove some/all of them. (I only tested with one version, namely the one where all these backports where necessary)
* In the SDK only catkin_make and catkin_make_isolated (not tested) will work (see the catkin [environment hook](https://github.com/rapyuta/meta-ros/blob/catkin-fix-v2/recipes-ros/catkin/catkin/environment.d-catkin.sh) for the alias used in case you want to test catkin-tools)

Yaaaaao, 2020-03-13:
Hi everyone!

May I ask if current meta-ros (ROS2 Eloquent distro) SDK still work or not? Is it possible to cross-compile a ROS package using an Eloquent OE SDK?

Best Regard,
Yao