ros/meta-ros #384: Task for catkin_runtime throws error

t0mmer, 2016-05-14:
Hello,

I am trying to do build an core-image-ros-roscore image according to "Usage" found at https://github.com/bmwcarit/meta-ros for qemux86.
I am using a common download and sstate-cache folder on my hard disk which are also used for other platforms.
With fresh clones of all needed repositories I get following error:

```
DEBUG: Executing python function sysroot_cleansstate
DEBUG: Python function sysroot_cleansstate finished
DEBUG: Executing shell function do_configure
NOTE: cmake.bbclass no longer uses OECMAKE_BUILDPATH.  The default behaviour is now out-of-tree builds with B=WORKDIR/build.
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/gcc
-- Check for working C compiler: /usr/bin/gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/g++
-- Check for working CXX compiler: /usr/bin/g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/btr8fe/workspace/sandbox/yocto/ros_hydro/devel/build/qemux86/tmp-glibc/work/x86_64-linux/catkin-runtime-native/0.6.16-r0/build/devel
-- Using CMAKE_PREFIX_PATH: /home/btr8fe/workspace/sandbox/yocto/ros_hydro/devel/build/qemux86/tmp-glibc/sysroots/x86_64-linux/opt/ros/indigo
-- This workspace overlays: /home/btr8fe/workspace/sandbox/yocto/ros_hydro/devel/build/qemux86/tmp-glibc/sysroots/x86_64-linux/opt/ros/indigo
-- Found PythonInterp: /home/btr8fe/workspace/sandbox/yocto/ros_hydro/devel/build/qemux86/tmp-glibc/sysroots/x86_64-linux/usr/bin/python-native/python (found version "2.7.11") 
-- Using PYTHON_EXECUTABLE: /home/btr8fe/workspace/sandbox/yocto/ros_hydro/devel/build/qemux86/tmp-glibc/sysroots/x86_64-linux/usr/bin/python-native/python
-- Using default Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: 0
-- catkin 0.6.16
/home/btr8fe/workspace/sandbox/yocto/ros_hydro/devel/build/qemux86/tmp-glibc/work/x86_64-linux/catkin-runtime-native/0.6.16-r0/build/devel/env.sh: 59: /home/btr8fe/workspace/sandbox/yocto/ros_hydro/devel/build/qemux86/tmp-glibc/work/x86_64-linux/catkin-runtime-native/0.6.16-r0/build/devel/setup.sh: /home/btr8fe/workspace/sandbox/yocto/ros_hydro/devel/build/qemux86/tmp-glibc/work/x86_64-linux/catkin-runtime-native/0.6.16-r0/build/devel/_setup_util.py: not found
Traceback (most recent call last):
  File "/home/btr8fe/workspace/sandbox/yocto/ros_hydro/devel/build/qemux86/tmp-glibc/work/x86_64-linux/catkin-runtime-native/0.6.16-r0/build/catkin_generated/generate_cached_setup.py", line 22, in <module>
    code = generate_environment_script('/home/btr8fe/workspace/sandbox/yocto/ros_hydro/devel/build/qemux86/tmp-glibc/work/x86_64-linux/catkin-runtime-native/0.6.16-r0/build/devel/env.sh')
  File "/home/btr8fe/workspace/sandbox/yocto/ros_hydro/devel/build/qemux86/tmp-glibc/work/x86_64-linux/catkin-runtime-native/0.6.16-r0/catkin-0.6.16/cmake/../python/catkin/environment_cache.py", line 62, in generate_environment_script
    env_after = ast.literal_eval(output.decode('utf8'))
  File "/home/btr8fe/workspace/sandbox/yocto/ros_hydro/devel/build/qemux86/tmp-glibc/sysroots/x86_64-linux/usr/lib/python2.7/ast.py", line 49, in literal_eval
    node_or_string = parse(node_or_string, mode='eval')
  File "/home/btr8fe/workspace/sandbox/yocto/ros_hydro/devel/build/qemux86/tmp-glibc/sysroots/x86_64-linux/usr/lib/python2.7/ast.py", line 37, in parse
    return compile(source, filename, mode, PyCF_ONLY_AST)
  File "<unknown>", line 1
    Failed to run '"/home/btr8fe/workspace/sandbox/yocto/ros_hydro/devel/build/qemux86/tmp-glibc/work/x86_64-linux/catkin-runtime-native/0.6.16-r0/build/devel/_setup_util.py" /home/btr8fe/workspace/sandbox/yocto/ros_hydro/devel/build/qemux86/tmp-glibc/sysroots/x86_64-linux/usr/bin/python-native/python -c import os; print(dict(os.environ))': return code 127
            ^
SyntaxError: invalid syntax
CMake Error at cmake/safe_execute_process.cmake:11 (message):
  
  execute_process(/home/btr8fe/workspace/sandbox/yocto/ros_hydro/devel/build/qemux86/tmp-glibc/sysroots/x86_64-linux/usr/bin/python-native/python
  "/home/btr8fe/workspace/sandbox/yocto/ros_hydro/devel/build/qemux86/tmp-glibc/work/x86_64-linux/catkin-runtime-native/0.6.16-r0/build/catkin_generated/generate_cached_setup.py")
  returned error code 1
Call Stack (most recent call first):
  cmake/all.cmake:184 (safe_execute_process)
  CMakeLists.txt:8 (include)

-- Configuring incomplete, errors occurred!
See also "/home/btr8fe/workspace/sandbox/yocto/ros_hydro/devel/build/qemux86/tmp-glibc/work/x86_64-linux/catkin-runtime-native/0.6.16-r0/build/CMakeFiles/CMakeOutput.log".
WARNING: exit code 1 from a shell command.
ERROR: Function failed: do_configure (log file is located at /home/btr8fe/workspace/sandbox/yocto/ros_hydro/devel/build/qemux86/tmp-glibc/work/x86_64-linux/catkin-runtime-native/0.6.16-r0/temp/log.do_configure.6308)
```

Does anyone has a clue?

Thanks a lot in advance and best regards,
Tom

bulwahn, 2016-05-27:
Dear Tom,

Sorry for the late response; I have been busy with other projects, and sadly did not find any working time for the meta-ros layer.

I can confirm that I have also seen this error on the recent master branches of meta-openembedded and meta-oe on my build machine some weeks ago. Unfortunately, due to moving our office into a new building, my build machine has been switched off and was put into moving boxes for the weekend.

There are two options to get meta-ros working:

  1. Use the last released version of the poky distribution. My CI reports that the current meta-ros is fully working with that.

  2. Use git bisect on meta-oe and build meta-ros on the different changesets until you find which recent changeset in meta-oe causes this issue. The commit message of that changeset is probably going to give you some insights how to address the issue above.

If you pursue the second option, please share your insights here, as other users can probably help once issue is narrowed down to own specific change in the meta-oe layer.
Once my build machine is installed in the new office building, I can hopefully spend also some time on this issue as well.

bulwahn, 2016-05-31:
Dear Tom,

First, I bisected the issue assuming that there has been a change in oe-core that caused the issue, but actually it turned out that the issue occurred with various changesets back until March this year.

Then I looked into the actual steps of the failure:
In my case, cmake calls "/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/catkin-runtime/0.6.18-r0/build/devel/_setup_util.py" with the further arguments.
However, as I tried to reproduce that in my bash terminal, I noticed that /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/catkin-runtime/0.6.18-r0/build/devel/_setup_util.py fails (even without arguments) with:

-bash: /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/catkin-runtime/0.6.18-r0/build/devel/_setup_util.py: /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-gl: bad interpreter: Permission denied

The shebang line of the _setup_util.py script points to 'python-native/python' interpreter which is actually a quite long line (in my case: 134 characters); hence the script fails in its first line as shebang line limit is beyond my kernel's limit according to some stackoverflow answer [1].

Next, I simply removed the _setup_util.py invocation in the 
'File "/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/catkin-runtime/0.6.18-r0/catkin-0.6.18/cmake/../python/catkin/environment_cache.py", line 61, in generate_environment_script:

and shortened it to:

output = subprocess.check_output([sys.executable, '-c', python_code])

This hack did let that error message disappear; however, the quick hack only shows that the problem is part of these shell invocations and environment setup; it is not a solution to issue yet, as I am uncertain if you can simply drop the environment setup script safely.

My investigations will continue to check for further errors if the script is dropped or if the first line of the _setup_util.py script is modified to be shorter.

[1] http://stackoverflow.com/questions/10813538/shebang-line-limit-in-bash-and-linux-kernel

bulwahn, 2016-12-03:
As we did not encounter any further issue in this direction, we can consider this issue resolved with the merged commit that patches catkin.

closed 2016-12-03 by bulwahn