ros/meta-ros #388: sound-play: recipe recommends python-gst, but it is not available in any layers' master branch

bulwahn, 2016-06-01:
The python-gst recipe was once in the meta-multimedia layer, but was dropped in changeset 1ec074eca:

http://cgit.openembedded.org/meta-openembedded/commit/?id=1ec074eca0165de351ee45f173321f14408e3862

Without any python-gst recipe in any layer, the build of core-image-ros-world fails with:

```
ERROR: Nothing RPROVIDES 'python-gst' (but /home/lukas/work/repositories/github.com/bmwcarit/meta-ros/recipes-ros/audio-common/sound-play_0.2.9.bb RDEPENDS on or otherwise requires it)
NOTE: Runtime target 'python-gst' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['python-gst']
ERROR: Required build target 'sound-play' has no buildable providers.
Missing or unbuildable dependency chain was: ['sound-play', 'python-gst']
```

bulwahn, 2016-06-02:
If anyone needs sound-play with the recent versions, he/she must address this issue. I will not look into this issue.

herb-kuta-lge, 2020-02-07:
@bulwahn < Can this issue be closed?

closed 2020-02-21 by herb-kuta-lge