ros/meta-ros #389: libepoxy was skipped: missing required distro feature 'opengl' (not in DISTRO_FEATURES)

bulwahn, 2016-06-02:
All recipes that rely on cv-bridge and hence on opencv are unbuildable if the distro feature 'opengl' is not in DISTRO_FEATURES.

bitbake reports a number of NOTE errors during parsing of this kind:
```
NOTE: Runtime target 'compressed-depth-image-transport' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['compressed-depth-image-transport', 'cv-bridge', 'opencv', 'gtk+3', 'libepoxy']
```

The Build Configuration for this failure was:
```
BB_VERSION        = "1.31.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-14.04"
TARGET_SYS        = "i586-oe-linux"
MACHINE           = "qemux86"
DISTRO            = "nodistro"
DISTRO_VERSION    = "nodistro.0"
TUNE_FEATURES     = "m32 i586"
TARGET_FPU        = ""
meta              = "master:b746223787a0195c3a4d16523003c62ec0ac8451"
meta-oe           
meta-multimedia   
meta-python       = "master:aa8e156d56d487d5419dd353da50778d56d47703"
meta-ros          = "master:e8f3e9c255731a88a9001abcc8340695a4def583"
```

bulwahn, 2016-06-07:
The additional DISTRO_FEATURE check was added in commit 3edf08b38b:

http://cgit.openembedded.org/openembedded-core/commit/?id=3edf08b38b0af93cef0933b061349264dc86d54c

herb-kuta-lge, 2019-11-17:
`libepoxy` builds successfully for **meldoic** when `opengl` appears in `DISTRO_FEATURES` (as it does when `DISTRO` is `webos`). @bulwahn < Can this issue be closed?

closed 2019-12-03 by herb-kuta-lge