ros/meta-ros #390: log4cxx: compile errors with gcc version 6

bulwahn, 2016-06-02:
| ../../../i586-oe-linux-libtool  --tag=CXX   --mode=compile i586-oe-linux-g++  -m32 -march=i586 --sysroot=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86 -DPACKAGE_NAME=\"\" -DPACKAGE_TARNAME=\"\" -DPACKAGE_VERSION=\"\" -DPACKAGE_STRING=\"\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -DPACKAGE=\"log4cxx\" -DVERSION=\"0.10.0\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DHAVE_DLFCN_H=1 -DLT_OBJDIR=\".libs/\" -DHAVE_MBSRTOWCS=1 -DHAVE_WCSTOMBS=1 -DHAVE_SYSLOG=1 -DHAVE_FWIDE=1 -I. -I../../../src/main/include -I../../../src/main/include  -DLINUX -D_REENTRANT -D_GNU_SOURCE  -I/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86/usr/include   -I/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86/usr/include -I/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86/usr/include  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/log4cxx/0.10.0-r0=/usr/src/debug/log4cxx/0.10.0-r0 -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/x86_64-linux= -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86=  -fvisibility-inlines-hidden -c -o rootlogger.lo rootlogger.cpp
| locationinfo.cpp: In member function 'void log4cxx::spi::LocationInfo::write(log4cxx::helpers::ObjectOutputStream&, log4cxx::helpers::Pool&) const':
| locationinfo.cpp:163:21: error: narrowing conversion of '237' from 'int' to 'char' inside { } [-Wnarrowing]
|           0x78, 0x70 };
|                      ^
| locationinfo.cpp:163:21: error: narrowing conversion of '153' from 'int' to 'char' inside { } [-Wnarrowing]
| locationinfo.cpp:163:21: error: narrowing conversion of '187' from 'int' to 'char' inside { } [-Wnarrowing]
| locationinfo.cpp:163:21: error: narrowing conversion of '225' from 'int' to 'char' inside { } [-Wnarrowing]
| locationinfo.cpp:163:21: error: narrowing conversion of '145' from 'int' to 'char' inside { } [-Wnarrowing]
| locationinfo.cpp:163:21: error: narrowing conversion of '165' from 'int' to 'char' inside { } [-Wnarrowing]
| make[3]: *** [locationinfo.lo] Error 1
| make[3]: *** Waiting for unfinished jobs....
| objectoutputstream.cpp: In constructor 'log4cxx::helpers::ObjectOutputStream::ObjectOutputStream(log4cxx::helpers::OutputStreamPtr, log4cxx::helpers::Pool&)':
| objectoutputstream.cpp:39:44: error: narrowing conversion of '172' from 'int' to 'char' inside { } [-Wnarrowing]
|     char start[] = { 0xAC, 0xED, 0x00, 0x05 };
|                                             ^
| objectoutputstream.cpp:39:44: error: narrowing conversion of '237' from 'int' to 'char' inside { } [-Wnarrowing]
| objectoutputstream.cpp: In member function 'void log4cxx::helpers::ObjectOutputStream::writeObject(const Map&, log4cxx::helpers::Pool&)':
| objectoutputstream.cpp:92:39: error: narrowing conversion of '187' from 'int' to 'char' inside { } [-Wnarrowing]
|          0x6F, 0x6C, 0x64, 0x78, 0x70  };
|                                        ^
| objectoutputstream.cpp:92:39: error: narrowing conversion of '228' from 'int' to 'char' inside { } [-Wnarrowing]
| objectoutputstream.cpp:92:39: error: narrowing conversion of '184' from 'int' to 'char' inside { } [-Wnarrowing]

bulwahn, 2016-06-02:
This issue was already reported and resolved in:

https://issues.apache.org/jira/browse/LOGCXX-400

closed 2016-06-02 by KristofRobot

aleafboat, 2019-04-30:
how to fix

aleafboat, 2019-04-30:
The problem here is that some of the values do not fit into char if you use **unsigned char** instead it should work

commit b4baa4a8 (2016-06-02): log4cxx: update to current svn version (resolves #390)

The svn version of log4cxx does not require the previous patches
and hence, they are dropped.