ros/meta-ros #392: errors during compiling a number of recipes due to change default GCC from 5.3 to 6.1

bulwahn, 2016-06-03:
On 2015-05-25, the commit c20d86 [1] in openembedded-core changed the default GCC from 5.3 to 6.1.
This causes a number of recipes to fail during compiling:

TODO: add list of all recipes known to fail due to the new GCC version
(tf and more...)

[1] http://cgit.openembedded.org/openembedded-core/commit/?id=c20d863da57004e6824a03bcf53d6cf24ba8815b

KristofRobot, 2016-07-15:
As a temporary work around, you can override the default GCC in your ``local.conf``:

```
GCCVERSION = "5.%"
```

And comment out the blacklisting of opencv in ``meta-openembedded/recipes-support/opencv/opencv_3.1.bb``

```
#PNBLACKLIST[opencv] ?= "BROKEN: fails to build with gcc-6"
```

bulwahn, 2016-09-22:
This issue still occurs in the following recipes with the initiated solutions:
  -  ROS packages: 
     - Required upstream changes for tf, urdf, geometric_shapes, polled_camera, dynamic_reconfigure, image_geometry, cv_bridge are tracked at https://github.com/ros/rosdistro/issues/12783.
     - diagnostic-aggregator is fixed upstream and only needs update of the recipe. For this to happen, we must re-visit gtest issue in diagnostics to make the update possible.
  - Other software:
    - libflann just needs an update of the recipe.
    - collada_dom seems not further maintained at sourceforge, another (inofficial) fork exists in github.
    - it seems that pcl and opencv have been fixed already since June.

bulwahn, 2016-09-23:
The diagnostic-aggregator recipe is updated, cf. pull request #414.

commit 621669d0 (2016-10-15): Merge pull request #423 from bulwahn/support-gcc6

patch all ROS packages to build with gcc6 (resolves #392)
closed 2016-10-15 by KristofRobot