ros/meta-ros #395: Python-Imaging Error

mdangler, 2016-06-13:
Hello

May be asking in the wrong place but trying to build a ROS project in Yocto/Poky and always hit the same failure with python-imaging. Seems the one calling for it is rosbag but even without a ROS request I get the following - 

michael@michael-Kudu-Pro:~/yocto/build$ bitbake python-imaging
NOTE: Started PRServer with DBfile: /home/michael/yocto/build/cache/prserv.sqlite3, IP: 127.0.0.1, PORT: 37498, PID: 12511
Parsing recipes: 100% |##########################################################################################################################################################################| Time: 00:00:58
Parsing of 2312 .bb files complete (0 cached, 2312 parsed). 2852 targets, 150 skipped, 9 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION        = "1.26.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-14.04"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "overo"
DISTRO            = "poky"
DISTRO_VERSION    = "1.8"
TUNE_FEATURES     = "arm armv7a vfp neon callconvention-hard cortexa8"
TARGET_FPU        = "vfp-neon"
meta              
meta-yocto        = "(nobranch):83aa565d93aacae484976562ef1ae8dbbb6b2bc0"
meta-gnome        
meta-oe           
meta-xfce         
meta-systemd      
meta-networking   
meta-multimedia   
meta-python       = "(nobranch):10d3c8f85280a0bf867a8e4f84bcda81c290d28e"
meta-gumstix      = "(nobranch):8ea631d8ef26cb13102dac0910be52674cb7b395"
meta-gumstix-extras = "(nobranch):15cfa9fcff75d3d95f66cfa190e1b604ff5ea4f6"
meta-ros          = "master:a5a19128a82a1c04a65d06eb615dbcf254dd1ae8"
meta-ti           = "(nobranch):03cb3ff15f3b8535ca409ce682f44ec1b81da0c3"
meta-java         = "(nobranch):66c97ae7461f4c1a13917131787bb76dc34e6b6f"

NOTE: Preparing RunQueue
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: QA Issue: python-imaging: The compile log indicates that host include and/or library paths were used.
         Please check the log '/home/michael/yocto/build/tmp/work/cortexa8hf-vfp-neon-poky-linux-gnueabi/python-imaging/1.1.7-r5/temp/log.do_compile' for more information. [compile-host-path]
ERROR: QA run found fatal errors. Please consider fixing them.
ERROR: Function failed: do_package_qa
ERROR: Logfile of failure stored in: /home/michael/yocto/build/tmp/work/cortexa8hf-vfp-neon-poky-linux-gnueabi/python-imaging/1.1.7-r5/temp/log.do_package_qa.22115
ERROR: Task 9 (/home/michael/yocto/poky/meta-ros/recipes-devtools/python/python-imaging_1.1.7.bb, do_package_qa) failed with exit code '1'
NOTE: Tasks Summary: Attempted 715 tasks of which 0 didn't need to be rerun and 1 failed.
Waiting for 0 running tasks to finish:

Summary: 1 task failed:
  /home/michael/yocto/poky/meta-ros/recipes-devtools/python/python-imaging_1.1.7.bb, do_package_qa

The failure seems to be with the following - 

cc1: warning: include location "/usr/include/tcl8.6" is unsafe for cross-compilation [-Wpoison-system-directories]

Any help would be great, or if you can point me in the right direction. A little lost since OE pulled python-imaging from there index. Thank you.

Cheers, Michael

Minipada, 2016-09-18:
It is a bit old, but I had the same issue when updating. Thanks for the hint of the tcl library, I confirmed noticing **/usr/include/tcl8.6**, this patch solves it on my machine

yongli3, 2017-03-06:
I encountered the same issue too, I think the patch should be merged :)

bulwahn, 2017-03-06:
@yongli3 Could you please report your build configuration, and the reasons why you chose the build configuration?
I never encountered a build configuration to reproduce the issue.

yongli3, 2017-03-06:
@bulwahn  I am using Ubuntu 1604 64 bit as the bitbake host system, the latest yocto(poky) version.

bulwahn, 2017-03-06:
@yongli3 Please provide the revisions of all layers, just as bitbake prints them at the start of a build.

bulwahn, 2017-03-06:
For this issue, note that considered merging the proposed patch in pull request #418 and then decided to drop the python-imaging recipe in meta-ros in pull request #421 instead.

yongli3, 2017-03-06:
@bulwahn Below is the log for your reference:
Build Configuration:
BB_VERSION        = "1.33.2"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "universal"
TARGET_SYS        = "x86_64-poky-linux"
MACHINE           = "intel-corei7-64"
DISTRO            = "poky"
DISTRO_VERSION    = "2.2+snapshot-20170306"
TUNE_FEATURES     = "m64 corei7"
TARGET_FPU        = ""
meta              
meta-poky         = "master:4188e53f803aefdb26768abfad591283662cd27e"
meta-x1000-mips   = "master:6a9aa4cfc8055e2927b2ed50e57fda88bf12412a"
meta-oe           
meta-multimedia   
meta-python       
meta-networking   = "master:131255dd561225d7e4ebbf5155420a685dfebf6e"
meta-intel        = "master:f6e8abbfdee736e273d2943fceaec828501b5606"
meta-ros          = "master:8a3adc1a6542f2c8a40b36021a78dc1057e3ed08"

bulwahn, 2017-03-07:
@yongli3 As I currently do not have access to my build machine, I cannot check the issue with the reported build configuration this week, but only next week. However, as you are using the current master of meta-ros, which does not provide the python-imaging recipe anymore, the patch for the tcl linking issue should be contributed to the meta-python layer in the meta-openembedded repository. There is however little to do in the meta-ros repository to address this python-imaging issue. You, just as anyone else who encounters the python-imaging issue, can submit a commit to the meta-openembedded mailing list to address the issue. In pull request #418, we discussed some improvements to the original patch. If you submit a patch to the meta-openembedded repository, you might consider taking the comments from #418 into account.

herb-kuta-lge, 2019-11-14:
`python-imaging` successfully builds for **melodic** using `[thud-draft]`. @mdangler < Can I close this issue?

closed 2019-12-03 by herb-kuta-lge