ros/meta-ros #396: image-view: fails due to missing gtk header files

bulwahn, 2016-06-14:
The updated image-view recipe with version 1.12.16 fails with:

| [ 76%] Building CXX object CMakeFiles/image_view.dir/src/nodelets/image_nodelet.cpp.o
| [ 82%] Building CXX object CMakeFiles/image_view.dir/src/nodelets/disparity_nodelet.cpp.o
| /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/x86_64-linux/usr/bin/i586-oe-linux/i586-oe-linux-g++   -DHAVE_GTK -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"image_view\" -Dimage_view_EXPORTS -I/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/image-view/1.12.16-r0/build/devel/include -isystem /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86/usr/include -isystem /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86/opt/ros/indigo/include -isystem /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86/usr/include/opencv  -m32 -march=i586  --sysroot=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/image-view/1.12.16-r0=/usr/src/debug/image-view/1.12.16-r0 -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/x86_64-linux= -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86=  -fvisibility-inlines-hidden -fPIC   -o CMakeFiles/image_view.dir/src/nodelets/image_nodelet.cpp.o -c /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/image-view/1.12.16-r0/image_pipeline-1.12.16/image_view/src/nodelets/image_nodelet.cpp
| /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/x86_64-linux/usr/bin/i586-oe-linux/i586-oe-linux-g++   -DHAVE_GTK -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"image_view\" -Dimage_view_EXPORTS -I/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/image-view/1.12.16-r0/build/devel/include -isystem /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86/usr/include -isystem /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86/opt/ros/indigo/include -isystem /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86/usr/include/opencv  -m32 -march=i586  --sysroot=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/image-view/1.12.16-r0=/usr/src/debug/image-view/1.12.16-r0 -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/x86_64-linux= -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86=  -fvisibility-inlines-hidden -fPIC   -o CMakeFiles/image_view.dir/src/nodelets/disparity_nodelet.cpp.o -c /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/image-view/1.12.16-r0/image_pipeline-1.12.16/image_view/src/nodelets/disparity_nodelet.cpp
| /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/image-view/1.12.16-r0/image_pipeline-1.12.16/image_view/src/nodelets/disparity_nodelet.cpp:42:21: fatal error: gtk/gtk.h: No such file or directory
| compilation terminated.
| make[2]: *** [CMakeFiles/image_view.dir/src/nodelets/disparity_nodelet.cpp.o] Error 1
| make[2]: *** Waiting for unfinished jobs....
| /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/image-view/1.12.16-r0/image_pipeline-1.12.16/image_view/src/nodelets/image_nodelet.cpp:46:21: fatal error: gtk/gtk.h: No such file or directory
| compilation terminated.
| make[2]: *** [CMakeFiles/image_view.dir/src/nodelets/image_nodelet.cpp.o] Error 1
| make[2]: Leaving directory `/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/image-view/1.12.16-r0/build'
| make[1]: *** [CMakeFiles/image_view.dir/all] Error 2
| make[1]: *** Waiting for unfinished jobs....
| /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/image-view/1.12.16-r0/image_pipeline-1.12.16/image_view/src/nodes/stereo_view.cpp:53:21: fatal error: gtk/gtk.h: No such file or directory
| compilation terminated.
| make[2]: *** [CMakeFiles/stereo_view.dir/src/nodes/stereo_view.cpp.o] Error 1
| make[2]: Leaving directory `/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/image-view/1.12.16-r0/build'
| make[1]: *** [CMakeFiles/stereo_view.dir/all] Error 2
| make[1]: Leaving directory `/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/image-view/1.12.16-r0/build'
| make: *** [all] Error 2
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/image-view/1.12.16-r0/temp/log.do_compile.12246)
ERROR: Task 7 (/home/lukas/work/repositories/github.com/bmwcarit/meta-ros/recipes-ros/image-pipeline/image-view_1.12.16.bb, do_compile) failed with exit code '1'

JochiPochi, 2016-08-01:
Interesting.
I can build image-view 1.12.16 just fine in my environment.
my bblayers.conf looks like this:
`LCONF_VERSION = "6"

BBPATH = "${TOPDIR}"
BSPDIR := "${@os.path.abspath(os.path.dirname(d.getVar('FILE', True)) + '/../..')}"

BBFILES ?= ""
BBLAYERS = " \
  ${BSPDIR}/sources/poky/meta \
  ${BSPDIR}/sources/poky/meta-yocto \
  \
  ${BSPDIR}/sources/meta-openembedded/meta-oe \
  ${BSPDIR}/sources/meta-openembedded/meta-multimedia \
  ${BSPDIR}/sources/meta-openembedded/meta-python \
  ${BSPDIR}/sources/meta-openembedded/meta-networking \
  \
  ${BSPDIR}/sources/meta-fsl-arm \
  ${BSPDIR}/sources/meta-fsl-arm-extra \
  ${BSPDIR}/sources/meta-fsl-demos \
  \
  ${BSPDIR}/sources/meta-ros \
"
`

JochiPochi, 2016-08-01:
however it doesn't behave the same during runtime as the previous version of the image_pipeline. Thet error I get is:

`Attempt to unlock mutex that was not locked
Aborted`

And then image_view exists

JochiPochi, 2016-08-01:
I'll poke around some more to see what's up

bulwahn, 2016-08-07:
I usually test with the latest upstream openembedded-core repository versions. So to reproduce the issue above, you would need to use the openembedded-core repository (and not poky) probably at some changeset around June 14th when I opened this issue. If you would like to dig deeper into this issue, it would be good to know if the issue still occurs with the current upstream version.

The runtime error you report is a different one to one above (but possibly related), maybe there is already some discussion to your reported issue in the issue tracker of image_view or on ros question website.

bulwahn, 2016-12-02:
Today, I revisited this issue. I tried to `bitbake image-view` and it still fails with:
```
| [ 76%] Building CXX object CMakeFiles/image_view.dir/src/nodelets/image_nodelet.cpp.o
| [ 82%] Building CXX object CMakeFiles/image_view.dir/src/nodelets/disparity_nodelet.cpp.o
| /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/x86_64-linux/usr/bin/i586-oe-linux/i586-oe-linux-g++   -DHAVE_GTK -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"image_view\" -Dimage_view_EXPORTS -I/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/image-view/1.12.19-r0/build/devel/include -I/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86/usr/include -I/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86/opt/ros/indigo/include -I/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86/usr/include/opencv  -m32 -march=i586  --sysroot=/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/image-view/1.12.19-r0=/usr/src/debug/image-view/1.12.19-r0 -fdebug-prefix-map=/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/x86_64-linux= -fdebug-prefix-map=/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86=  -fvisibility-inlines-hidden  -m32 -march=i586  --sysroot=/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86 -fPIC   -o CMakeFiles/image_view.dir/src/nodelets/image_nodelet.cpp.o -c /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/image-view/1.12.19-r0/image_pipeline-1.12.19/image_view/src/nodelets/image_nodelet.cpp
| ERROR: oe_runmake failed
| /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/x86_64-linux/usr/bin/i586-oe-linux/i586-oe-linux-g++   -DHAVE_GTK -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"image_view\" -Dimage_view_EXPORTS -I/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/image-view/1.12.19-r0/build/devel/include -I/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86/usr/include -I/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86/opt/ros/indigo/include -I/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86/usr/include/opencv  -m32 -march=i586  --sysroot=/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/image-view/1.12.19-r0=/usr/src/debug/image-view/1.12.19-r0 -fdebug-prefix-map=/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/x86_64-linux= -fdebug-prefix-map=/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86=  -fvisibility-inlines-hidden  -m32 -march=i586  --sysroot=/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86 -fPIC   -o CMakeFiles/image_view.dir/src/nodelets/disparity_nodelet.cpp.o -c /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/image-view/1.12.19-r0/image_pipeline-1.12.19/image_view/src/nodelets/disparity_nodelet.cpp
| /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/image-view/1.12.19-r0/image_pipeline-1.12.19/image_view/src/nodelets/disparity_nodelet.cpp:42:21: fatal error: gtk/gtk.h: No such file or directory
|  #include <gtk/gtk.h>
|                      ^
| compilation terminated.
| CMakeFiles/image_view.dir/build.make:89: recipe for target 'CMakeFiles/image_view.dir/src/nodelets/disparity_nodelet.cpp.o' failed
| make[2]: *** [CMakeFiles/image_view.dir/src/nodelets/disparity_nodelet.cpp.o] Error 1
| make[2]: *** Waiting for unfinished jobs....
| /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/image-view/1.12.19-r0/image_pipeline-1.12.19/image_view/src/nodelets/image_nodelet.cpp:46:21: fatal error: gtk/gtk.h: No such file or directory
|  #include <gtk/gtk.h>
|                      ^
| compilation terminated.
| CMakeFiles/image_view.dir/build.make:65: recipe for target 'CMakeFiles/image_view.dir/src/nodelets/image_nodelet.cpp.o' failed
| make[2]: *** [CMakeFiles/image_view.dir/src/nodelets/image_nodelet.cpp.o] Error 1
| make[2]: Leaving directory '/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/image-view/1.12.19-r0/build'
| CMakeFiles/Makefile2:1065: recipe for target 'CMakeFiles/image_view.dir/all' failed
| make[1]: *** [CMakeFiles/image_view.dir/all] Error 2
| make[1]: *** Waiting for unfinished jobs....
| /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/image-view/1.12.19-r0/image_pipeline-1.12.19/image_view/src/nodes/stereo_view.cpp:53:21: fatal error: gtk/gtk.h: No such file or directory
|  #include <gtk/gtk.h>
|                      ^
| compilation terminated.
| CMakeFiles/stereo_view.dir/build.make:65: recipe for target 'CMakeFiles/stereo_view.dir/src/nodes/stereo_view.cpp.o' failed
| make[2]: *** [CMakeFiles/stereo_view.dir/src/nodes/stereo_view.cpp.o] Error 1
[... cut ...]
TARGET_SYS        = "i586-oe-linux"
MACHINE           = "qemux86"
DISTRO            = "nodistro"
DISTRO_VERSION    = "nodistro.0"
TUNE_FEATURES     = "m32 i586"
TARGET_FPU        = ""
meta              = "master:9f1fe76727e98e58fc9e46ea2b49cf5c0cb48e6c"
meta-oe           
meta-python       
meta-multimedia   = "master:761639b9d7681c81dd69eaf3a37c32791d6e97fd"
meta-intel-realsense = "master:c74766ff2dc7383562bdd4aa1f10687c94261c53"
meta-ros          = "indigo-updates:8c8349b317992caa9e961d2a843ffa13f4dfe1d1"
```

herb-kuta-lge, 2020-02-07:
@bulwahn < Can this issue be closed?

closed 2020-02-21 by herb-kuta-lge