ros/meta-ros #399: Error: Cross-compile ros package for beagleond black

huyht1205, 2016-06-15:
Hello,

Compiling newly written ros package error:

[package.zip](https://github.com/bmwcarit/meta-ros/files/315740/package.zip)

> ...
| /data/02_angstrom/setup-scripts/build/tmp-angstrom_v2014_12-glibc/sysroots/x86_64-linux/usr/libexec/arm-angstrom-linux-gnueabi/gcc/arm-angstrom-linux-gnueabi/4.9.3/ld: error: /opt/ros/indigo/lib/libroscpp.so: incompatible target
| /data/02_angstrom/setup-scripts/build/tmp-angstrom_v2014_12-glibc/sysroots/x86_64-linux/usr/libexec/arm-angstrom-linux-gnueabi/gcc/arm-angstrom-linux-gnueabi/4.9.3/ld: error: /opt/ros/indigo/lib/librosconsole.so: incompatible target
| /data/02_angstrom/setup-scripts/build/tmp-angstrom_v2014_12-glibc/sysroots/x86_64-linux/usr/libexec/arm-angstrom-linux-gnueabi/gcc/arm-angstrom-linux-gnueabi/4.9.3/ld: error: /opt/ros/indigo/lib/librosconsole_log4cxx.so: incompatible target

Please see more detailed log in attached file and also my package.
FYI:
Git repo:
beagle-ros: branch v1.1
   |---setup-script: bfb121b555d4a6160da39d709de330b9d568170e

Thank you very much.
Best regards,
Huy Hoang

bulwahn, 2016-06-16:
Dear Huy, your package.zip file is 166 bytes large and carries no content. In general, I feel uncomfortable to open any .zip files locally on my machine. Please provide a github fork of your development and a README how to re-construct your issue. As far as I know, beagle-ros is currently unmaintained, but probably large part of the installation can still be followed. 
Angstrom and meta-ros are maintained, however, no one has checked if they still are compatible with each other. meta-ros is currently only maintained against the master branch and the fido branch (as far as possible).

huyht1205, 2016-06-16:
Dear Lukas,

Thank you very much for your reply and I'm sorry for the inconvenience zip file caused.
Please let me report the issue again as below:

My target: cross-compile my C++ package for to run with ros on beaglebond black.
What I did:

**Step-1:** Clone and configure "angstrom-v2014.12-yocto1.7". Refer: http://www.angstrom-distribution.org/building-angstrom/
- $ git clone git://github.com/Angstrom-distribution/setup-scripts.git
- $ MACHINE=beagleboard ./oebb.sh config beagleboard
- $ MACHINE=beagleboard ./oebb.sh update
- $ cd $BUILD_DIR/setup-scripts && source environment-angstrom-v2014.12

**Step-2**: Clone "beagle-ros"
- $ git clone https://github.com/vmayoral/beagle-ros.git
- Add "beagle-ros" to $BUILD_DIR/setup-scripts/conf/bblayers.conf

**Step-3**: Clone and replace "beagle-ros/recipe-ros/beginner-tutorials" by "meta-ros-example"
- $ git clone https://github.com/ymkim92/meta-ros-example.git
- $ rm $BUILD_DIR/beagle-ros/recipe-ros/beginner-tutorials/*
- $ cp -r $BUILD_DIR/meta-ros-example/* $BUILD_DIR/beagle-ros/recipe-ros/beginner-tutorials/

**Step-4**: Build "beginner-tutorials" by bitbake
- $ cd $BUILD_DIR/setup-scripts/build
- $ bitbake beginner-tutorials => error
[error_compile_log.txt](https://github.com/bmwcarit/meta-ros/files/318129/error_compile_log.txt)

Thank you so much for your help.
Best regards,
Huy Hoang

bulwahn, 2016-06-20:
Dear Huy,

do you ever clone the meta-ros layer and add meta-ros to bblayers.conf? This seems to be missing completely somewhere between step 1 and step 3?

huyht1205, 2016-06-21:
Dear Lukas,

Thank you.

> do you ever clone the meta-ros layer and add meta-ros to bblayers.conf?

=> I think yes. The script clones it for me automatically.

Actually, I followed instructions in Angstrom distribution (http://www.angstrom-distribution.org/building-angstrom/) with following commands:
- MACHINE=beagleboard ./oebb.sh config beagleboard
- MACHINE=beagleboard ./oebb.sh update

After that, "meta-ros" other "meta-*" are cloned and added into "bblayers.conf" as below:

\# LAYER_CONF_VERSION is increased each time build/conf/bblayers.conf
\# changes incompatibly
LCONF_VERSION = "5"
TOPDIR := "${@os.path.dirname(os.path.dirname(d.getVar('FILE', True)))}"

BBPATH = "${TOPDIR}"

BBFILES = ""

\# These layers hold recipe metadata not found in OE-core, but lack any machine or distro content
BASELAYERS ?= " \
  ${TOPDIR}/sources/meta-openembedded/meta-oe \
  ${TOPDIR}/sources/meta-openembedded/meta-efl \
  ${TOPDIR}/sources/meta-openembedded/meta-gpe \
  ${TOPDIR}/sources/meta-openembedded/meta-gnome \
  ${TOPDIR}/sources/meta-openembedded/meta-xfce \
  ${TOPDIR}/sources/meta-openembedded/meta-initramfs \
  ${TOPDIR}/sources/meta-openembedded/toolchain-layer \
  ${TOPDIR}/sources/meta-openembedded/meta-multimedia \
  ${TOPDIR}/sources/meta-openembedded/meta-networking \
  ${TOPDIR}/sources/meta-openembedded/meta-webserver \
  ${TOPDIR}/sources/meta-openembedded/meta-ruby \
  ${TOPDIR}/sources/meta-openembedded/meta-filesystems \
  ${TOPDIR}/sources/meta-openembedded/meta-perl \
  ${TOPDIR}/sources/meta-openembedded/meta-python \
  ${TOPDIR}/sources/meta-kde4 \
  ${TOPDIR}/sources/meta-java \
  ${TOPDIR}/sources/meta-browser \
  ${TOPDIR}/sources/meta-mono \
  ${TOPDIR}/sources/meta-qt5 \
  ${TOPDIR}/sources/meta-openembedded/meta-systemd \
  ${TOPDIR}/sources/meta-ros \
  ${TOPDIR}/sources/meta-uav \
  ${TOPDIR}/sources/meta-telephony \
  ${TOPDIR}/sources/meta-beagleboard/meta-beagleboard-extras \
  ${TOPDIR}/sources/meta-photography \
  ${TOPDIR}/sources/meta-ci \
  ${TOPDIR}/sources/meta-intel-iot-middleware \
  ${TOPDIR}/sources/meta-maker \
"

\# These layers hold machine specific content, aka Board Support Packages
BSPLAYERS ?= " \
  ${TOPDIR}/sources/meta-beagleboard/common-bsp \
  ${TOPDIR}/sources/meta-ti \
  ${TOPDIR}/sources/meta-fsl-arm \
  ${TOPDIR}/sources/meta-fsl-arm-extra \
  ${TOPDIR}/sources/meta-nslu2 \
  ${TOPDIR}/sources/meta-smartphone/meta-htc \
  ${TOPDIR}/sources/meta-smartphone/meta-nokia \
  ${TOPDIR}/sources/meta-smartphone/meta-openmoko \
  ${TOPDIR}/sources/meta-smartphone/meta-palm \
  ${TOPDIR}/sources/meta-handheld \
  ${TOPDIR}/sources/meta-intel \
  ${TOPDIR}/sources/meta-intel/meta-sugarbay \
  ${TOPDIR}/sources/meta-intel/meta-crownbay \
  ${TOPDIR}/sources/meta-intel/meta-emenlow \
  ${TOPDIR}/sources/meta-intel/meta-fri2 \
  ${TOPDIR}/sources/meta-intel/meta-jasperforest \
  ${TOPDIR}/sources/meta-sunxi \
  ${TOPDIR}/sources/meta-raspberrypi \
  ${TOPDIR}/sources/meta-minnow \
  ${TOPDIR}/sources/meta-dominion \
  ${TOPDIR}/sources/meta-atmel \
  ${TOPDIR}/sources/meta-exynos \
  ${TOPDIR}/sources/meta-gumstix-community \
  ${TOPDIR}/sources/meta-qualcomm \
  ${TOPDIR}/sources/meta-edison \
  ${TOPDIR}/sources/meta-96boards \
"

\# Add your overlay location to EXTRALAYERS
\# Make sure to have a conf/layers.conf in there
EXTRALAYERS ?= " \
  ${TOPDIR}/sources/meta-linaro/meta-linaro \
  ${TOPDIR}/sources/meta-linaro/meta-linaro-toolchain \
  ${TOPDIR}/sources/meta-linaro/meta-aarch64 \
"

BBLAYERS = " \
  ${TOPDIR}/sources/meta-angstrom \
  ${BASELAYERS} \
  ${BSPLAYERS} \
  ${EXTRALAYERS} \
  ${TOPDIR}/sources/openembedded-core/meta \
  "

Thank you very much for your time.
Best regards,
Huy Hoang

bulwahn, 2016-06-21:
Okay. Here's a first guess and some hints that might solve the problem:

It seems that when building your package bitbake links against the locally installed packages instead of the ones that should be provided by the bitbake tool chain.
So avoid that error, you can try to:
- add needed dependencies in your recipe (as the recipe's dependencies are preferred before searching in the local system.)
- build your image on a machine that has no ROS packages installed (a dedicated build machine). Then, the compilation will fail rather than linking against the wrong installation.
- update the layers.txt in Angstrom to the current commit of meta-ros and do some minor layer compatibility tuning. The meta-ros version you use is a couple of months old, and possibly the issue has been addressed by recent improvements and changes.

I hope this helps.

huyht1205, 2016-06-27:
Dear Lukas,

Thank you very much for your information.

   - I checked and confirmed that dependencies is mentioned in recipe by
   _DEPENDS = "roscpp catkin std-msgs"_. I also removed native ROS packages
     and tried using newer version of meta-ros but all of them seem to be not enough for this issue.

   - I also tried manually cross compiling and met wrong linking library issue.
     As a result, I think your points about linking libraries is very potentially the direct cause
     and "version compatible of layers" should be the root cause.

=> I planned to check the suitable version and go deep into logs to understanding what happened.

However, due to the time constrain of my project, we need to go forward to next phase by using "qemu-arm-static" and "chroot" as work around solution for cross compilation.
So I would like to temporarily close this issue here and I will come back to this issue later when I have more time.

Thank you very much for your help. I really appreciate it!
Best regards,
Huy Hoang.

closed 2016-06-27 by huyht1205