ros/meta-ros #400: sphinxbase: linking libsphinxad.so fails due to undefined reference in do_compile

bulwahn, 2016-06-20:
The recipe sphinxbase currently fails with the build configuration:
```
Build Configuration:
BB_VERSION        = "1.31.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-14.04"
TARGET_SYS        = "i586-oe-linux"
MACHINE           = "qemux86"
DISTRO            = "nodistro"
DISTRO_VERSION    = "nodistro.0"
TUNE_FEATURES     = "m32 i586"
TARGET_FPU        = ""
meta              = "master:fba321ae9a1916d64f5c49c65c60c276f3346e7a"
meta-oe           
meta-multimedia   
meta-python       = "master:9cd117a5dc5f26133a855d1cc3bf9604ae178685"
meta-ros          = "ci:c8b0688503be4f30e9ea525ab6b8cf645a2dc78f"
```

The complete log is:
```
| Making all in sphinx_adtools
| make[2]: Entering directory `/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/sphinxbase/0.8-r0/build/src/sphinx_adtools'
| ../../i586-oe-linux-libtool  --tag=CC   --mode=link i586-oe-linux-gcc  -m32 -march=i586 --sysroot=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/sphinxbase/0.8-r0=/usr/src/debug/sphinxbase/0.8-r0 -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/x86_64-linux= -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86=   -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -o sphinx_cont_adseg cont_adseg.o ../../src/libsphinxbase/libsphinxbase.la ../../src/libsphinxad/libsphinxad.la -lpthread -lm
| ../../i586-oe-linux-libtool  --tag=CC   --mode=link i586-oe-linux-gcc  -m32 -march=i586 --sysroot=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/sphinxbase/0.8-r0=/usr/src/debug/sphinxbase/0.8-r0 -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/x86_64-linux= -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86=   -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -o sphinx_cont_fileseg cont_fileseg.o ../../src/libsphinxbase/libsphinxbase.la ../../src/libsphinxad/libsphinxad.la -lpthread -lm
| i586-oe-linux-libtool: link: i586-oe-linux-gcc -m32 -march=i586 --sysroot=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86 -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/sphinxbase/0.8-r0=/usr/src/debug/sphinxbase/0.8-r0 -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/x86_64-linux= -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86= -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -o .libs/sphinx_cont_fileseg cont_fileseg.o  ../../src/libsphinxbase/.libs/libsphinxbase.so ../../src/libsphinxad/.libs/libsphinxad.so /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86/usr/lib/libsamplerate.so /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/sphinxbase/0.8-r0/build/src/libsphinxbase/.libs/libsphinxbase.so -lpthread -lm
| i586-oe-linux-libtool: link: i586-oe-linux-gcc -m32 -march=i586 --sysroot=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86 -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/sphinxbase/0.8-r0=/usr/src/debug/sphinxbase/0.8-r0 -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/x86_64-linux= -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86= -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -o .libs/sphinx_cont_adseg cont_adseg.o  ../../src/libsphinxbase/.libs/libsphinxbase.so ../../src/libsphinxad/.libs/libsphinxad.so /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86/usr/lib/libsamplerate.so /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/sphinxbase/0.8-r0/build/src/libsphinxbase/.libs/libsphinxbase.so -lpthread -lm
| ../../src/libsphinxad/.libs/libsphinxad.so: undefined reference to `pa_strerror'
| ../../src/libsphinxad/.libs/libsphinxad.so: undefined reference to `pa_simple_read'
| ../../src/libsphinxad/.libs/libsphinxad.so: undefined reference to `pa_simple_new'
| ../../src/libsphinxad/.libs/libsphinxad.so: undefined reference to `pa_simple_free'
| ../../src/libsphinxad/.libs/libsphinxad.so: undefined reference to `pa_strerror'
| ../../src/libsphinxad/.libs/libsphinxad.so: undefined reference to `pa_simple_read'
| ../../src/libsphinxad/.libs/libsphinxad.so: undefined reference to `pa_simple_new'
| ../../src/libsphinxad/.libs/libsphinxad.so: undefined reference to `pa_simple_free'
| collect2: error: ld returned 1 exit status
| collect2: error: ld returned 1 exit status
| make[2]: *** [sphinx_cont_fileseg] Error 1
| make[2]: *** Waiting for unfinished jobs....
| make[2]: *** [sphinx_cont_adseg] Error 1
| make[2]: Leaving directory `/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/sphinxbase/0.8-r0/build/src/sphinx_adtools'
| make[1]: *** [all-recursive] Error 1
| make[1]: Leaving directory `/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/sphinxbase/0.8-r0/build/src'
| make: *** [all-recursive] Error 1
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/sphinxbase/0.8-r0/temp/log.do_compile.27713)
ERROR: Task 5943 (/home/lukas/work/repositories/github.com/bmwcarit/meta-ros/recipes-extended/cmusphinx/sphinxbase_0.8.bb, do_compile) failed with exit code '1'
```

herb-kuta-lge, 2020-02-21:
@bulwahn < Can this issue be closed?

shr-project, 2020-07-22:
"Silence implies consent" (see http://wiki.c2.com/?SilenceImpliesConsent).

closed 2020-07-22 by shr-project