ros/meta-ros #408: collada_urdf: fatal error: Eigen/Core: No such file or directory

bulwahn, 2016-09-05:
On the experimental kinetic branch, compiling collada_urdf currently fails with:

```
| [ 33%] Building CXX object CMakeFiles/collada_urdf.dir/src/collada_urdf.cpp.o
| /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/x86_64-linux/usr/bin/i586-oe-linux/i586-oe-linux-g++   -DASSIMP_EXPORT_API -DASSIMP_UNIFIED_HEADER_NAMES -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"collada_urdf\" -Dcollada_urdf_EXPORTS -I/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/collada-urdf/1.12.4-r0/robot_model-1.12.4/collada_urdf/include -I/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86/usr/include/assimp -I/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86/usr/include -I/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86/usr/include/collada-dom2.4/1.5 -I/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86/usr/include/collada-dom2.4 -I/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86/opt/ros/indigo/include  -m32 -march=i586  --sysroot=/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/collada-urdf/1.12.4-r0=/usr/src/debug/collada-urdf/1.12.4-r0 -fdebug-prefix-map=/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/x86_64-linux= -fdebug-prefix-map=/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86=  -fvisibility-inlines-hidden -fPIC   -std=c++11 -o CMakeFiles/collada_urdf.dir/src/collada_urdf.cpp.o -c /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/collada-urdf/1.12.4-r0/robot_model-1.12.4/collada_urdf/src/collada_urdf.cpp
| In file included from /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86/opt/ros/indigo/include/eigen_stl_containers/eigen_stl_containers.h:40:0,
|                  from /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86/opt/ros/indigo/include/geometric_shapes/mesh_operations.h:41,
|                  from /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/collada-urdf/1.12.4-r0/robot_model-1.12.4/collada_urdf/src/collada_urdf.cpp:89:
| /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/sysroots/qemux86/opt/ros/indigo/include/eigen_stl_containers/eigen_stl_vector_container.h:40:22: fatal error: Eigen/Core: No such file or directory
| ERROR: oe_runmake failed
| compilation terminated.
| CMakeFiles/collada_urdf.dir/build.make:65: recipe for target 'CMakeFiles/collada_urdf.dir/src/collada_urdf.cpp.o' failed
| make[2]: *** [CMakeFiles/collada_urdf.dir/src/collada_urdf.cpp.o] Error 1
| make[2]: Leaving directory '/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/collada-urdf/1.12.4-r0/build'
| CMakeFiles/Makefile2:144: recipe for target 'CMakeFiles/collada_urdf.dir/all' failed
| make[1]: *** [CMakeFiles/collada_urdf.dir/all] Error 2
| make[1]: *** Waiting for unfinished jobs....
...
| Makefile:130: recipe for target 'all' failed
| make: *** [all] Error 2
| WARNING: /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/collada-urdf/1.12.4-r0/temp/run.do_compile.7870:1 exit 1 from 'exit 1'
| ERROR: Function failed: do_compile (log file is located at /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/collada-urdf/1.12.4-r0/temp/log.do_compile.7870)
ERROR: Task (/home/lukas/dev/github.com/bmwcarit/meta-ros/recipes-ros/robot-model/collada-urdf_1.12.4.bb:do_compile) failed with exit code '1'
```

Build configuration:
BB_VERSION        = "1.31.1"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Fedora-23"
TARGET_SYS        = "i586-oe-linux"
MACHINE           = "qemux86"
DISTRO            = "nodistro"
DISTRO_VERSION    = "nodistro.0"
TUNE_FEATURES     = "m32 i586"
TARGET_FPU        = ""
meta              = "master:9b08503eabf78bc1b114416523b41dcce3449f58"
meta-oe           
meta-python       
meta-multimedia   = "master:fef7b1feb79c7b43623760e2cdf4e68bd0dbd526"
meta-ros          = "kinetic-branch"

bulwahn, 2016-09-05:
The open pull request https://github.com/ros/robot_model/pull/143 addresses this issue. Possibly this patch also resolves the issue here.

herb-kuta-lge, 2019-11-14:
`collada-urdf` successfully builds for **melodic** using `[thud-draft]`. @bulwahn < Can I close this issue?

closed 2019-12-03 by herb-kuta-lge