ros/meta-ros #411: console_bridge library not installed properly

vjayara1, 2016-09-16:
Hello all, I am trying to build meta-ros v2.0 on Ubuntu and am seeing the error,

 CMake Error at /home/venkat/internal/krogoth/build/tmp/sysroots/genericx86-64/usr/share/console_bridge/cmake/console_bridge-config.cmake:15 (message):
|   Library 'console_bridge' in package console_bridge is not installed
|   properly
| Call Stack (most recent call first):
|   CMakeLists.txt:3 (find_package)

Any idea what I am missing?

Build Configuration:
BB_VERSION        = "1.30.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "universal"
TARGET_SYS        = "x86_64-poky-linux"
MACHINE           = "genericx86-64"
DISTRO            = "poky"
DISTRO_VERSION    = "2.1.1"
TUNE_FEATURES     = "m64 core2"
TARGET_FPU        = ""
meta              
meta-poky         
meta-yocto-bsp    = "krogoth:aad7166704021d82ad3a5ec468552f8f10360d41"
meta-oe           = "krogoth:895b25b387cae428c38990f4d2747603ee9dc92d"
meta-intel-aero-librealsense = "krogoth:aaede82379ebcf1ae84716855bc2b4efc4a9d53b"
meta-intel        = "master:9a2d3d844c9650697846dc2bbfe5005af83e2af4"
meta-intel-aero-connectivity = "master:4d76a35605150f6b755076bd7fa9cdabc30ffa10"
meta-intel-aero   = "master:bc09e289508fa443929ebf3ae5975071f7b733bc"
meta-python       
meta-networking   = "krogoth:895b25b387cae428c38990f4d2747603ee9dc92d"
meta-ros          = "HEAD:fd9394c2871213ca3c151f7678a6c32cddffe736"

bulwahn, 2016-09-19:
I have never noticed this issue in the past. I will need to check if I can reproduce this issue with a build configuration that close to the build configuration that you use. This will need some time, so please do have a little patience. I will look into it. If you find something out yourself, just let me know here.

bulwahn, 2016-09-19:
Dea Vjayara1,

I tried bitbake console-bridge on this configuration and it succeeded:
```
$ bitbake console-bridge
Loaded 2505 entries from dependency cache.
Parsing of 1961 .bb files complete (1956 cached, 5 parsed). 2509 targets, 114 skipped, 0 masked, 0 errors.

Build Configuration:
BB_VERSION        = "1.30.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "universal"
TARGET_SYS        = "x86_64-poky-linux"
MACHINE           = "genericx86-64"
DISTRO            = "poky"
DISTRO_VERSION    = "2.1.1"
TUNE_FEATURES     = "m64 core2"
TARGET_FPU        = ""
meta              
meta-poky         
meta-yocto-bsp    = "krogoth:aad7166704021d82ad3a5ec468552f8f10360d41"
meta-oe           
meta-python       
meta-multimedia   = "krogoth:895b25b387cae428c38990f4d2747603ee9dc92d"
meta-ros          = "HEAD:fd9394c2871213ca3c151f7678a6c32cddffe736"

NOTE: Preparing RunQueue
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
NOTE: Tasks Summary: Attempted 723 tasks of which 723 didn't need to be rerun and all succeeded.
```

bulwahn, 2016-09-20:
I tested a build configuration very close to @vjayara1, but with the current commit on master (changeset 253d1fcd544caefa2058139a9421e1b8b57503d9) and did not have any build problems:  
```
$ bitbake core-image-ros-world -k
Parsing of 1977 .bb files complete (0 cached, 1977 parsed). 2527 targets, 114 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION        = "1.30.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Fedora-23"
TARGET_SYS        = "x86_64-poky-linux"
MACHINE           = "genericx86-64"
DISTRO            = "poky"
DISTRO_VERSION    = "2.1.1"
TUNE_FEATURES     = "m64 core2"
TARGET_FPU        = ""
meta              
meta-poky         
meta-yocto-bsp    = "krogoth:aad7166704021d82ad3a5ec468552f8f10360d41"
meta-oe           
meta-python       
meta-multimedia   = "krogoth:895b25b387cae428c38990f4d2747603ee9dc92d"
meta-ros          = "master:253d1fcd544caefa2058139a9421e1b8b57503d9"
(...)
NOTE: Tasks Summary: Attempted 7780 tasks of which 17 didn't need to be rerun and all succeeded.
```
I strongly propose to update to the current master and not use version 0.2 on yocto krogoth.

bulwahn, 2016-10-10:
If not further vetoed by @vjayara1, I consider this issue resolved.

closed 2016-10-10 by bulwahn