ros/meta-ros #413: do_fetch timeout for svn incubator project log4cxx

vjayara1, 2016-09-20:
Hello all,

I am trying to include log4cxx in my build and keep constantly running into the error below,

ERROR: log4cxx-svn-r0 do_fetch: Fetcher failure: Fetch command failed with exit code 1, output:
svn: E170013: Unable to connect to a repository at URL 'svn://svn.apache.org/repos/asf/incubator/log4cxx/trunk'
svn: E000110: Can't connect to host 'svn.apache.org': Connection timed out

I do have http-proxy-host and http-proxy-port set to 3128 defined in my servers file for .subversion. Any other ideas on what could be causing this timeout and any ways to work around this ? THANKS.

Build Configuration:
BB_VERSION        = "1.30.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "universal"
TARGET_SYS        = "x86_64-poky-linux"
MACHINE           = "genericx86-64"
DISTRO            = "poky"
DISTRO_VERSION    = "2.1.1"
TUNE_FEATURES     = "m64 core2"
TARGET_FPU        = ""
meta              
meta-poky         
meta-yocto-bsp    = "krogoth:aad7166704021d82ad3a5ec468552f8f10360d41"
meta-ros          = "HEAD:d47a27bc60a4cbdf8d91bfde8c8dec47cfed649a"
meta-intel-aero-librealsense = "krogoth:aaede82379ebcf1ae84716855bc2b4efc4a9d53b"
meta-intel        = "master:9a2d3d844c9650697846dc2bbfe5005af83e2af4"
meta-intel-aero-connectivity = "master:4d76a35605150f6b755076bd7fa9cdabc30ffa10"
meta-intel-aero   = "master:bc09e289508fa443929ebf3ae5975071f7b733bc"
meta-intel-private = "master:90846240887c23dfe3360bfe5374fbd52ceca8c9"
meta-oe           
meta-python       
meta-networking   = "krogoth:895b25b387cae428c38990f4d2747603ee9dc92d"

bulwahn, 2016-09-20:
Certainly, something is strange with the SVN fetcher or the apache's SVN repository.
I noticed some do_fetch failure of the at my very first experiments.
However then, I updated bitbake and the various openembedded repositories and I could not
reproduce the failure anymore (although I am uncertain if it is due to the update that the failure disappeared).
So currently, these two setups work on my machine:
With the latest changesets from the bitbake repository and openembedded repositories, it always reproducibly and repeatedly works:
```
[lukas@hulk build]$ bitbake log4cxx -c cleanall && bitbake log4cxx -c fetch
Build Configuration:
BB_VERSION        = "1.31.1"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Fedora-23"
TARGET_SYS        = "i586-oe-linux"
MACHINE           = "qemux86"
DISTRO            = "nodistro"
DISTRO_VERSION    = "nodistro.0"
TUNE_FEATURES     = "m32 i586"
TARGET_FPU        = ""
meta              = "master:5ac0604fdc7d5b783011c43d476210b427b5dae0"
meta-oe           
meta-python       
meta-multimedia   = "master:24e387aaf78bc0f68a46abb1fab3a76364804bd1"
meta-ros          = "master:253d1fcd544caefa2058139a9421e1b8b57503d9"
(...)
NOTE: Tasks Summary: Attempted 225 tasks of which 224 didn't need to be rerun and all succeeded.
```

And with yocto krogoth, it always reproducibly and repeatedly works:
```
$ bitbake log4cxx -c cleanall && bitbake log4cxx -c fetch
(...)
Build Configuration:
BB_VERSION        = "1.30.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "universal"
TARGET_SYS        = "x86_64-poky-linux"
MACHINE           = "genericx86-64"
DISTRO            = "poky"
DISTRO_VERSION    = "2.1.1"
TUNE_FEATURES     = "m64 core2"
TARGET_FPU        = ""
meta              
meta-poky         
meta-yocto-bsp    = "krogoth:aad7166704021d82ad3a5ec468552f8f10360d41"
meta-oe           
meta-python       
meta-multimedia   = "krogoth:895b25b387cae428c38990f4d2747603ee9dc92d"
meta-ros          = "master:253d1fcd544caefa2058139a9421e1b8b57503d9"
(...)
NOTE: Tasks Summary: Attempted 204 tasks of which 203 didn't need to be rerun and all succeeded.
```
However, with the current openembedded master branch

vjayara1, 2016-09-20:
Hello Bulwahn, Thanks for the effort in reproducing the problem. For some reason, even an "svn co .." in the build machine shell returns the same error. I guess there is something wrong with the proxy. But then, i downloaded the contents of the log4cxx from http://cgit.openembedded.org/openembedded/tree/recipes/log4cxx I was able to get past the error. It doesn't make sense why now, but this issue has now become less critical as i can move on with the build now. Thanks again. Will close this issue once i am able to make a complete build.

vjayara1, 2016-09-29:
Hello all, it was a proxy issue, after adding the following lines in ~/.subversion/servers the issue got fixed,
[global]
http-proxy-host = proxy-chain.intel.com
http-proxy-port = 911 

Thanks

closed 2016-09-29 by vjayara1