ros/meta-ros #416: [Errno 101] Network is unreachable -- Some packages may not be found!

vjayara1, 2016-09-29:
Hello,

When i try to build meta-ros with some ROS tools, i keep hitting the below error, any way to fix this or get around this ?

ERROR: python-rosdep-0.11.5-r0 do_compile: python setup.py build execution failed.
ERROR: python-rosdep-0.11.5-r0 do_compile: Function failed: do_compile 
Log data follows:
| DEBUG: Executing shell function do_compile
| Download error on https://pypi.python.org/simple/nose/: [Errno 101] Network is unreachable -- Some packages may not be found!
| Couldn't find index page for 'nose' (maybe misspelled?)
| Download error on https://pypi.python.org/simple/: [Errno 101] Network is unreachable -- Some packages may not be found!
| No local packages or download links found for nose>=1.0

Thanks

bulwahn, 2016-09-30:
Dear @vjayara1, I have never encountered that issue. In the python-rosdep recipe [1], it states that it depends on python-nose, so I assume python-nose is installed in the sysroot... While I was writing this, I wondered if python-nose is actually a native dependency, but this is not clear to me at the moment.

If I would have to guess the error, I would assume that your internet connection is has been disconnected or you are not allowed to make https connections from your build environment.

[1] https://github.com/bmwcarit/meta-ros/blob/master/recipes-devtools/python/python-rosdep_0.11.5.bb

vjayara1, 2016-10-03:
Hello Lukas,

What I am wondering is why the recipe file is attempting to “fetch” during do_compile. During do_compile, the proxy settings are not taking effect and those of us who are behind a corporate firewall will not be able to build. is there a way that the fetch can be avoided during the compile task and done instead during do_fetch ?

Thanks
Vjayara1

bulwahn, 2016-10-04:
That is a good question, and I was also expecting that the setup is made in such a way that it should not fetch further sources during the do_compile step. I do not know in detail why the recipe makes do_compile fetch that; it is defined in OpenEmbedded-Core's python-setuptools bbclass and python's setuptools package. This might be a good question to ask on the OpenEmbedded-Core mailing list.

bulwahn, 2016-10-10:
@vjayara1, could people involved in OpenEmbedded-Core help you further with this question?

xsteve, 2017-01-11:
I had a similar problem and solved it by providing http_proxy settings during the distutils3_do_compile step.

I added the following function to the .bb file of the recipe that does not build:

```bash
distutils3_do_compile_prepend() {
    export http_proxy=${http_proxy}
    export https_proxy=${https_proxy}
}
```

herb-kuta-lge, 2020-02-07:
@vjayara1 < Can this issue be closed?

closed 2020-02-21 by herb-kuta-lge