ros/meta-ros #425: fialed at "Built target pcl_tracking"

peter73421, 2016-10-14:
can you please help what issue might be?

```
| [ 59%] [ 58%] [ 60%] [ 58%] [ 60%] Building CXX object features/CMakeFiles/pcl_features.dir/src/spin_image.cpp.o
| Building CXX object features/CMakeFiles/pcl_features.dir/src/rift.cpp.o
| Building CXX object features/CMakeFiles/pcl_features.dir/src/statistical_multiscale_interest_region_extraction.cpp.o
| Building CXX object features/CMakeFiles/pcl_features.dir/src/principal_curvatures.cpp.o
| Building CXX object features/CMakeFiles/pcl_features.dir/src/rops_estimation.cpp.o
| cd /mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/build/features && /mnt/build-x11-full/tmp/sysroots/x86_64-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-g++   -DEIGEN_USE_NEW_STDVECTOR -DEIGEN_YES_I_KNOW_SPARSE_MODULE_IS_NOT_STABLE_YET -DPCLAPI_EXPORTS -Dqh_QHpointer -march=armv7-a -mfloat-abi=hard -mfpu=neon -mtune=cortex-a9  --sysroot=/mnt/build-x11-full/tmp/sysroots/imx6qpsabreauto  -O2 -pipe -g -feliminate-unused-debug-types -fvisibility-inlines-hidden -ffloat-store -pthread -fopenmp -O2 -g -DNDEBUG -fPIC -isystem /mnt/build-x11-full/tmp/sysroots/imx6qpsabreauto/usr/include -isystem /mnt/build-x11-full/tmp/sysroots/imx6qpsabreauto/usr/include/eigen3 -I/mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/recognition/include/pcl/recognition/3rdparty -I/mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/build/include -I/mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/common/include -I/mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/search/include -I/mnt/fsl-release-3.14.52-1.1.0_ga/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/kdtree/include -I/mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/octree/include -I/mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/filters/include -I/mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/features/include    -DBOOST_DISABLE_ASSERTS -DEIGEN_NO_DEBUG -o CMakeFiles/pcl_features.dir/src/rift.cpp.o -c /mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/features/src/rift.cpp
| cd /mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/build/features && /mnt/build-x11-full/tmp/sysroots/x86_64-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-g++   -DEIGEN_USE_NEW_STDVECTOR -DEIGEN_YES_I_KNOW_SPARSE_MODULE_IS_NOT_STABLE_YET -DPCLAPI_EXPORTS -Dqh_QHpointer -march=armv7-a -mfloat-abi=hard -mfpu=neon -mtune=cortex-a9  --sysroot=/mnt/build-x11-full/tmp/sysroots/imx6qpsabreauto  -O2 -pipe -g -feliminate-unused-debug-types -fvisibility-inlines-hidden -ffloat-store -pthread -fopenmp -O2 -g -DNDEBUG -fPIC -isystem /mnt/build-x11-full/tmp/sysroots/imx6qpsabreauto/usr/include -isystem /mnt/build-x11-full/tmp/sysroots/imx6qpsabreauto/usr/include/eigen3 -I/mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/recognition/include/pcl/recognition/3rdparty -I/mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/build/include -I/mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/common/include -I/mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/search/include -I/mnt/fsl-release-3.14.52-1.1.0_ga/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/kdtree/include -I/mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/octree/include -I/mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/filters/include -I/mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/features/include    -DBOOST_DISABLE_ASSERTS -DEIGEN_NO_DEBUG -o CMakeFiles/pcl_features.dir/src/rops_estimation.cpp.o -c /mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/features/src/rops_estimation.cpp
| cd /mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/build/features && /mnt/build-x11-full/tmp/sysroots/x86_64-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-g++   -DEIGEN_USE_NEW_STDVECTOR -DEIGEN_YES_I_KNOW_SPARSE_MODULE_IS_NOT_STABLE_YET -DPCLAPI_EXPORTS -Dqh_QHpointer -march=armv7-a -mfloat-abi=hard -mfpu=neon -mtune=cortex-a9  --sysroot=/mnt/build-x11-full/tmp/sysroots/imx6qpsabreauto  -O2 -pipe -g -feliminate-unused-debug-types -fvisibility-inlines-hidden -ffloat-store -pthread -fopenmp -O2 -g -DNDEBUG -fPIC -isystem /mnt/build-x11-full/tmp/sysroots/imx6qpsabreauto/usr/include -isystem /mnt/build-x11-full/tmp/sysroots/imx6qpsabreauto/usr/include/eigen3 -I/mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/recognition/include/pcl/recognition/3rdparty -I/mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/build/include -I/mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/common/include -I/mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/search/include -I/mnt/fsl-release-3.14.52-1.1.0_ga/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/kdtree/include -I/mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/octree/include -I/mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/filters/include -I/mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/features/include    -DBOOST_DISABLE_ASSERTS -DEIGEN_NO_DEBUG -o CMakeFiles/pcl_features.dir/src/spin_image.cpp.o -c /mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/features/src/spin_image.cpp
| cd /mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/build/features && /mnt/build-x11-full/tmp/sysroots/x86_64-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-g++   -DEIGEN_USE_NEW_STDVECTOR -DEIGEN_YES_I_KNOW_SPARSE_MODULE_IS_NOT_STABLE_YET -DPCLAPI_EXPORTS -Dqh_QHpointer -march=armv7-a -mfloat-abi=hard -mfpu=neon -mtune=cortex-a9  --sysroot=/mnt/build-x11-full/tmp/sysroots/imx6qpsabreauto  -O2 -pipe -g -feliminate-unused-debug-types -fvisibility-inlines-hidden -ffloat-store -pthread -fopenmp -O2 -g -DNDEBUG -fPIC -isystem /mnt/build-x11-full/tmp/sysroots/imx6qpsabreauto/usr/include -isystem /mnt/build-x11-full/tmp/sysroots/imx6qpsabreauto/usr/include/eigen3 -I/mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/recognition/include/pcl/recognition/3rdparty -I/mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/build/include -I/mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/common/include -I/mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/search/include -I/mnt/fsl-release-3.14.52-1.1.0_ga/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/kdtree/include -I/mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/octree/include -I/mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/filters/include -I/mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/features/include    -DBOOST_DISABLE_ASSERTS -DEIGEN_NO_DEBUG -o CMakeFiles/pcl_features.dir/src/statistical_multiscale_interest_region_extraction.cpp.o -c /mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/pcl-pcl-1.7.2/features/src/statistical_multiscale_interest_region_extraction.cpp
| /mnt/build-x11-full/tmp/sysroots/x86_64-linux/usr/bin/cmake -E cmake_progress_report /mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/build/CMakeFiles 21
| [ 61%] Building CXX object features/CMakeFiles/pcl_features.dir/src/vfh.cpp.o
[... cut ...]
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /mnt/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.7.2-r0/temp/log.do_compile.25800)
ERROR: Task 602 (/mnt/sources/meta-openembedded/meta-ros/recipes-extended/pcl/pcl_1.7.2.bb, do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 5764 tasks of which 5719 didn't need to be rerun and 1 failed.
No currently running tasks (5764 of 5773)

```

bulwahn, 2016-10-14:
I have not seen this issue before. If you look into one of the pending pull requests, you will find a commit that updates to pcl version 1.8.0, maybe the error does not occur anymore with the latest version 1.8.0.

peter73421, 2016-10-15:
thank you, please can you point me which "pending pull" has pcl 1.8.0?
Thank you.

bulwahn, 2016-10-15:
I think my co-maintainer has merged it just a few hours ago. If you checkout the current HEAD you will have pcl version 1.8.0.

peter73421, 2016-10-17:
build issue is still there even for pcl  1.8.0

| make[2]: Leaving directory
'/mnt/fsl-release-3.14.52-1.1.0_ga/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.8.0-r0/build'
|
/mnt/fsl-release-3.14.52-1.1.0_ga/build-x11-full/tmp/sysroots/x86_64-linux/usr/bin/cmake
-E cmake_progress_report
/mnt/fsl-release-3.14.52-1.1.0_ga/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.8.0-r0/build/CMakeFiles
 96 97
| [ 63%] Built target pcl_tracking
| make[2]: Leaving directory
'/mnt/fsl-release-3.14.52-1.1.0_ga/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.8.0-r0/build'
| CMakeFiles/Makefile2:1080: recipe for target
'features/CMakeFiles/pcl_features.dir/all' failed
| make[1]: *** [features/CMakeFiles/pcl_features.dir/all] Error 2
| make[1]: Leaving directory
'/mnt/fsl-release-3.14.52-1.1.0_ga/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.8.0-r0/build'
| Makefile:146: recipe for target 'all' failed
| make: *** [all] Error 2
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at
/mn//fsl-release-3.14.52-1.1.0_ga/build-x11-full/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/pcl/1.8.0-r0/temp/log.do_compile.25528)
ERROR: Task 629
(/mnt/fsl-release-3.14.52-1.1.0_ga/sources/meta-openembedded/meta-ros-master/recipes-extended/pcl/
pcl_1.8.0.bb, do_compile) failed with exit code '1'

bulwahn, 2016-10-17:
Could you please provide your Build Configuration for me to reproduce the issue?

peter73421, 2016-10-17:
```
repo init -u git://git.freescale.com/imx/fsl-arm-yocto-bsp.git -b imx-3.14.52-1.1.0_ga
repo sync
MACHINE=imx6qpsabreauto source fsl-setup-release.sh -b build-x11 e x11
bitbake fsl-image-multimedia-full
```

local.conf:

```

MACHINE ??= 'imx6qpsabreauto'
DISTRO ?= 'fsl-imx-x11'
PACKAGE_CLASSES ?= "package_rpm"
EXTRA_IMAGE_FEATURES = "debug-tweaks dev-pkgs ssh-server-openssh tools-debug tools-sdk"
USER_CLASSES ?= "buildstats image-mklibs image-prelink"
PATCHRESOLVE = "noop"
BB_DISKMON_DIRS = "\
    STOPTASKS,${TMPDIR},1G,100K \
    STOPTASKS,${DL_DIR},1G,100K \
    STOPTASKS,${SSTATE_DIR},1G,100K \
    ABORT,${TMPDIR},100M,1K \
    ABORT,${DL_DIR},100M,1K \
    ABORT,${SSTATE_DIR},100M,1K" 
PACKAGECONFIG_append_pn-qemu-native = " sdl"
PACKAGECONFIG_append_pn-nativesdk-qemu = " sdl"
ASSUME_PROVIDED += "libsdl-native"
CONF_VERSION = "1"

BB_NUMBER_THREADS = '22'
PARALLEL_MAKE = '-j 22'

DL_DIR ?= "${BSPDIR}/downloads/"
ACCEPT_FSL_EULA = ""

OPENCV_CONF_OPT += "DWITH_FFMPEG=ON"

CORE_IMAGE_EXTRA_INSTALL += "libopencv-core-dev libopencv-highgui-dev libopencv-imgproc-dev \ 
                             libopencv-objdetect-dev libopencv-ml-dev imx-gpu-viv imx-gpu-viv-dev \ 
                             libclc-mx6 libclc-mx6-dev libopencl-mx6 \ 
                             packagegroup-fsl-gstreamer gst-fsl-plugin gst-fsl-plugin gst1.0-fsl-plugin \
                             kernel-dev v4l-utils libv4l-dev i2c-tools"
LICENSE_FLAGS_WHITELIST="commercial"
IMAGE_INSTALL_append = " tmux imx-test python-core python-terminal python-pip python-modules python-numpy python-distribute git"
IMAGE_INSTALL_append = " cmake parted e2fsprogs e2fsprogs-resize2fs"
IMAGE_INSTALL_append = " yaml-cpp pcl"

```

jvgomez, 2016-10-17:
I get similar errors sometimes (internal compiler error) and they are because the PCL compilation takes all the resources from the computer, to a point compilation cannot continue. To solve it I increased swap file size like crazy (16GB)  and retry many times.

peter73421, 2016-10-17:
you mean "swap" size of build machine?

jvgomez, 2016-10-17:
Yes, swap partition size or create a swap file on the build machine

peter73421, 2016-10-19:
Fortunately as suggested I tried to compile for several times and last one build has been successful :)
I did not change swap of linux machine but regularly check with sudo swapon -s.
Thank you guys!

closed 2016-10-19 by peter73421