ros/meta-ros #430: Python packages and setup.* files missing when trying to build mavros with meta-ros

vjayara1, 2016-10-31:
Hello all,

I am trying to build mavros/mavlink along with meta-ros, but i see that in the final build, the setup.* files needed to set the environment variables for ROS are missing and PYTHONPATH is not set. Has anyone else seen this issue ? I have baselined meta-ros at tag v0,2. Are there any recent patches to address any python package issue or ROS integration issues? Thanks in advance.

bulwahn, 2016-10-31:
@vjayara1 We do not package these setup files on purpose. These files are completely filled with invalid paths due to the cross-compiling build environment. Instead you have to follow the instructions on the ROS wiki/in the README.md that tells you which bash environment variables you have set manually. I also suggest to move forward to the current master, as version 0.2 is quite old by now.

vjayara1, 2016-10-31:
Tghanks @bulwahn ... I will try manually setting the env up. For moving upto current "master" branch, is there a recommended known good point (commit) to pull from? It would be good to be able to recreate the same build if needed. Please advise.

bulwahn, 2016-10-31:
I propose to just follow the tip and rebase maybe once a week. meta-ros remains pretty stable and slightly improves over time.

bulwahn, 2017-05-01:
I believe this user question has been answered and hence I will now close this issue.

closed 2017-05-01 by bulwahn