ros/meta-ros #432: mavros_msgs not being installed on the platform

vjayara1, 2016-11-02:
when trying to build with mavros and mavros_msgs, only mavros seems to make it to the install. As a result, no messages are printed out on any topic. when trying to print from mavros like "rostopic echo mavros/state", it gives an error saying, 
"ERROR: Cannot load message class for [mavros_msgs/State]. Are your messages built?"
Any suggestions on what could be missing? Responses will be greatly appreciated. Thanks.

bulwahn, 2016-11-02:
Can you please share your extensions to meta-ros on github and provide a pointer to your meta-ros repository? I cannot reproduce the issue without knowing the recipes and the setup.

vjayara1, 2016-11-03:
This unfortunately is an internal repo which still hasnt been made public, but this build uses 17.4 meta-ros recipe files for mavros. I do change the /etc/hosts file and set up the environment like what is mentioned in the readme and able to start roscore successfully and list the mavros topics, but when trying to view the nodes like state for instance. it gives the above error message. What could be causing this. I cannot find the msg definitions anywhere in the build.

bulwahn, 2016-11-03:
As it seems mavros-msg's content, the .msg file and the generated python source, are not shipped. I believe rostopic requires the .msg file and the message's generated python source to work properly.

That is unfortunately all I can tell you without seeing the recipes and re-producing it. If the recipes are publicly available and you intend to integrate them back into meta-ros, we can continue to do a deeper inspection.

vjayara1, 2016-11-03:
Hi Lukas, basically the source base is meta-ros repo tag v0.2 and the patches from pull request #424. This has all the required mav recipes and the top level bb file has "IMAGE_INSTALL_append = " mavros mavros". Let me know if you need any other info. The mavros source is from tag 0.17.4 of the indigo-devel branch, Thank you.

JochiPochi, 2016-11-14:
Can you try building with mavros-msgs and mavros-msgs-dev please? There is a chance that the package-split may not be doing this correctly. All the tests I made were including -dev packages too so I could have missed this error.

Also, can you post the exact sequence of commands you use to cause the error? I'd like to recreate it if I can.

JochiPochi, 2016-11-14:
I found a careless error on my part, please try with the latest of #424.
Basically, the dependencies for Mavros build were being met but they weren't being installed.
The solution was to add `mavlink libmavconn mavros-msgs mavros-extras mavros` to your image. 
or
me adding that those are runtime dependencies to the recipes.

I fixed the recipes but just keep that in mind in case you may have other dependencies missing. mavros has a lot of build dependencies but I'm not 100% sure how many of those are also runtime dependencies.

vjayara1, 2016-11-14:
I can try with the -dev packages to see if there is any difference. Just clone the meta-ros and tag v0.2 with your patches on top and do a bitbake build. After loading the build, here are the things that i do on the platform, Create a setup.bash file and run it manually since that file is also not created in  the final build; export proper values for ROS_IP and ROS_MASTER_URI and modify the  /etc/hosts/ file on the platform and replace localhost with the hostname (as sepcified in the build). Once i do this i can start roscore successfully. I start the PX4 SITL on another host machine connected over SSH with IP addresses assigned; Run "rostopic list" on the platform and see that mavros nodes are created. After all this though, when attempting to run "rostopic echo <any mavros topic>" , i get the error, 
"Are the messages being built?" meaning it did not find the message definitions anywhere. Let  me know if you have any other questions/comments. 
Thanks.

vjayara1, 2016-11-15:
Hello John, Have you been able to compile the new version of the patch successfully with tag 17.4 ? I am getting a few recipe and compile errors. Let me know how it goes for you. Thanks.

bulwahn, 2016-11-15:
@vjayara1 I tried to follow your instructions, but I failed at the step `I start the PX4 SITL on another host machine connected over SSH with IP addresses assigned` because I would like to have a setup where I can test everything within the qemu image. I do not want setup and mess up yet another system just to do this failure diagnostics.
Can you describe the setup such that all commands can be done within a qemu emulator.
Then the instructions should be close to this:
```
Build image (link to bb recipe) with ...
Start qemu and then type:
$ export ...
$ export ...
$ roslaunch ... &
$ rostopic
... 
```

Even without executing, I asked myself if you did `export ROS_PACKAGE_PATH=/opt/ros/indigo/share`? This step is needed for rostopic to work.

JochiPochi, 2016-11-15:
Hello @vjayara1. I had compilation issues too, discovered I had pushed the wrong branch...

I just pushed the right modification so you should be able to compile it now. I tested all the recipes compile successfully but I wont be able to test if they still run properly until tomorrow. 

Can you give it a run in the mean time?

vjayara1, 2016-11-15:
Hello @JochiPochi, i addresses the license typos already. I was facing compile errors like "fatal error: mavlink/v1.0/mavlink_types.h: No such file or directory" on top of this. Were you able to compile successfully. libmavconn do_compile failed as above.

JochiPochi, 2016-11-15:
Odd, everything seems good on my end. Does the following:
`bitbake mavlink`
compile fine?

vjayara1, 2016-11-16:
mavlink comepiles fine. "libmavconn" fails with the following error. All errors are for missing file mavlink_types.h. Not sure if any of you have encountered this before.

| [ 50%] Building CXX object CMakeFiles/mavconn.dir/src/serial.cpp.o
| /home/venkat/Nov15/poky/build/tmp/sysroots/x86_64-linux/usr/bin/x86_64-poky-linux/x86_64-poky-linux-g++   -DMAVLINK2_COMPAT -DMAVLINK_DIALECT=ardupilotmega -Dmavconn_EXPORTS -I/home/venkat/Nov15/poky/build/tmp/work/core2-64-poky-linux/libmavconn/0.17.4-r0/git/libmavconn/include -I/home/venkat/Nov15/poky/build/tmp/sysroots/aer/usr/include -I/opt/ros/indigo/include  -m64 -march=core2 -mtune=core2 -msse3 -mfpmath=sse  --sysroot=/home/venkat/Nov15/poky/build/tmp/sysroots/aer  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/venkat/Nov15/poky/build/tmp/work/core2-64-poky-linux/libmavconn/0.17.4-r0=/usr/src/debug/libmavconn/0.17.4-r0 -fdebug-prefix-map=/home/venkat/Nov15/poky/build/tmp/sysroots/x86_64-linux= -fdebug-prefix-map=/home/venkat/Nov15/poky/build/tmp/sysroots/aer=  -fvisibility-inlines-hidden -std=c++11 -fPIC   -o CMakeFiles/mavconn.dir/src/serial.cpp.o -c /home/venkat/Nov15/poky/build/tmp/work/core2-64-poky-linux/libmavconn/0.17.4-r0/git/libmavconn/src/serial.cpp
| [ 66%] Building CXX object CMakeFiles/mavconn.dir/src/udp.cpp.o
| In file included from /home/venkat/Nov15/poky/build/tmp/work/core2-64-poky-linux/libmavconn/0.17.4-r0/git/libmavconn/src/mavlink_helpers.cpp:21:0:
| /home/venkat/Nov15/poky/build/tmp/work/core2-64-poky-linux/libmavconn/0.17.4-r0/git/libmavconn/include/mavconn/mavlink_dialect.h:32:40: fatal error: mavlink/v1.0/mavlink_types.h: No such file or directory
| compilation terminated.

JochiPochi, 2016-11-16:
Mmmmm...... I ran two different compilation environments and I can't seem to replicate that error:

a) meta-ros as described on the [readme instructions](https://github.com/bmwcarit/meta-ros/blob/master/README.md)

b) meta-intel-aero [environment](https://github.com/intel-aero/meta-intel-aero/wiki/Quickstart-Guide#intel-aero-development-environment-setup) without meta-uav

However, one thing I noticed is that the meta-intel-aero instructions add the meta-uav layer that has a different [mavlink recipe](https://github.com/intel-aero/meta-uav/tree/master/recipes-mavlink/mavlink). Maybe the installed files from that recipe are interfering with mine if you built it previously.

Can you do a clean build excluding the meta-uav layer? that may put us on a common ground

vjayara1, 2016-11-16:
same result... tried option b), compiling without meta-uav...which would also disable dronekit and pymavlink for now...

JochiPochi, 2016-11-16:
Can you post your bblayers.config and local.config please? 

Also, can you go to `tmp/work/core2-64-poky-linux/mavlink` and check that things get created properly by the recipe?

you should see a folder named `2016.5.20-r0` as that is the PN of the recipe.

I have attached the directory structure I get there for you to compare, if you post yours I can take a look as well. I generated it using `tree >> tree.txt`

[tree.txt](https://github.com/bmwcarit/meta-ros/files/596170/tree.txt)

vjayara1, 2016-11-17:
Here is my dir tree 
[tree.txt](https://github.com/bmwcarit/meta-ros/files/596377/tree.txt)

vjayara1, 2016-11-17:
Please use the following tree 
[tree.txt](https://github.com/bmwcarit/meta-ros/files/596463/tree.txt)

bulwahn, 2016-11-17:
@vjayara1: I happy to assist by independently confirming the issue, however currrently your setup is not sufficiently described to do so.

JochiPochi, 2016-11-17:
@bulwahn, are you able to compile the libmavconn recipe on any setup you have?
I didn't quite get if you were successful or not at it by your comment, just making sure.

bulwahn, 2016-11-22:
I could compile the pull-request #424 with the changes I made available in my personal branch. I started the image in qemu but I did not try running anything because I did not what I should expect.

vjayara1, 2016-11-22:
Thank you Lukas. I tried out with PR #434 and now i see that mavros-msgs are being installed on the target. Closing this issue...

closed 2016-11-22 by vjayara1