ros/meta-ros #433: rosconsole recipe do_compile error

JochiPochi, 2016-11-17:
As of [dc0748974564b541710e187902b62b05d43e20ff](https://github.com/bmwcarit/meta-ros/commit/dc0748974564b541710e187902b62b05d43e20ff) rosconsole_1.11.20 fails to build while doing 
`bitbake core-image-ros-roscore`

This was a fresh build environement made following the instructions on the README.md

Logfile with the error attached.

[log.do_compile.14287.txt](https://github.com/bmwcarit/meta-ros/files/596506/log.do_compile.14287.txt)

JochiPochi, 2016-11-17:
rosconsole seems to work just fine when building with Poky though. Maybe something changed on upstream openembedded?

bulwahn, 2016-11-17:
I can confirm that rosconsole fails. We will need to do some `git bisect`; I did not fail a few weeks ago.

JochiPochi, 2016-11-17:
@bulwahn would you be interested in using a repo [manifests file ](https://github.com/Freescale/fsl-community-bsp-platform/blob/krogoth/default.xml) to avoid this problem in the future?

I first learned bitbake while working with the Freescale community projects so I got used to having those. I can craft up one really quickly for this project if you think it would add some value :)

bulwahn, 2016-11-18:
@JochiPochi Sure, we can do so, just create a pull request to meta-ros. It is certainly helpful to clearly state which build configuration we tested and know that it works. It does not really help to avoid that upstream development might require changes, though. We still have to follow the upstream development, just to continuously repair minor issues as the other layers develop, but for that I am also trying to find time to get a sufficiently stable and public CI system running.

commit 2cd378a1 (2016-11-30): rosconsole: add include in console.h (resolves #433)

Compiling rosconsole failed with:
```
[...]/ros_comm-1.11.20/tools/rosconsole/include/ros/console.h:121:14: error: 'vector' in namespace 'std' does not name a template type
 typedef std::vector<TokenPtr> V_Token;
```

The console.h assumed that vector is included already by one of its
dependencies. This bold assumption has been uncovered by the update
of the boost library to version 1.62.0 [1, 2] in openembedded-core
repository.

Coincidently, this issue was also noticed by ROS users on Gentoo and
Arch Linux, which probably also use the lat [... cut]
commit 78101ac3 (2016-11-30): Merge pull request #436 from bulwahn/resolve-433

rosconsole: add include in console.h (resolves #433)
closed 2016-11-30 by KristofRobot