ros/meta-ros #435: Cartographer

Minipada, 2016-11-28:
I was wondering if there was any proper way to include [Cartographer](https://github.com/googlecartographer/cartographer/) in the meta-ros layer.

I have made some tests home and it's close to compile for my machine.
The thing is it depends on [Ceres-solver](https://ceres-solver.googlesource.com/ceres-solver) depending itself on atlas and other dependencies (ATLAS, BLAS, LAPACK and Suitesparse shared and static libraries).

ATLAS is not supposed to be cross compiled as the [documentation](http://www.vesperix.com/arm/atlas-arm/faq/index.html) mentions it. I built a system, compiled all the dependencies to Ceres-Solver which cannot be cross-compiled on it, created recipes adding the files manually. It was a big effort and at the end,  I'll have to recompile for optimization purpose or for new machines.

I was wondering if you had any knowledge regarding this topic. Cartographer is pretty interesting and got more than 2k stars on their repo pretty fast. Any idea how to proceed in this case? Should I send this question to the Yocto guys?

bulwahn, 2016-11-28:
I am happy to include carthographer in meta-ros. How to handle ATLAS properly is probably a good question for the yocto mailing list. I assume that one has to cross-compile ATLAS with a reasonable conservative setting, and then actually create specific settings for actual machines on which the ATLAS tests have already been executed.

If you have a first draft that is a good basis to review and continue to work on open points, please share it here either via pull request or via link to a repository.

Minipada, 2016-11-28:
Great, I will do a pull request as soon as I get it working on my machine

herb-kuta-lge, 2019-11-14:
`cartographer` successfully builds for **melodic** using `[thud-draft]`. @bulwahn < Can I close this issue?

closed 2019-12-03 by herb-kuta-lge