ros/meta-ros #448: rocon-ebnf: No generic license file exists for: GPL in any provider

bulwahn, 2017-01-19:
The rocon_ebnf package is published under GNU Public License, but the sources do not state which GPL version they refer to.

shr-project, 2018-04-10:
There are also some files under BSD license there:
tmp-glibc/work/core2-64-oe-linux/rocon-ebnf/0.1.23-r0/rocon_tools-0.1.23/rocon_ebnf/tests/test_rule_parser.py:# License: BSD
tmp-glibc/work/core2-64-oe-linux/rocon-ebnf/0.1.23-r0/rocon_tools-0.1.23/rocon_ebnf/tests/test_rule_parser.py:#   https://raw.github.com/robotics-in-concert/rocon_tools/license/LICENSE
tmp-glibc/work/core2-64-oe-linux/rocon-ebnf/0.1.23-r0/rocon_tools-0.1.23/rocon_ebnf/tests/demo_substring_problem.py:# License: BSD
tmp-glibc/work/core2-64-oe-linux/rocon-ebnf/0.1.23-r0/rocon_tools-0.1.23/rocon_ebnf/tests/demo_substring_problem.py:#   https://raw.github.com/robotics-in-concert/rocon_tools/license/LICENSE
tmp-glibc/work/core2-64-oe-linux/rocon-ebnf/0.1.23-r0/rocon_tools-0.1.23/rocon_ebnf/src/rocon_ebnf/rule_parser.py:#copyright 2008 under the GNU General Public License
tmp-glibc/work/core2-64-oe-linux/rocon-ebnf/0.1.23-r0/rocon_tools-0.1.23/rocon_ebnf/src/rocon_ebnf/__init__.py:# License: BSD
tmp-glibc/work/core2-64-oe-linux/rocon-ebnf/0.1.23-r0/rocon_tools-0.1.23/rocon_ebnf/src/rocon_ebnf/__init__.py:#   https://raw.github.com/robotics-in-concert/rocon_tools/license/LICENSE
tmp-glibc/work/core2-64-oe-linux/rocon-ebnf/0.1.23-r0/rocon_tools-0.1.23/rocon_ebnf/doc/manifest.yaml:license: GPL
tmp-glibc/work/core2-64-oe-linux/rocon-ebnf/0.1.23-r0/rocon_tools-0.1.23/rocon_ebnf/package.xml:  <license>GPL</license>

herb-kuta-lge, 2019-11-14:
`rocon-ebnf` successfully builds for **melodic** using `[thud-draft]`. @bulwahn < Can I close this issue?

closed 2019-12-03 by herb-kuta-lge