ros/meta-ros #450: do_rootfs: Cannot satisfy dependencies while bitbake core-image-ros-roscore on Ubuntu15.10

den4ix, 2017-01-28:
While building core-image-ros-roscore image on Ubuntu 15.10 with arm toolchain I experience the following issue:
[log.do_rootfs.9347.txt](https://github.com/bmwcarit/meta-ros/files/736629/log.do_rootfs.9347.txt)
Also on Gentoo machine using the same toolchain build finish successfully.

bulwahn, 2017-01-28:
This issue is unknown to us and I have not seen any similar issue in the past. It is still unclear to me if this is caused by something in meta-ros, or if it is due to bitbake or the system environment. Please let us know if you find out more about this issue.

bulwahn, 2017-04-18:
As we did not get any further feedback, the issue is closed now.

closed 2017-04-18 by bulwahn