ros/meta-ros #456: Compile error "undefined reference to"

fuowang, 2017-02-26:
Arch Linux 64bit cross-compiles for MACHINE ??= "qemux86"
gcc (GCC) 6.3.1 20170109
Python 2.7.13 (default, Dec 21 2016, 07:16:46)     [GCC 6.2.1 20160830] on linux2

Error Log:
	/home/fuowang/devel/build/tmp-glibc/work/i586-oe-linux/rosout/1.11.20-r0/recipe-sysroot/lib/libdl.so.2: undefined reference to `_dl_catch_error@GLIBC_PRIVATE'
	/home/fuowang/devel/build/tmp-glibc/work/i586-oe-linux/rosout/1.11.20-r0/recipe-sysroot/lib/libdl.so.2: undefined reference to `_dl_signal_error@GLIBC_PRIVATE'
	/home/fuowang/devel/build/tmp-glibc/work/i586-oe-linux/rosout/1.11.20-r0/recipe-sysroot/lib/ERROR: oe_runmake failed
	libcrypt.so.1: undefined reference to `__explicit_bzero_chk@GLIBC_2.25'
	/home/fuowang/devel/build/tmp-glibc/work/i586-oe-linux/rosout/1.11.20-r0/recipe-sysroot/lib/libcrypt.so.1: undefined reference to `__snprintf@GLIBC_PRIVATE'
	collect2: error: ld returned 1 exit status

I'm followed this paper: http://wiki.ros.org/hydro/Installation/OpenEmbedded

How can I do for this?

rojkov, 2017-02-27:
Looks like either glibc is missing in rosout's recipe-sysroot or the linker is not instructed to look for glibc in the correct directory. A complete compile log would help to give more insights on what's wrong.

fuowang, 2017-02-28:
Thanks, it's my fault.
I compile glibc-2.25 on a 32bit virtual machine,and replace /usr/lib32/libc.so.6 and /usr/lib32/libpthread.so.0 to new version(old version is 2.24),resolve it eventually.

closed 2017-02-28 by fuowang