ros/meta-ros #458: python-nose conflicts with python3-nose

yongli3, 2017-03-07:
During the ros-world build, there is a conflict about python-nose and python3-nose. Since opencv dpeneds on python3-nose, by tf and rostest depend on python-nose. bitbake will install both of python3-nose and python-nose, this causes the ""that file is already provided by package" error message.

I think we need to change the python-nose to python3-nose in the DEPENDS in meta-ros

bulwahn, 2017-03-07:
Thanks for reporting this issue. I am happy to review pull requests that addresses this issue. We should make sure that all ROS packages still work with python3, which sounds like a quite large task to me right now, but we will see once we start tackling this issue.

yongli3, 2017-03-07:
Thanks @bulwahn, please check on PR #459

yongli3, 2017-03-09:
As a workaround, we can modify the local.conf to disable python3 in opencv:
PACKAGECONFIG_remove_pn-opencv = " python3"
PACKAGECONFIG_append_pn-opencv = " python"

rojkov, 2017-03-30:
I guess I'm volunteering to check if python3 is enough for ROS because the project I'm working on [1] wants to get rid of python2 completely.

[1] https://github.com/intel/intel-iot-refkit/

bulwahn, 2017-03-30:
Well that sounds like a bigger task...

closed 2017-04-13 by bulwahn

bulwahn, 2017-04-14:
@yongli3 We have now included python3 support. Could you try if the issue still occurs in your image build when you set meta-ros to use python3?

commit 3e201940 (2017-04-13): enable python3 compatible builds of ROS (resolves #458)

By default meta-ros uses python2. In case you want to get
python3-compatible builds add the following line to your local.conf:

    ROS_USE_PYTHON3 = "yes"