ros/meta-ros #460: boost: type error when including libs/python/src/numpy/dtype.cpp while compiling

bulwahn, 2017-03-13:
With the update from boost 1.62 to 1.63 (openembedded-core commit ef603f41b5df4772bb598ec9d389dd5f858592afef603f41b5df4772bb598ec9d389dd5f858592af@openembedded/openembedded-core) [1],  `bitbake boost` fails during do_compile.

It fails with meta-ros probably due to appending PACKAGECONFIG with python in the boost_%.bbappend file [2].

[1] http://cgit.openembedded.org/openembedded-core/commit/?id=ef603f41b5df4772bb598ec9d389dd5f858592af
[2] https://github.com/bmwcarit/meta-ros/blob/master/recipes-support/boost/boost_%25.bbappend

The tested build configuration is:
```
Build Configuration:
BB_VERSION        = "1.33.2"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "fedora-23"
TARGET_SYS        = "i586-oe-linux"
MACHINE           = "qemux86"
DISTRO            = "nodistro"
DISTRO_VERSION    = "nodistro.0"
TUNE_FEATURES     = "m32 i586"
TARGET_FPU        = ""
meta              = "HEAD:ef603f41b5df4772bb598ec9d389dd5f858592af"
meta-oe           
meta-python       
meta-multimedia   = "master:1dff2351aa6cdafa5a501e8956cb853ab17ed9ae"
meta-qt4          = "master:2c7f8df9039be498f8a2232d1428adb7f4e5e800"
meta-ros          = "rojkov-drop-librealsense-v1:fe4f0c1c2b3608306c981e61ca60674b46763911"
```

I did not do any further failure analysis yet.

rojkov, 2017-03-15:
Uninstalling python2-numpy from the host system helps to work around this failure.

bulwahn, 2017-03-15:
Issue is also tracked at https://bugzilla.yoctoproject.org/show_bug.cgi?id=11152

bulwahn, 2017-03-25:
@BilalArain This issue has only appeared very recently in the openembedded-core development, and is currently in the progress of being addressed and resolved. You can avoid this failure until it is resolved in the master branch by choosing the parent commit ac5c08ea [1]  in the openembedded-core repository, or by reverting ef603f41 [2] in your openembedded-core repository for the time being.

[1] http://cgit.openembedded.org/openembedded-core/commit/?id=ac5c08ea00816852a712a8d7025bce7e865460ec
[2] http://cgit.openembedded.org/openembedded-core/commit/?id=ef603f41b5df4772bb598ec9d389dd5f858592af

bulwahn, 2017-04-13:
The commit http://cgit.openembedded.org/openembedded-core/commit/?id=0f5418eb0ce12811b16d2e3c28c28140a509f685 in openembedded-core addresses the issue; however, it now requires to use python3 as the boost-python package for python2 is not provided anymore.

herb-kuta-lge, 2019-11-14:
`boost` successfully builds for **melodic** using `[thud-draft]`. @bulwahn < Can I close this issue?

closed 2019-12-03 by herb-kuta-lge