ros/meta-ros #461: Yocto build of rosout fails when ROS is installed locally on host

fabianoboril, 2017-03-14:
Dear all,

I am currently building a yocto image with ROS included. In the process of setting up this image, I came across the issue, that the build of rosout fails on my machine, while it builds flawlessly on another machine. The only difference between these two machines is, that my machine has a local ROS installation.

I figured out, that the build of rosout, but also other packages, access the libraries of the local host ROS installation, which inevitably leads to build failures. It seems that this behavior is rooted in the way catkin generates cmake config files. As it turns out, these cmake config files always points to the short path /opt/ros/<version>/lib in first place to look for libraries. If the path does not exist, the yocto-based directories are checked. However, if a local ROS installation is available, the short path creates a hit and the host libraries are accessed during the build process.

To overcome this issue, I modified the catkin_package.cmake file by removing the lines with 
`list(INSERT PKG_CONFIG_LIB_PATHS 0 ${PROJECT_SPACE_DIR}/lib)`
I did this by adding an extra bbappend for the catkin package. 

I don't know if this is the preferred way to resolve this issue, but at least it works. So maybe this is useful for some that face similar issues. 

Best regards,
Fabian Oboril

bulwahn, 2017-03-14:
Could you report which build configuration you are using? The problem you mentioned has been addressed already a few times by a number of authors; in each iteration, I tried to improve the situation of avoiding to link against host libraries.
If you are using the master branch of meta-ros and openembedded-core, you should open a pull request so that we can review, test and then potentially merge your suggestion into the master branch.

fabianoboril, 2017-03-15:
Hi,

in fact, I was not using the newest commit, since I started with the project a while ago. After your answer yesterday, I updated to the most recent version, which seems to fix the issue. Thank you very much for pointing this out!

Best regards.

bulwahn, 2017-03-15:
Nice to see that we solved the issue you encountered already; this gives us a good feeling that we have a grip on the number of not yet discovered issues. As this issue seems to be solved now, I will close it. Just continue to create issues when you find another problem in your development.

closed 2017-03-15 by bulwahn