ros/meta-ros #463: Add more ROS packages

yongli3, 2017-03-18:
@bulwahn  Please check on https://github.com/yongli3/meta-ros/commit/eba9ddd281424a5b8dea6017a1a727cf54593e09, based on issue #245, I add more ROS related packages into recipes-extended, to work with turtlebot device. I would like to know if we can add them into meta-ros ? If YES, I will submit a PR

bulwahn, 2017-03-18:
@yongli3 I have similar work on-going. Before you spend significant time on your recipes, you should have a look at my recipes, as I believe I am already a step further (I started in December working on them). I will provide you a link this afternoon (European time). Is this okay for you?

yongli3, 2017-03-18:
@bulwahn Thanks for your information. It will be very useful with your recipes, I can help to test them :)

bulwahn, 2017-03-19:
@yongli3 Please have a look at #465. You would help us a lot if you could test them.

bulwahn, 2017-03-19:
There is a further branch wip-turtlebot on my personal fork to get turtlebot running. This is still some work in progress. Turtlebot requires openni2-camera, which is still quite some work.

yongli3, 2017-03-19:
Got it! @bulwahn  I will test the build image on hardware and update the status

dkargin, 2019-04-03:
There is a fork of [openni2](https://github.com/aldebaran/openni2)  which uses CMake instead of handcrafted gnumake magic. We use it as a base for building openni2 in yocto. Our fork is at https://github.com/sybotech/openni2. There were some changes to make this fork be able to read data from orbbec sensors as well. It works mostly fine on aarch64 machines, like Rpi3 or odroid-c2

herb-kuta-lge, 2020-02-07:
@yongli3 < Can this issue be closed?

closed 2020-02-21 by herb-kuta-lge