ros/meta-ros #464: libgphoto2: configure fails with `cp: cannot stat ‘/.../usr/share/gettext/po/Makefile.in.in’: No such file or directory

bulwahn, 2017-03-18:
libgphoto2 is located in meta-oe, and is required in meta-ros through this dependency `libgphoto2 -> opencv -> cv-bridge`.

The build configuration is:
```
Build Configuration:
BB_VERSION        = "1.33.2"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "fedora-23"
TARGET_SYS        = "i586-oe-linux"
MACHINE           = "qemux86"
DISTRO            = "nodistro"
DISTRO_VERSION    = "nodistro.0"
TUNE_FEATURES     = "m32 i586"
TARGET_FPU        = ""
meta              = "HEAD:ac5c08ea00816852a712a8d7025bce7e865460ec"
meta-oe           
meta-python       
meta-multimedia   = "master:1dff2351aa6cdafa5a501e8956cb853ab17ed9ae"
meta-qt4          = "master:2c7f8df9039be498f8a2232d1428adb7f4e5e800"
meta-intel-realsense = "master:1d50cb5b82926f8e69a78999e9e8d2ffe2169897"
meta-ros          = "master:8a3adc1a6542f2c8a40b36021a78dc1057e3ed08"
```

The error log is:
```
| Configuration (libgphoto2_port 0.12.0):
| 
|   Build
|     Source directory:              ../../libgphoto2-2.5.8/libgphoto2_port
|     Compiler:                      i586-oe-linux-gcc  -m32 -march=i586 --sysroot=/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/libgphoto2/2.5.8-r0/recipe-sysroot
|     libltdl includes:
|     libltdl library:               -lltdl
|     Use translations:              no
| 
|   Serial ports
|     Serial support:                no
| 
|   USB ports
|     libusb-1.0:                    yes
|     libusb0 support:               detected, but disabled (libusb1 support is enabled)
|     libusb:                        yes
| 
|   Documentation
|     build API docs with gtk-doc:   no (not requested)
| 
| You may run "make" and "make install" now.
| 
| Configuration (libgphoto2 2.5.8):
| 
|   Build
|     Source code location:          ../libgphoto2-2.5.8
|     GP2DDB support:                no  (EXPERIMENTAL)
|     Compiler:                      i586-oe-linux-gcc  -m32 -march=i586 --sysroot=/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/libgphoto2/2.5.8-r0/recipe-sysroot
|     libltdl includes:
|     libltdl library:               -lltdl
| 
|   Features
|     Use translations:              no
|     JPEG mangling support:         yes
|     libxml-2.0:                    yes
|     XML support for Olympus E s..: yes
|     gdlib:                         no
|     libGD conversion support:      no
|     Camlibs:                       all
|     build API docs with gtk-doc:   no (not requested)
|     libexif:                       yes
| 
|     Subprojects:                   libgphoto2_port
| 
| You may run "make" and "make install" now.
| Please check whether the configuration I detected matches what you
| would like to have. E.g. make sure that USB support is there if you
| intend to use USB cameras with libgphoto2 photo camera library.
| 
| Please also check that PKG_CONFIG_PATH contains
|     ${libdir}/pkgconfig
| before compiling any libgphoto2 frontend.
| 
| cp: cannot stat ‘/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/libgphoto2/2.5.8-r0/recipe-sysroot/usr/share/gettext/po/Makefile.in.in’: No such file or directory
| WARNING: /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/libgphoto2/2.5.8-r0/temp/run.do_configure.9287:1 exit 1 from 'cp /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/libgphoto2/2.5.8-r0/recipe-sysroot/usr/share/gettext/po/Makefile.in.in /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/libgphoto2/2.5.8-r0/libgphoto2-2.5.8/libgphoto2_port/po/'
| ERROR: Function failed: do_configure (log file is located at /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/libgphoto2/2.5.8-r0/temp/log.do_configure.9287)
ERROR: Task (/home/lukas/dev/openembedded.org/meta-openembedded/master/meta-oe/recipes-graphics/gphoto2/libgphoto2_2.5.8.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2430 tasks of which 2429 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/lukas/dev/openembedded.org/meta-openembedded/master/meta-oe/recipes-graphics/gphoto2/libgphoto2_2.5.8.bb:do_configure
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```

bulwahn, 2017-03-18:
When building recipes in meta-ros, the libgphoto2 issue can be avoided by changing the PACKAGECONFIG of opencv to not depend on libgphoto2.

There are two simple ways to do that:

1. Add a `opencv_%.bbappend` file with:
```
PACKAGECONFIG_remove ?= "gphoto2"
```

2. Add `PACKAGECONFIG_remove_pn-opencv = "gphoto2"` to your local.conf.

herb-kuta-lge, 2019-11-14:
`libgphoto2` successfully builds for **melodic** using `[thud-draft]`. @bulwahn < Can I close this issue?

closed 2019-12-03 by herb-kuta-lge