ros/meta-ros #467: mavros: do_compile fails including eigen headers

bulwahn, 2017-03-19:
`bitbake mavros` fails with:
```
| /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/mavros/0.17.4-r0/mavros-0.17.4/mavros/include/mavros/mavros_uas.h:23:23: fatal error: Eigen/Eigen: No such file or directory
|  #include <Eigen/Eigen>
|                        ^
| compilation terminated.
```

The build configuration is:
```
Build Configuration:
BB_VERSION        = "1.33.2"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "fedora-23"
TARGET_SYS        = "i586-oe-linux"
MACHINE           = "qemux86"
DISTRO            = "nodistro"
DISTRO_VERSION    = "nodistro.0"
TUNE_FEATURES     = "m32 i586"
TARGET_FPU        = ""
meta              = "HEAD:ac5c08ea00816852a712a8d7025bce7e865460ec"
meta-oe           
meta-python       
meta-multimedia   = "master:1dff2351aa6cdafa5a501e8956cb853ab17ed9ae"
meta-qt4          = "master:2c7f8df9039be498f8a2232d1428adb7f4e5e800"
meta-intel-realsense = "master:1d50cb5b82926f8e69a78999e9e8d2ffe2169897"
meta-ros          = "repair-mavros:898028f49bf0d890d37cfef8f504f31dc0d17487"
```

bulwahn, 2017-03-19:
@rojkov Did you come across similar issues when addressing the adjustments for recipe-specific sysroots? mavros did work in January 2017, and I could not see an obvious mistake in the CMakeLists.txt.

rojkov, 2017-03-22:
Yes, I bumped into exactly this problem. For some reason the fix still doesn't work for mavros. I could reproduce the build failure locally.

commit 47568013 (2017-05-08): mavros: properly find Eigen (resolves #467)
closed 2017-05-08 by rojkov