ros/meta-ros #469: python-orocos-kdl:

bulwahn, 2017-03-20:
On the current pull request #465, `bitbake python-orocos-kdl` fails with:
```
| /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/python-orocos-kdl/1.3.0-r0/recipe-sysroot-native/usr/bin/i586-oe-linux/i586-oe-linux-g++   -Dpython_module_PyKDL_EXPORTS -I/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/orocos-kdl/1.3.0-r0/recipe-sysroot/usr/include/eigen3 -I/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/python-orocos-kdl/1.3.0-r0/recipe-sysroot/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/x86_64-linux/sip-native/4.19-r0/image/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/python-orocos-kdl/1.3.0-r0/recipe-sysroot-native/usr/include -I/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/python-orocos-kdl/1.3.0-r0/recipe-sysroot/usr/include/python2.7  -m32 -march=i586  --sysroot=/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/python-orocos-kdl/1.3.0-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/python-orocos-kdl/1.3.0-r0=/usr/src/debug/python-orocos-kdl/1.3.0-r0 -fdebug-prefix-map=/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/python-orocos-kdl/1.3.0-r0/recipe-sysroot-native= -fdebug-prefix-map=/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/python-orocos-kdl/1.3.0-r0/recipe-sysroot=  -fvisibility-inlines-hidden  -m32 -march=i586  --sysroot=/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/python-orocos-kdl/1.3.0-r0/recipe-sysroot -fPIC   -o CMakeFiles/python_module_PyKDL.dir/PyKDL/sipPyKDLpart0.o -c /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/python-orocos-kdl/1.3.0-r0/build/PyKDL/sipPyKDLpart0.cpp
...
| In file included from /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/python-orocos-kdl/1.3.0-r0/recipe-sysroot/usr/include/kdl/jntarray.hpp:26:0,
|                  from /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/python-orocos-kdl/1.3.0-r0/orocos_kinematics_dynamics-1.3.0/python_orocos_kdl/PyKDL/kinfam.sip:172:
| /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/python-orocos-kdl/1.3.0-r0/recipe-sysroot/usr/include/kdl/jacobian.hpp:26:22: fatal error: Eigen/Core: No such file or directory
|  #include <Eigen/Core>
|                       ^
| compilation terminated.
| CMakeFiles/python_module_PyKDL.dir/build.make:92: recipe for target 'CMakeFiles/python_module_PyKDL.dir/PyKDL/sipPyKDLpart0.o' failed
| make[2]: *** [CMakeFiles/python_module_PyKDL.dir/PyKDL/sipPyKDLpart0.o] Error 1
```

The build configuration is:
```
Build Configuration:
BB_VERSION        = "1.33.2"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "fedora-23"
TARGET_SYS        = "i586-oe-linux"
MACHINE           = "qemux86"
DISTRO            = "nodistro"
DISTRO_VERSION    = "nodistro.0"
TUNE_FEATURES     = "m32 i586"
TARGET_FPU        = ""
meta              = "HEAD:ac5c08ea00816852a712a8d7025bce7e865460ec"
meta-oe           
meta-python       
meta-multimedia   = "master:1dff2351aa6cdafa5a501e8956cb853ab17ed9ae"
meta-qt4          = "master:2c7f8df9039be498f8a2232d1428adb7f4e5e800"
meta-intel-realsense = "master:1d50cb5b82926f8e69a78999e9e8d2ffe2169897"
meta-ros          = "kobuki-v1:c15bf07f941927d977220c46f41cdb9f4bcb0e6d"
```

bulwahn, 2017-03-20:
The python-orocos-kdl recipe did work in January 2017.
@rojkov Did you encounter a similar issue when updating to recipe-specific sysroots?

rojkov, 2017-03-20:
Yes, I did. But those errors were due to catkin's peculiarities. This package seems to be purely CMake-based.

Hopefully I'll have time to look at this tomorrow or on Wed.

herb-kuta-lge, 2019-11-14:
`python-orocos-kdl` successfully builds for **melodic** using `[thud-draft]`. @bulwahn < Can I close this issue?

closed 2019-12-03 by herb-kuta-lge