ros/meta-ros #470: log4cxx svn fetch

BilalArain, 2017-03-23:
Previous issues: #390, resolved on 3 Jun 2016 commit  b4baa4a

Failed to fetch log4cxx provided in recipe.
Warning:
log4cxx-svn-r0 do_fetch: Failed to fetch URL svn://svn.apache.org/repos/asf;module=incubator/log4cxx/trunk;protocol=http, attempting MIRRORS if available

Error:
 Fetcher failure for URL: 'svn://svn.apache.org/repos/asf;module=incubator/log4cxx/trunk;protocol=http'. Unable to fetch URL from any source.

Can be reproduced by using: bitbake log4cxx -c cleanall && bitbake log4cxx

rojkov, 2017-03-23:
@BilalArain I presume you're behind proxy. Then you need to configure your local subversion client. Bear in mind subversion doesn't respect `http_proxy=` and `https_proxy=` environment variables.

bulwahn, 2017-03-23:
Your issue looks like a duplicate of https://github.com/bmwcarit/meta-ros/issues/413.

BilalArain, 2017-03-24:
@bulwahn 
I can checkout the repository successfully using svn co https://svn.apache.org/repos/asf/incubator/log4cxx/trunk. Therefore it is not a problem of proxy. The build configuration is as follows:

Build Configuration:
BB_VERSION        = "1.33.2"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "universal"
TARGET_SYS        = "i586-poky-linux"
MACHINE           = "qemux86"
DISTRO            = "poky"
DISTRO_VERSION    = "2.2+snapshot-20170324"
TUNE_FEATURES     = "m32 i586"
TARGET_FPU        = ""
meta              
meta-poky         = "master:49c2df5652d25e6e9b2a4976658ab9bb17fe6694"
meta-oe           
meta-networking   
meta-python       = "master:d63e5687c7bf6b054c6ed2fc4790e4970b023219"
meta-ros          = "master:eb4f9edf2af38a854a2a8d28e45ea02f6bdb7f12"

I have also changed the protocol in log4cxx recipe from 'http' to 'https' but no avail. The log 
[log_do_fetch.txt](https://github.com/bmwcarit/meta-ros/files/866565/log_do_fetch.txt)
[log_do_fetch.txt](https://github.com/bmwcarit/meta-ros/files/866566/log_do_fetch.txt)

file of the error is attached. 

I can confirm that the work around suggested in #413 using http://cgit.openembedded.org/openembedded/tree/recipes/log4cxx works but then I end up compile errors due to gcc mentioned in #390.

Is it possible to download the trunk and use SRC_URI = file://log4cxx.tar.gz? I have tried but then end up in License issues? Thanks for your help.

BilalArain, 2017-03-25:
When I get work around this failure I end up in #460. 
I close this issue and will work out to compile it directly on the development board.

closed 2017-03-25 by BilalArain