ros/meta-ros #471: tf-conversions: stdlib include error with gcc-6 after python-orocos-kdl update

bulwahn, 2017-03-24:
After the python-orocos-kdl update to the current git repository version, `bitbake tf-conversions`, which depends on python-orocos-kdl fails during do_compile with gcc-6:
```
Build Configuration:
BB_VERSION        = "1.33.2"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "fedora-23"
TARGET_SYS        = "i586-oe-linux"
MACHINE           = "qemux86"
DISTRO            = "nodistro"
DISTRO_VERSION    = "nodistro.0"
TUNE_FEATURES     = "m32 i586"
TARGET_FPU        = ""
meta              = "HEAD:ac5c08ea00816852a712a8d7025bce7e865460ec"
meta-oe           
meta-python       
meta-multimedia   = "master:1dff2351aa6cdafa5a501e8956cb853ab17ed9ae"
meta-qt4          = "master:2c7f8df9039be498f8a2232d1428adb7f4e5e800"
meta-intel-realsense = "master:1d50cb5b82926f8e69a78999e9e8d2ffe2169897"
meta-ros          = "python-orocos-kdl:41bbe1e6a57f94e5516ad27a75672982aa0a00a8"

ERROR: tf-conversions-1.11.8-r0 do_compile: oe_runmake failed
ERROR: tf-conversions-1.11.8-r0 do_compile: Function failed: do_compile (log file is located at /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/temp/log.do_compile.27805)
ERROR: Logfile of failure stored in: /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/temp/log.do_compile.27805
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 32
| /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/recipe-sysroot-native/usr/bin/cmake -H/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/geometry-1.11.8/tf_conversions -B/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/build --check-build-system CMakeFiles/Makefile.cmake 0
| /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/recipe-sysroot-native/usr/bin/cmake -E cmake_progress_start /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/build/CMakeFiles /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/build/CMakeFiles/progress.marks
| make -f CMakeFiles/Makefile2 all
| make[1]: Entering directory '/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/build'
| make -f CMakeFiles/tf_conversions.dir/build.make CMakeFiles/tf_conversions.dir/depend
| make[2]: Entering directory '/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/build'
| cd /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/build && /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/recipe-sysroot-native/usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/geometry-1.11.8/tf_conversions /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/geometry-1.11.8/tf_conversions /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/build /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/build /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/build/CMakeFiles/tf_conversions.dir/DependInfo.cmake --color=
| make[2]: Leaving directory '/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/build'
| make -f CMakeFiles/tf_conversions.dir/build.make CMakeFiles/tf_conversions.dir/build
| make[2]: Entering directory '/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/build'
| [ 33%] Building CXX object CMakeFiles/tf_conversions.dir/src/tf_kdl.cpp.o
| /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/recipe-sysroot-native/usr/bin/i586-oe-linux/i586-oe-linux-g++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"tf_conversions\" -Dtf_conversions_EXPORTS -isystem /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/recipe-sysroot/usr/include/eigen3 -isystem /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/recipe-sysroot/usr/share/orocos_kdl/../../include -isystem /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/orocos-kdl/git-r0/recipe-sysroot/usr/include/eigen3 -I/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/geometry-1.11.8/tf_conversions/include -I/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/recipe-sysroot/opt/ros/indigo/include -I/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/recipe-sysroot/usr/include  -m32 -march=i586  --sysroot=/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0=/usr/src/debug/tf-conversions/1.11.8-r0 -fdebug-prefix-map=/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/recipe-sysroot-native= -fdebug-prefix-map=/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/recipe-sysroot=  -fvisibility-inlines-hidden  -m32 -march=i586  --sysroot=/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/recipe-sysroot -fPIC   -o CMakeFiles/tf_conversions.dir/src/tf_kdl.cpp.o -c /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/geometry-1.11.8/tf_conversions/src/tf_kdl.cpp
| [ 66%] Building CXX object CMakeFiles/tf_conversions.dir/src/tf_eigen.cpp.o
| /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/recipe-sysroot-native/usr/bin/i586-oe-linux/i586-oe-linux-g++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"tf_conversions\" -Dtf_conversions_EXPORTS -isystem /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/recipe-sysroot/usr/include/eigen3 -isystem /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/recipe-sysroot/usr/share/orocos_kdl/../../include -isystem /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/orocos-kdl/git-r0/recipe-sysroot/usr/include/eigen3 -I/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/geometry-1.11.8/tf_conversions/include -I/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/recipe-sysroot/opt/ros/indigo/include -I/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/recipe-sysroot/usr/include  -m32 -march=i586  --sysroot=/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0=/usr/src/debug/tf-conversions/1.11.8-r0 -fdebug-prefix-map=/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/recipe-sysroot-native= -fdebug-prefix-map=/home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/recipe-sysroot=  -fvisibility-inlines-hidden  -m32 -march=i586  --sysroot=/home/lukas/dev/openembedded.org/openembedded- [... cut ...]
| ERROR: Function failed: do_compile (log file is located at /home/lukas/dev/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf-conversions/1.11.8-r0/temp/log.do_compile.27805)
ERROR: Task (/home/lukas/dev/github.com/bmwcarit/meta-ros/recipes-ros/geometry/tf-conversions_1.11.8.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 8253 tasks of which 8252 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/lukas/dev/github.com/bmwcarit/meta-ros/recipes-ros/geometry/tf-conversions_1.11.8.bb:do_compile
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

With gcc-5.4, the issue does not occur.

bulwahn, 2017-03-24:
This probably the line to blame: https://github.com/ros/geometry/blob/indigo-devel/tf_conversions/CMakeLists.txt#L17

I will use the next few hours to check if this issue is quickly addressed by changing the CMakeLists.txt.

ckrooss, 2017-03-24:
I'm having the exact same error with dynamic_reconfigure. It's caused by `-isystem /usr/include`, I think I had the same problem with opencv some time ago, workaround was to use `-I /usr/include`

herb-kuta-lge, 2019-06-28:
@bulwahn < Can this issue be closed?

closed 2019-08-13 by herb-kuta-lge