ros/meta-ros #483: log4cxx-svn-r0 gives 400, cannot build a clean target for roslaunch

prajankya, 2017-04-22:
svn: E170013: Unable to connect to a repository at URL 'http://svn.apache.org/repos/asf/incubator/log4cxx/trunk'
svn: E175002: Unexpected HTTP status 400 'Bad Request' on '/repos/asf/incubator/log4cxx/trunk'

``` bash
$ bitbake roslaunch

Loading cache: 100% |###############################################################################################| Time: 0:00:01
Loaded 3698 entries from dependency cache.
Parsing recipes: 100% |#############################################################################################| Time: 0:00:07
Parsing of 2916 .bb files complete (2491 cached, 425 parsed). 3700 targets, 297 skipped, 0 masked, 0 errors.
WARNING: No recipes available for:
  /home/prajankya/work/toradex/w/build/../layers/meta-jetson-tk1/recipes-graphics/mesa/mesa_11.1.1.bbappend
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION        = "1.32.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-16.04"
TARGET_SYS        = "arm-angstrom-linux-gnueabi"
MACHINE           = "apalis-t30"
DISTRO            = "angstrom"
DISTRO_VERSION    = "v2016.12"
TUNE_FEATURES     = "arm armv7a vfp thumb neon       callconvention-hard"
TARGET_FPU        = "hard"
meta-angstrom     = "HEAD:ee90314737ce65dcc902a75ea7a1941bb38401aa"
meta-toradex-nxp  = "HEAD:ee77ef53d35284bcacc77e849dd30a36fe263ef2"
meta-freescale    = "HEAD:6ec09e6a0f09d1cb9c3761b339590c0a0baeeeb3"
meta-freescale-3rdparty = "HEAD:69ed94bb781a9c1bfbba4fb3026e4cbf3a194878"
meta-toradex-tegra = "HEAD:c411daec746975195f1590c727449adffb8587e0"
meta-jetson-tk1   = "HEAD:e8b87fe8da7c6fcffa37ab245f50082953cc1ee1"
meta-toradex-bsp-common = "HEAD:05faa6a4346eeb7631fe72bcbb9aa7d414bee791"
meta-linaro-toolchain = "HEAD:407fe60740240b3115531cbed129f1e186bfbe3e"
meta-oe           
meta-efl          
meta-gpe          
meta-gnome        
meta-xfce         
meta-initramfs    
meta-systemd      
meta-networking   
meta-multimedia   
meta-python       = "HEAD:1efa5d623bc64659b57389e50be2568b1355d5f7"
meta-lxde         = "HEAD:cb926caf26762a68bb735fd8bd20843fdf1b0cfe"
meta-browser      = "HEAD:1edcce7791b4cee9a515c1f11b351753a4f8b12e"
meta-qt4          = "HEAD:2c7f8df9039be498f8a2232d1428adb7f4e5e800"
meta-qt5          = "HEAD:9aa870eecf6dc7a87678393bd55b97e21033ab48"
meta-freescale-distro = "HEAD:ae27e8a8a068b960d6c1219f50b2e8ccc97f0bea"
meta-toradex-demos = "HEAD:9fa810182cbb085554ad3a596db6dbf1fc9a0a73"
meta-nodejs       = "HEAD:5878455edb0b62cbd501927466a19700fe4c70e6"
meta-ros          = "HEAD:a5647ad610ce929ec8bf5d7e3321feaf7e10be3d"
meta              = "HEAD:c8d96b10ee3bc2eae0fd269d2564286fd0bc82ed"

Initialising tasks: 100% |##########################################################################################| Time: 0:00:04
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
WARNING: log4cxx-svn-r0 do_fetch: Failed to fetch URL svn://svn.apache.org/repos/asf;module=incubator/log4cxx/trunk;protocol=http, attempting MIRRORS if available
ERROR: log4cxx-svn-r0 do_fetch: Fetcher failure: Fetch command export PATH="/home/prajankya/work/toradex/w/layers/openembedded-core/scripts:/home/prajankya/work/toradex/w/build/tmp-glibc/sysroots/x86_64-linux/usr/bin/arm-angstrom-linux-gnueabi:/home/prajankya/work/toradex/w/build/tmp-glibc/sysroots/apalis-t30/usr/bin/crossscripts:/home/prajankya/work/toradex/w/build/tmp-glibc/sysroots/x86_64-linux/usr/sbin:/home/prajankya/work/toradex/w/build/tmp-glibc/sysroots/x86_64-linux/usr/bin:/home/prajankya/work/toradex/w/build/tmp-glibc/sysroots/x86_64-linux/sbin:/home/prajankya/work/toradex/w/build/tmp-glibc/sysroots/x86_64-linux/bin:/home/prajankya/work/toradex/w/layers/openembedded-core/scripts:/home/prajankya/work/toradex/w/layers/openembedded-core/bitbake/bin:/home/prajankya/work/toradex/q1/layers/openembedded-core/scripts:/home/prajankya/work/toradex/q1/layers/openembedded-core/bitbake/bin:/home/prajankya/bin:/home/prajankya/.local/bin:/home/prajankya/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin"; export HOME="/home/prajankya"; /usr/bin/env svn --non-interactive --trust-server-cert co --no-auth-cache -r 1738416 http://svn.apache.org/repos/asf/incubator/log4cxx/trunk@1738416 incubator/log4cxx/trunk failed with exit code 1, output:
svn: E170013: Unable to connect to a repository at URL 'http://svn.apache.org/repos/asf/incubator/log4cxx/trunk'
svn: E175002: Unexpected HTTP status 400 'Bad Request' on '/repos/asf/incubator/log4cxx/trunk'

ERROR: log4cxx-svn-r0 do_fetch: Fetcher failure for URL: 'svn://svn.apache.org/repos/asf;module=incubator/log4cxx/trunk;protocol=http'. Unable to fetch URL from any source.
ERROR: log4cxx-svn-r0 do_fetch: Function failed: base_do_fetch
ERROR: Logfile of failure stored in: /home/prajankya/work/toradex/w/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/log4cxx/svn-r0/temp/log.do_fetch.17996
ERROR: Task (/home/prajankya/work/toradex/w/build/../layers/meta-ros/recipes-devtools/log4cxx/log4cxx_svn.bb:do_fetch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1218 tasks of which 1216 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory

Summary: 1 task failed:
  /home/prajankya/work/toradex/w/build/../layers/meta-ros/recipes-devtools/log4cxx/log4cxx_svn.bb:do_fetch
Summary: There were 2 WARNING messages shown.
Summary: There were 3 ERROR messages shown, returning a non-zero exit code.
```

Attached is the log file
[log.do_fetch.17996.txt](https://github.com/bmwcarit/meta-ros/files/948700/log.do_fetch.17996.txt)

bulwahn, 2017-04-22:
If you do not use gcc6, you can simply revert commit https://github.com/bmwcarit/meta-ros/commit/b4baa4a896f6179465be233940e6ea4f4bf0c269 with `git revert b4baa4a896f6179465be233940e6ea4f4bf0c269`. Then your problem should be solved. I started working on a proper resolution a few weeks ago, but did not finish it yet.

prajankya, 2017-04-22:
I will check and get back to you right away..

prajankya, 2017-04-22:
No, its still trying to download the same package, log4cxx-svn-r0, and when it does, HTTP status 400 is given.

Any other ways to solving it ?

prajankya, 2017-04-22:
Ok, I did as per you told to revert it, and it worked, but as stated above, I tried to change the revision to download that commit then it didn't worked, I know its out of the scope of this project but can you help me out with this problem?

bulwahn, 2017-04-22:
I am sorry, but I do not understand your question. Could you rephrase what are you trying to achieve?

prajankya, 2017-04-22:
If I go in the folder of meta ros and do the git revert as you said, it compiles without errors.

But as I updated my repo file to get me the commit you referred to in https://github.com/SnappyRobotics/snappy-toradex/commit/772cf15000c54da5a31472382bb64d6b7301ef20

In "repo"(the tool which I am using to download multiple repos with given the revision/commit number) it gave the message of `reverted 270 commits` meaning it went back to the commit you described, but it didn't worked. 

So ,knowing that this is not related to your project, my question is if do you know what wrong I am doing in "repo", if possible can you help me out.

bulwahn, 2017-04-22:
Okay now, I understand. You are probably mis-understanding what `git revert` does. When you apply git revert on the meta-ros repository, you selectively revert one commit from the past and apply that reverted commit on top of the current head. When you update the repo file to that commit, you go back to the state of that commit, so going 270 commits back of the current head. Those 270 commits are of course a number of improvements and possibly some crucial changes for the overall build to work. My proposition what to do will follow in a minute... (once I am done with a commit to address the log4cxx issue.)

bulwahn, 2017-04-22:
I have provided a commit that should resolve your issue in a personal branch. So you must now just update the repo file to use https://github.com/bulwahn/meta-ros/commit/290766c50482a25deba51a71ddf05f96553911cf.

In more detail, update:
```
<remote alias="repo" fetch="http://github.com/bmwcarit" name="ros" />
```
to:
```
<remote alias="repo" fetch="http://github.com/bulwahn" name="ros" />
```

and update:
```
  <project name="meta-ros.git" path="layers/meta-ros" remote="ros" revision="b4baa4a896f6179465be233940e6ea4f4bf0c269" />
```
to:
```
  <project name="meta-ros.git" path="layers/meta-ros" remote="ros" revision="290766c50482a25deba51a71ddf05f96553911cf" />
```

I will open a pull request with this change soon, once I have completely tested it on my local setup. Then the second maintainer of meta-ros will review it and in a couple of days, this is merged in the mainline development.

prajankya, 2017-04-22:
Thanks for solving this issue. solved for now, looking forward for it to be merged into mainline.

closed 2017-04-22 by prajankya

bulwahn, 2017-04-24:
We have merged the commit to the main-line repository. So I would suggest to update the repo file to point the current master of the mainline repository. I will be deleting my personal branch in a couple of days, just to keep my personal space a bit cleaner (but I will check again if your repo file has been changed already).
By the way, if meta-ros is useful for you, please star this repository on github.

prajankya, 2017-04-27:
Thank you @bulwahn the problem is solved in the main-line.