ros/meta-ros #485: meta-ros release: Are there any plans to update to Kinetic version?

TSC21, 2017-04-23:
I saw some comments regarding a dev branch for Kinetic but can't seem to find it. Are there any progresses on this matter? Any issues that can be tackled together to help this get through?

Thanks in advance.

(btw, I'm one of the active developers of `mavros` and I have interest on this to happen since we actually don't maintain the indigo version and left it as is, but Travis still builds the master version for indigo though - in any case, the interest is to sync `meta-ros` with the current version).

TSC21, 2017-04-23:
@JochiPochi and @guludo probably this also interests you?

TSC21, 2017-04-24:
@rojkov @bulwahn any updates you can give us on this? Is this on close future plans?

rojkov, 2017-04-24:
@TSC21 I can only speak for myself. Currently my customers don't require Kinetic. And given how little of my time I can spend on meta-ros I have no plans for the update.

In my personal time I'm working on automating test builds for meta-ros (TravisCI is not an option unfortunately for such big projects). If I complete it soon enough I'll have a look at Kinetic.

TSC21, 2017-04-24:
@rojkov thanks for your feedback. That would be great! Let me know if you need help on that. I would be glad to contribute to it.

bulwahn, 2017-04-24:
@TSC21 
The kinetic branch I reported last year is available on my personal fork on the kinetic-experimental branch. I am now actually in the progress of creating a new version of that kinetic-experimental, but there are still some open issues. I have published the current state on my personal fork on kinetic-experimental-v2-alpha1. On this branch, fcl, collada-urdf and moveit-ros-perception currently fails.

In general, we have been discussing moving to the next distribution in the past on the meta-ros mailing list [1, 2]. Our discussion forum has now moved to discourse.ros.org. I am pragmatic and would not reject if the majority of contributors & users want to move forward to kinetic. Then we can do so.
However, there are some concerns from my side:
- If we want to provide meta-ros for two distributions (indigo and kinetic), we would need to maintain two branches. This could be challenging, especially because we currently do not have any CI set up yet. So, I would not vote for maintaining two branches, if the number of contributors and maintainers does not significantly increase and we do not have a solid CI in place.
- If we want to just maintain one branch, we would just update all recipes from indigo to kinetic. However, before we do so, the majority should be in favour of doing that. Also, I would also prefer to create a proper release of meta-ros for indigo before moving away from indigo. This should then also work with the latest yocto project distribution at the time of our release. Currently, however, the current master does not work with the currently latest yocto distribution. So, I propose to wait at least until the next yocto distribution release is released.

[1] https://groups.google.com/forum/#!topic/meta-ros/xXyfMG5-frI
[2] https://groups.google.com/forum/#!topic/meta-ros/XpfLXJrjQUI

TSC21, 2017-04-24:
@bulwahn makes total sense to me what you just said. And based on the current state, I think the 2nd option would be the best and open an issue on it would be good to know people's opinion.

> Currently, however, the current master does not work with the currently latest yocto distribution

I tested already current master of yocto with current `master` of `meta-ros` and `bitbake` run till the end perfectly just with 3 warnings related to failed fetching.

Minipada, 2017-04-25:
I tried last week and still got issues with gcc6, for navfn and amcl, not an issue on meta-ros, but the packages themselves.
@TSC21 the easiest for you is to know which packages you need to update and use your own recipes for this

bulwahn, 2017-04-25:
@Minipada The gcc6 issue is strange!? I thought we have addressed this in the current master of meta-ros with provided patches and fixed it already in the upstream repository. Are you using the current master?

Minipada, 2017-04-25:
I was using

* meta-openembedded @ 2cdea6c
* meta-ros @ b3ffbe4
* poky @ fe47bff

I didn't go deeply into it, I'll update you after new tests

TSC21, 2017-04-25:
> @TSC21 the easiest for you is to know which packages you need to update and use your own recipes for this

As you make reference to that, what is the easiest way of selecting which group of packages I want so that bitbake makes an image out of those?

JochiPochi, 2017-04-25:
@bulwahn I have some dirty ansible code to spin up a CI server that just builds every bitbake recipe and reports errors. Are there other requirements for a meta-ros CI? Maybe we can just start using that if I clean up a bit.

TSC21, 2017-04-25:
@JochiPochi I'm totally intersted on that as I'm also working with Ansible myself. Any thing I can assist?

rojkov, 2017-04-26:
I'm working on a CI solution based on Azure's server-less functions and their message queues with VMs fired up only for actual builds. I guess it's cheaper than hosting an instance of Jenkins. I wanted to use Google cloud first, but as a private user I'm not ready to bother with VAT reporting and AWS looks a bit pricey.

The end goal is to create a simple web-based tool for generating cloud-based (public or private) CI deployments for other projects using Yocto. I doubt there's a big enough market for such tool, but it's just my way of having fun besides robots.

@JochiPochi we can start with ansible indeed.

bulwahn, 2017-04-26:
I created a CI system based on docker, and published that work on my personal repository in the ci branch (https://github.com/bulwahn/meta-ros/tree/ci). The CI was running on my company's internal systems, and I was looking into providing a public instance, but then time for that got short, and I stopped that activity and at some point even killed the internal CI as I could not keep up the maintenance.

At around the same time last year, I had a discussion with Tully Foote who is maintaining the ROS CI system a couple of months ago, and he was willing to integrate meta-ros CI testing into the ROS CI system. He pointed me to the repository https://github.com/ros-infrastructure/ros_buildfarm and suggested to simply add the needed Jenkins definitions in there. Unfortunately, I did not make any progress there as well.

TSC21, 2017-04-26:
> He pointed me to the repository https://github.com/ros-infrastructure/ros_buildfarm and suggested to simply add the needed Jenkins definitions in there. Unfortunately, I did not make any progress there as well.

I guess this can be the best approach as we will integrate `meat-ros` on ROS CI system.

JochiPochi, 2017-04-26:
Plenty of us have been looking into CI for meta-ros then. May I suggest to move this conversation to https://github.com/bmwcarit/meta-ros/issues/490 ?

bulwahn, 2017-10-04:
I keep an experimental local branch for the kinetic distribution. I have rebased it to the current master, updated and adjusted the recipes to get most recipes working again.

It is now published on my local github repository at:

  https://github.com/bulwahn/meta-ros/tree/kinetic-experimental-v3-alpha1

I have tested to build it, and currently, most packages build without issues. The packages that fail on kinetic are tracked in the meta-ros issue tracker. Feel free to use it if you like to work with the kinetic distribution.

However, please remember that this is an experimental local branch:
First, do not expect that this branch has been tested with many different build configurations, so possibly, your build configuration could cause build failure we are unaware of. Second, this branch is work in progress, and hence, the commits of this branch will be rebased and modified, and the git history will change in a non-monotonic way in the future. Nevertheless, go ahead and try this branch and report issues if you encounter them.

I already plan to merge this kinetic branch (or a branch that has evolved from this one) into the main repository, and maintain two branches for indigo and kinetic. Before I do this, there are still a few open topics that I would like to have working on in the current master branch of meta-ros.

herb-kuta-lge, 2019-06-28:
We are planning to support **melodic**, at least at first, as it's listed as the recommended ROS 1 distro by http://wiki.ros.org/Distributions. (See https://github.com/ros/meta-ros/wiki/Superflore-OE-Recipe-Generation-Scheme#milestones for our plans.) But if the community would rather have **kinetic** be supported ahead of **melodic**, we can certain change our plans.

herb-kuta-lge, 2019-11-04:
We now have partial support for **melodic** and there are no plans to support **kinetic**.

closed 2019-11-04 by herb-kuta-lge