ros/meta-ros #489: Update Mavros to v0.18.x to get MAVLink 2

mindThomas, 2017-04-25:
When can we expect this meta-ros layer to be updated to include mavros of version v0.18.xx ?
We are currently working with the Intel Aero drone, using this meta-ros layer. Unfortunately we are unable to do any onboard flight logging because the PX4 ULogger script requires MAVLink 2.

I have tried to do some of the changes in meta-ros myself, pulling in the latest MAVLink, but without any luck.
Instead I ran into problems with the message generator not generating proper C++ header files (common.hpp etc.) and after solving that the Python based message generator (mavgen.py) within pymavlink complained about missing the Python library, "future"

Hopefully anyone can point us in any usefull way?

Kind regards
Thomas Jespersen

TSC21, 2017-04-25:
@mindThomas I'm currently interested on this also (as one of the maintainers of mavros). Problem starts when the `meta-ros` layer still depends on indigo, while us on mavros do not currently maintain master for indigo (we have an indigo-devel branch which is not up to date with the current master). Indigo also uses the previous C msg generator and parser, while we are already using the C++ one. So the idea could be trying to update `ros-mavlink` (which gets the mavlink package) and also `mavros` to current master and check out it goes with current `meta-ros` implementation (this of course is not the ideal implementation as the deb package version is indigo, while with this you have to build it from src). Haven't tried yet, but seems you are getting problems about `future`. Have you tried to add `python-future` deb package to dependencies on the bb file?

mindThomas, 2017-04-25:
I have tried to include the python-future from source and also tried then adding it to pymavlink as a "DEPENDS". No luck though.

TSC21, 2017-04-25:
Can you paste the error here? Also could be interesting that you share your current `bblayers.conf` and `local.conf` here also so to help debug this. Thanks

TSC21, 2017-04-25:
Also, whenever you do a mod, try doing `bitbake -c cleanall <package>`.

mindThomas, 2017-04-25:
My BB files are unmodified from the original repository.
bblayers.conf here as the template: https://github.com/intel-aero/meta-intel-aero/blob/master/conf/bblayers.conf.sample
local.conf here as the template: https://github.com/intel-aero/meta-intel-aero/blob/master/conf/local.conf.sample
I have managed to also come around the thing of "forcing a rebuild" by increasing the PR number in the recipe.

TSC21, 2017-04-25:
Have you include `ROS_USE_PYTHON3 = "yes"` on `local.conf`? Are you using latest `meta-python` layer?

mindThomas, 2017-04-25:
Well, I have just realized that Future is already a part on the image.
And keep in mind that Future was also used for old message generations with Mavros 0.17.5, which is the version that is contained within meta-ros. Without any changes to the image I can successfully bitbake everything with this old mavros version, hence these settings must be set right?
Or am I wrong? Would anything important change when I change the mavros version?

TSC21, 2017-04-25:
> Would anything important change when I change the mavros version?

The most important one is related with the msg parse as I said (c++ instead of c now). If it implies Future then it has to be `pymavlink` related so I suppose that you have to update it also.

mindThomas, 2017-04-25:
The ros-mavlink uses http://github.com/mavlink/mavlink-gbp-release
Which branch would you take from this? Currently I am trying with "release/kinetic/mavlink"

TSC21, 2017-04-25:
https://github.com/mavlink/mavros/blob/master/dependencies.rosinstall

So you have the right release. As I said, this has to be `pymavlink` related, and you will have to install the correct dependencies for it. Usually `pymavlink` is installed using `pip3 install` if you are using `python3`, which installs the latest dependencies for `pymavlink` - so please recheck the if you have latest `future` (https://pypi.python.org/pypi/future) and `lxml` (http://lxml.de/installation.html).

TSC21, 2017-04-25:
Try adding `python3-future` to `DEPENDS` instead of `python-future`.

mindThomas, 2017-04-25:
Do you know if that has changed? Because I am able to build with the indigo release without depending on anything.

mindThomas, 2017-04-25:
Depending on python3-future does not seem to work: `Nothing PROVIDES 'python3-future'`

TSC21, 2017-04-25:
I know that I had to use `pip3 install future` to install future when your system default python is version 3.

mindThomas, 2017-04-25:
Well pymavlink does not seem to depend on Python3. At least it requires/inherits setuptools and not setuptools3 : https://github.com/ArduPilot/pymavlink/blob/0e949ae41c0db266c8a903a5bcc8d6d7e1bb5c20/setup.py#L3

TSC21, 2017-04-25:
It doesn't have to be with dependency on python 3. It has to be with lib links. So you can tun pymavlink scripts on both python 2.7 and python 3 but the link with the version depends on what you have on your system or the version of pip you use.

TSC21, 2017-04-26:
@mindThomas any luck with this in the meantime? Just to reinvoke one of my comments, as I consider it may be important:
> Have you included `ROS_USE_PYTHON3 = "yes"` on `local.conf`? Are you using latest `meta-python` layer?

mindThomas, 2017-04-26:
I were using the latest meta-python, yes.
But now we have "given up" and went back and did a fresh pull of the manifest.
Any suggestions are welcome though.

bulwahn, 2017-04-29:
I could successfully update the mavros recipe to the current kinetic version, and get ros-mavlink to build. Then, the mavros build fails with the already known Eigen include issue. This is available in the update-mavros branch of my personal fork. 
By the way, I am using python3 with `ROS_USE_PYTHON3 = "yes"` and the current commits are currently limited to python3 and will not directly work with python2, but it should feasible to adjust them to python2.

If you update to my build configuration, you should be able to get at least to compile mavros and re-produce the Eigen include issue. If you tell me your exact build configuration, I can look to port my commits to your build configuration.

My build configuration is:
```
Build Configuration:
BB_VERSION        = "1.33.4"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "ubuntu-14.04"
TARGET_SYS        = "x86_64-oe-linux"
MACHINE           = "qemux86-64"
DISTRO            = "nodistro"
DISTRO_VERSION    = "nodistro.0"
TUNE_FEATURES     = "m64 core2"
TARGET_FPU        = ""
meta              = "master:e584be78f92ee6f08f570c239698d56ac78d05f9"
meta-oe           
meta-multimedia   
meta-python       = "master:2c21aa8633e4e3538099bcc56d76ab0a594411b4"
meta-ros          = "mavros-update:c5479e13929bb1d6caa9ec196dcf4b88379a4dd8"
```

TSC21, 2017-04-29:
@bulwahn thank you very much for the progress you got in this. I specifically don't have a build but would be interested this could be available to every possible machines as one of mavros developers.
What is that error you are having with Eigen include and how can I reproduce that?

bulwahn, 2017-04-29:
@TSC21 The Eigen issue in the mavros recipe is reported in #467.  I believe I have found a solution to #467, and will possibly provide a commit within the next hour. I think we will just need to repair mavros and keep it working, then it will be available for many machines and all new users.

TSC21, 2017-04-29:
Tracked! Thanks for this @bulwahn! It will for sure help many others with interest in Mavlink-based drones. The solution you are considering is for the the meta layer or for the mavros repo itself btw?

TSC21, 2017-04-29:
Just a suggestion: renaming `ros-mavlink` to `mavlink` only would be a better approach, as there's a [deprecated package](https://github.com/mavlink/mavlink_ros) which name may be misleading. Also I think it would be very interesting if you could present this current solution for Yocto Project/OpenEmbedded in the [Mavlink Github project](https://github.com/mavlink) to be integrated to it and extend the user list.

bulwahn, 2017-04-29:
@TSC21 Thanks for the suggestion. Once we have merged this upstream, we should continue to discuss where to present meta-ros in the mavlink project, e.g., possibly a blog/new entry on the mavlink webpage.

TSC21, 2017-04-29:
> @TSC21 Thanks for the suggestion. Once we have merged this upstream, we should continue to discuss where to present meta-ros in the mavlink project, e.g., possibly a blog/new entry on the mavlink webpage.

Totally agree. The person of contact will be Lorenz Meier - the project founder.

bulwahn, 2017-04-29:
I have found a first solution for #467. So now my personal branch should provide the recipe for the kinetic mavros package. We will continue to merge those commits in the main repository in the next week.

mindThomas, 2017-04-29:
At least we are looking forward to get meta-ros with a proper version of Mavros running on our Intel Aero.
Currently we are having great sucess with Mavros for our development of controllers, estimators and our current work with SLAM on the Intel Aero drone - but unfortunately due to the meta-ros issues we have to stick to run all of our ROS nodes on a local PC with Ubuntu running the latest version of Mavros and with a wireless connection to the drone.
I would be happy to test and share our results whenever we get Mavros updated and included in meta-ros.

TSC21, 2017-04-29:
> I have found a first solution for #467. So now my personal branch should provide the recipe for the kinetic mavros package. We will continue to merge those commits in the main repository in the next week.

Perfect! Thanks!

commit 118a1d07 (2017-05-08): Merge pull request #502 from bulwahn/mavros-update-v2

repair and update mavros to kinetic (resolves #489)
closed 2017-05-08 by rojkov