ros/meta-ros #490: Feature Request: Continuous Integration for meta-ros

JochiPochi, 2017-04-26:
This issue exists so anyone interested in CI/CD for the meta-ros project can talk about their progress, set of requirements and early stage documentation.
Some discussion about this started on:
https://github.com/bmwcarit/meta-ros/issues/485
But CI merits it's own conversation.

So far people interested are: @bulwahn @rojkov @TSC21

JochiPochi, 2017-04-26:
I put some work a few months ago to adapt the ostro project CI to just build all recipes found on all of the layers of a bitbake build.
`bitbake-layers show-recipes`
Maybe that was overkill as it would take way longer than just building core-image-ros-world

Here are a few links that got me started:
[Intel CI/CD OpenIoT Summit Presentation](http://events.linuxfoundation.org/sites/events/files/slides/Creating%20Continuous%20Delivery%20for%20Yocto%20Based%20IoT%20Distribution%2020161012%20LF.pdf)
[Ostro Project CI/CD ansible](https://github.com/ostroproject/ostroproject-ci-ansible)
[Ostro Project CI/CD](https://github.com/ostroproject/ostroproject-ci/tree/production)

Currently all my work was local and has some work specific code on it. Since there are people interested, I'll remove that and make it available. Given my current schedule, this will probably happen by the first or second week of May. (About two weeks from now)

JochiPochi, 2017-04-26:
We should talk about requirements thought, just so that all efforts go on the same direction.
I am aiming to produce a CI that given a bitbake layer it just builds everything on it. It was something generic so I could use it on other yocto/oe based projects too.
Maybe that's too little, maybe it's too much. It only ensures that recipes build. Thoughts?

bulwahn, 2017-09-01:
In the meantime, @rojkov has installed a continuous integration for meta-ros. We are surely open to improvements of any kind as long as it does not break what we already have in place.