ros/meta-ros #491: Help required

prajankya, 2017-04-27:
I am trying to add a custom Repo for ros as a recipe, but its not working.
This is the ROS project I want to integrate [/SnappyRobotics/warehouse_robot](/SnappyRobotics/warehouse_robot)

This is where I am keeping all my recipes [/SnappyRobotics/meta-snappy-setup](/SnappyRobotics/meta-snappy-setup)

and this is the commit where I have added this.[/SnappyRobotics/meta-snappy-setup#7760aa56d11142e66cd1aea3b8c2946ef967b8ce](/SnappyRobotics/meta-snappy-setup/tree/7760aa56d11142e66cd1aea3b8c2946ef967b8ce)

The recipe name is `recipes-warehouse_robot`

I am able to compile, but there is no file added in the built image, I had done it previously by adding it to [`IMAGE_INSTALL_append`](/SnappyRobotics/meta-snappy-setup/blob/7760aa56d11142e66cd1aea3b8c2946ef967b8ce/buildconf/local.conf#L275) but here the error is similar to *there is no recipe/layer providing warehouse_robot*

I tried following /blob/master/recipes-ros/chatter/chatter.inc, where `inherit catkin` is used I guess to solve the install issue. Can you help me out, for how to compile it ?

TSC21, 2017-04-27:
Try to follow [this](https://github.com/vmayoral/diving-meta-ros) and see if it helps.

bulwahn, 2017-04-27:
It is probably the underscore in the recipe filename `warehouse_robot` that causes the confusion. Just rename the filename in warehouse-robot; that is at least the default thing we do in meta-ros and other OpenEmbedded developments.

prajankya, 2017-04-28:
It worked like a charm thank you @TSC21  for that guide!

prajankya, 2017-04-28:
@bulwahn I have renamed the package, possibly that was the reason. Thanks.

prajankya, 2017-04-28:
Actually after following the tutorial I figured out that it just copies package.xml in the built image, and hence not complete repo. also there is no catkin_make, if I can git pull and make it there. @TSC21 in the tutorial it was python, and mine is cpp, does is matter which language I am using, as cpp need to be compiled.

should I do 
```
do_install() {
    	cp -r ${S}/* ${D}${PN}
}
```
in my .bb file, to copy ? but how can I include the *built binaries* into the image.

bulwahn, 2017-04-28:
Your CMakeLists.txt is incomplete: you need to install the files properly with the according CMake commands. There should be a documentation on the ROS wiki how to write a proper catkin CMakeLists.txt file.

prajankya, 2017-04-28:
Sorry for my lack of knowledge, I had been used build ros projects in user directory and hence never used install command in CMakeLists, I am trying it out now as you told, in [http://wiki.ros.org/catkin/CMakeLists.txt](http://wiki.ros.org/catkin/CMakeLists.txt) point 11.

But also, as asked in #494, the same method of making recipe I had followed for package [gennodejs](https://github.com/RethinkRobotics-opensource/gennodejs)  and it even has an **install** command in its CMakeLists.txt file. Still that is not working(more details on #494 ), so is the CMakeLists the only thing to configure or do I need to update something more for making gennodejs and my custom repo to work?

prajankya, 2017-05-05:
This issue is solved as you said @bulwahn I had to change my CMakelist, and as referred above for another issue, they are not linked. So, Thank you.

closed 2017-05-05 by prajankya