ros/meta-ros #494: gennodejs

prajankya, 2017-04-28:
can I run (gennodejs)[http://wiki.ros.org/gennodejs] as it is part of the ros_core, I thought gennodejs or genpy is available, but there is no recipe for it.

prajankya, 2017-04-28:
solved it with comment from #491 to use https://github.com/vmayoral/diving-meta-ros as a tutorial

prajankya, 2017-04-28:
The gennodejs is compiled, but there is no binary and hence its not working..

This is my package where recipe is kept [https://github.com/snappyrobotics/meta-snappy-setup](https://github.com/snappyrobotics/meta-snappy-setup)

This is the package I want to use which uses gennodejs [http://wiki.ros.org/rosnodejs/overview/messages](http://wiki.ros.org/rosnodejs/overview/messages)

On this page I want to use messages in 'from catkin' and cannot use 'on the fly'

bulwahn, 2017-05-03:
The `gennodejs` is not shipped in the indigo distribution, but only with the kinetic distribution. As the current master of meta-ros generally aims to be an indigo distribution, gennodejs is not included.
The experimental work-in-progress branch kinetic-experimental-v2-alpha2 contains gennodejs and might work (although it is experimental).

bulwahn, 2017-05-03:
I honestly do not understand this sentence:
`The gennodejs is compiled, but there is no binary and hence its not working..`

I have not used gennodejs, but I would expect it to provide a message generator just as the other message generators for python and C++. The other message generators are written in python, and do not provide a binary either. Could you be more specific what the error message is? Could you provide a  small example package that uses gennodejs?

prajankya, 2017-05-05:
on further testing and reproducing I found out that gennodejs will not create msgs files ( but it will provide a message generator)

as opposite to genpy, which generates and keeps the messages.

I think its not a issue with this repo, but rather the way in which the gennodejs is implemented, it need to run catkin_make once, with all the msgs files in built in it,.

As it is not possible to solve from this repo, I am closing this issue, and thank you for your support.!

closed 2017-05-05 by prajankya