ros/meta-ros #495: webkit-gtk issue

prajankya, 2017-04-28:
I am getting a error for 

ERROR: Nothing PROVIDES 'webkit-gtk' (but build/../layers/meta-ros/recipes-extended/wxwidgets/wxwidgets_2.9.5.bb DEPENDS on or otherwise requires it). Close matches:
  webkitgtk
  webkit-efl

I have installed all the dependencies declared in the README of this repo. Is it in any other package ?

bulwahn, 2017-04-29:
The joint-state-publisher, wxpython and wxwidgets recipes in meta-ros are not maintained anymore since quite some time. They were added by a contributor that is not active anymore, and we encountered an issue that we could not resolve in our available spare time. So, we removed the joint-state-publisher from our test builds, and simply left the existing recipes unmaintained for the first user to start investigating the needed work to get it building again.

prajankya, 2017-04-29:
Thank you, I will try if I can.

closed 2017-04-29 by prajankya

bulwahn, 2017-04-29:
Resolving the reported issue is however easy, just replace webkit-gtk with webkitgtk in the wxwidgets recipe. You will encounter more issues after that though.

prajankya, 2017-04-29:
true, I am trying the same. Further and further issues, anyways, will get back to you with results asap.

ioanb7, 2017-09-03:
Hi, any updates on this?

ioanb7, 2017-09-03:
My logs: https://hastebin.com/zipedubofe.pas .

It just returns 1, no further explanation. I wonder if there would be an easy fix to it though.

ioanb7, 2017-09-03:
So i've managed to get wxwidgets running: https://github.com/bmwcarit/meta-ros/pull/549

But I still have problems getting wxpython to work.

What I've tried (very similar to what it's already there, just changed an include path):
https://hastebin.com/osatemurec.nginx

Resulting in:
https://hastebin.com/yixebiduxe.sql

I think I am missing a g++ flag?