ros/meta-ros #496: robot-state-publisher clean bitbake error

prajankya, 2017-04-29:
Its a clean run for `bitbake robot-state-publisher`
(clean run meaning before this bitbake was never ran in that folder to build anything)

```
DEBUG: Executing shell function do_compile
NOTE: make -j 24
/home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/x86_64-linux/usr/bin/cmake -H/home/prajankya/work/toradex/cc/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/pluginlib/1.10.5-r0/pluginlib-1.10.5 -B/home/prajankya/work/toradex/cc/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/pluginlib/1.10.5-r0/build --check-build-system CMakeFiles/Makefile.cmake 0
/home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/x86_64-linux/usr/bin/cmake -E cmake_progress_start /home/prajankya/work/toradex/cc/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/pluginlib/1.10.5-r0/build/CMakeFiles /home/prajankya/work/toradex/cc/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/pluginlib/1.10.5-r0/build/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/prajankya/work/toradex/cc/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/pluginlib/1.10.5-r0/build'
make -f CMakeFiles/plugin_tool.dir/build.make CMakeFiles/plugin_tool.dir/depend
make[2]: Entering directory '/home/prajankya/work/toradex/cc/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/pluginlib/1.10.5-r0/build'
cd /home/prajankya/work/toradex/cc/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/pluginlib/1.10.5-r0/build && /home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/x86_64-linux/usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/prajankya/work/toradex/cc/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/pluginlib/1.10.5-r0/pluginlib-1.10.5 /home/prajankya/work/toradex/cc/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/pluginlib/1.10.5-r0/pluginlib-1.10.5 /home/prajankya/work/toradex/cc/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/pluginlib/1.10.5-r0/build /home/prajankya/work/toradex/cc/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/pluginlib/1.10.5-r0/build /home/prajankya/work/toradex/cc/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/pluginlib/1.10.5-r0/build/CMakeFiles/plugin_tool.dir/DependInfo.cmake --color=
Scanning dependencies of target plugin_tool
make[2]: Leaving directory '/home/prajankya/work/toradex/cc/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/pluginlib/1.10.5-r0/build'
make -f CMakeFiles/plugin_tool.dir/build.make CMakeFiles/plugin_tool.dir/build
make[2]: Entering directory '/home/prajankya/work/toradex/cc/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/pluginlib/1.10.5-r0/build'
[ 50%] Building CXX object CMakeFiles/plugin_tool.dir/src/plugin_tool.cpp.o
/home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/x86_64-linux/usr/bin/arm-angstrom-linux-gnueabi/arm-angstrom-linux-gnueabi-g++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"pluginlib\" -I/home/prajankya/work/toradex/cc/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/pluginlib/1.10.5-r0/pluginlib-1.10.5/include -I/home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/include -I/home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/apalis-t30/usr/include  -march=armv7-a -mthumb -mfpu=neon  -mfloat-abi=hard  --sysroot=/home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/apalis-t30 -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/prajankya/work/toradex/cc/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/pluginlib/1.10.5-r0=/usr/src/debug/pluginlib/1.10.5-r0 -fdebug-prefix-map=/home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/x86_64-linux= -fdebug-prefix-map=/home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/apalis-t30= -fvisibility-inlines-hidden   -o CMakeFiles/plugin_tool.dir/src/plugin_tool.cpp.o -c /home/prajankya/work/toradex/cc/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/pluginlib/1.10.5-r0/pluginlib-1.10.5/src/plugin_tool.cpp
[100%] Linking CXX executable devel/lib/pluginlib/plugin_tool
/home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/x86_64-linux/usr/bin/cmake -E cmake_link_script CMakeFiles/plugin_tool.dir/link.txt --verbose=1
/home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/x86_64-linux/usr/bin/arm-angstrom-linux-gnueabi/arm-angstrom-linux-gnueabi-g++    -march=armv7-a -mthumb -mfpu=neon  -mfloat-abi=hard  --sysroot=/home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/apalis-t30 -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/prajankya/work/toradex/cc/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/pluginlib/1.10.5-r0=/usr/src/debug/pluginlib/1.10.5-r0 -fdebug-prefix-map=/home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/x86_64-linux= -fdebug-prefix-map=/home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/apalis-t30= -fvisibility-inlines-hidden  -march=armv7-a -mthumb -mfpu=neon  -mfloat-abi=hard  --sysroot=/home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/apalis-t30 -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/prajankya/work/toradex/cc/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/pluginlib/1.10.5-r0=/usr/src/debug/pluginlib/1.10.5-r0 -fdebug-prefix-map=/home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/x86_64-linux= -fdebug-prefix-map=/home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/apalis-t30= -fvisibility-inlines-hidden -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed  -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed CMakeFiles/plugin_tool.dir/src/plugin_tool.cpp.o  -o devel/lib/pluginlib/plugin_tool -rdynamic -lboost_filesystem-mt -lboost_system-mt /home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/libclass_loader.so -lPocoFoundation -ldl /home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/librosconsole.so /home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/librosconsole_log4cxx.so /home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/librosconsole_backend_interface.so -llog4cxx -lboost_regex-mt /home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/librostime.so /home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/libcpp_common.so -lboost_thread-mt -lboost_chrono-mt -lboost_date_time-mt -lboost_atomic-mt -lconsole_bridge /home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/libroslib.so /home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/librospack.so -lpython2.7 -lboost_filesystem-mt -lboost_program_options-mt -lboost_system-mt -ltinyxml -ltinyxml -lboost_filesystem-mt /home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/libclass_loader.so -lPocoFoundation -ldl /home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/librosconsole.so /home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/librosconsole_log4cxx.so /home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/librosconsole_backend_interface.so -llog4cxx -lboost_regex-mt /home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/librostime.so /home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/libcpp_common.so -lboost_thread-mt -lboost_chrono-mt -lboost_date_time-mt -lboost_atomic-mt -lconsole_bridge /home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/libroslib.so /home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/librospack.so -lpython2.7 -lboost_program_options-mt -ltinyxml -Wl,-rpath,/home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib: 
/home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/libclass_loader.so: error: undefined reference to 'pthread_mutexattr_init'
/home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/libclass_loader.so: error: undefined reference to 'pthread_mutexattr_settype'
/home/prajankya/work/toradex/cc/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/libclass_loader.so: error: undefined reference to 'pthread_mutexattr_destroy'
ERROR: oe_runmake failed
[... cut ...]
/opt/ros/indigo/lib/libclass_loader.so: error: undefined reference to 'pthread_mutexattr_settype'
/opt/ros/indigo/lib/libclass_loader.so: error: undefined reference to 'pthread_mutexattr_destroy'
```

Its a clean Install, as seen in [packagegroup-ros-world.bb](https://github.com/bmwcarit/meta-ros/blob/master/recipes-ros/packagegroups/packagegroup-ros-world.bb) and also in #443 , the package `robot-state-publisher` haven't given any issue for you guys. Can anyone please help me find where am I doing wrong?

bulwahn, 2017-04-29:
Please provide your build configuration. That is at least a bare minimum for us to try to reproduce your build issue.

prajankya, 2017-04-29:
This is the board I am building for [Toradex Apalis-T30](http://developer.toradex.com/products/apalis-t30)
[link 2](https://www.toradex.com/computer-on-modules/apalis-arm-family/nvidia-tegra-3)

This is the ROS project I want to run on it :
[Warehouse-Robot](https://github.com/snappyrobotics/warehouse_robot)
and this is [the file](https://github.com/SnappyRobotics/warehouse_robot/blob/master/launch/state_publisher.launch) in which I am using robot_state_publisher.

My bblayers.conf file :

```cmake
# LAYER_CONF_VERSION is increased each time build/conf/bblayers.conf
# changes incompatibly
LCONF_VERSION = "7"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  ${TOPDIR}/../layers/meta-angstrom \
  \
  \
  ${TOPDIR}/../layers/meta-toradex-nxp \
  ${TOPDIR}/../layers/meta-freescale \
  ${TOPDIR}/../layers/meta-freescale-3rdparty \
  \
  ${TOPDIR}/../layers/meta-toradex-tegra \
  ${TOPDIR}/../layers/meta-jetson-tk1 \
  \
  \
  ${TOPDIR}/../layers/meta-toradex-bsp-common \
  \
  \
  ${TOPDIR}/../layers/meta-linaro/meta-linaro-toolchain \
  ${TOPDIR}/../layers/meta-openembedded/meta-oe \
  ${TOPDIR}/../layers/meta-openembedded/meta-efl \
  ${TOPDIR}/../layers/meta-openembedded/meta-gpe \
  ${TOPDIR}/../layers/meta-openembedded/meta-gnome \
  ${TOPDIR}/../layers/meta-openembedded/meta-xfce \
  ${TOPDIR}/../layers/meta-openembedded/meta-initramfs \
  ${TOPDIR}/../layers/meta-openembedded/meta-systemd \
  ${TOPDIR}/../layers/meta-openembedded/meta-networking \
  ${TOPDIR}/../layers/meta-openembedded/meta-multimedia \
  ${TOPDIR}/../layers/meta-openembedded/meta-python \
  \
  \
  ${TOPDIR}/../layers/meta-lxde \
  ${TOPDIR}/../layers/meta-browser \
  ${TOPDIR}/../layers/meta-qt4 \
  ${TOPDIR}/../layers/meta-qt5 \
  ${TOPDIR}/../layers/meta-freescale-distro \
  ${TOPDIR}/../layers/meta-toradex-demos \
  ${TOPDIR}/../layers/meta-snappy/meta-nodejs \
  \
  ${TOPDIR}/../layers/meta-snappy/meta-snappy-setup \
  \
  ${TOPDIR}/../layers/meta-ros \
  \
  ${TOPDIR}/../layers/openembedded-core/meta \
"
```

This is my local.conf
```cmake
WARN_QA_remove = "host-user-contaminated"

# meta-freescale otherwise defaults to wayland
DISTRO_FEATURES_remove = "wayland"

# This file is your local configuration file and is where all local user settings
# are placed. The comments in this file give some guide to the options a new user
[... 398 lines cut ...]
| CMakeFiles/Makefile2:166: recipe for target 'CMakeFiles/plugin_tool.dir/all' failed
| make[1]: *** [CMakeFiles/plugin_tool.dir/all] Error 2
| make[1]: Leaving directory '/home/prajankya/work/toradex/devel/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/pluginlib/1.10.5-r0/build'
| Makefile:130: recipe for target 'all' failed
| make: *** [all] Error 2
| WARNING: /home/prajankya/work/toradex/devel/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/pluginlib/1.10.5-r0/temp/run.do_compile.32833:1 exit 1 from 'exit 1'
| ERROR: Function failed: do_compile (log file is located at /home/prajankya/work/toradex/devel/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/pluginlib/1.10.5-r0/temp/log.do_compile.32833)
ERROR: Task (/home/prajankya/work/toradex/devel/build/../layers/meta-ros/recipes-ros/pluginlib/pluginlib_1.10.5.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2125 tasks of which 0 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory

Summary: 1 task failed:
  /home/prajankya/work/toradex/devel/build/../layers/meta-ros/recipes-ros/pluginlib/pluginlib_1.10.5.bb:do_compile
Summary: There were 2 WARNING messages shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

With these steps you can reproduce the issue >

```bash
repo init -u http://github.com/SnappyRobotics/snappy-toradex.git
repo sync
. export
bitbake robot-state-publisher
```

rojkov, 2017-05-02:
| [100%] Linking CXX executable devel/lib/pluginlib/plugin_tool

> | /home/prajankya/work/toradex/devel/build/tmp-glibc/sysroots/x86_64-linux/usr/bin/cmake -E cmake_link_script CMakeFiles/plugin_tool.dir/link.txt --verbose=1
> | /home/prajankya/work/toradex/devel/build/tmp-glibc/sysroots/x86_64-linux/usr/bin/arm-angstrom-linux-gnueabi/arm-angstrom-linux-gnueabi-g++    -march=armv7-a -mthumb -mfpu=neon  -mfloat-abi=hard  --sysroot=/home/prajankya/work/toradex/devel/build/tmp-glibc/sysroots/apalis-t30 -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/prajankya/work/toradex/devel/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/pluginlib/1.10.5-r0=/usr/src/debug/pluginlib/1.10.5-r0 -fdebug-prefix-map=/home/prajankya/work/toradex/devel/build/tmp-glibc/sysroots/x86_64-linux= -fdebug-prefix-map=/home/prajankya/work/toradex/devel/build/tmp-glibc/sysroots/apalis-t30= -fvisibility-inlines-hidden  -march=armv7-a -mthumb -mfpu=neon  -mfloat-abi=hard  --sysroot=/home/prajankya/work/toradex/devel/build/tmp-glibc/sysroots/apalis-t30 -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/prajankya/work/toradex/devel/build/tmp-glibc/work/armv7at2hf-neon-angstrom-linux-gnueabi/pluginlib/1.10.5-r0=/usr/src/debug/pluginlib/1.10.5-r0 -fdebug-prefix-map=/home/prajankya/work/toradex/devel/build/tmp-glibc/sysroots/x86_64-linux= -fdebug-prefix-map=/home/prajankya/work/toradex/devel/build/tmp-glibc/sysroots/apalis-t30= -fvisibility-inlines-hidden -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed  -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed CMakeFiles/plugin_tool.dir/src/plugin_tool.cpp.o  -o devel/lib/pluginlib/plugin_tool -rdynamic -lboost_filesystem-mt -lboost_system-mt /home/prajankya/work/toradex/devel/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/libclass_loader.so -lPocoFoundation -ldl /home/prajankya/work/toradex/devel/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/librosconsole.so /home/prajankya/work/toradex/devel/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/librosconsole_log4cxx.so /home/prajankya/work/toradex/devel/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/librosconsole_backend_interface.so -llog4cxx -lboost_regex-mt /home/prajankya/work/toradex/devel/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/librostime.so /home/prajankya/work/toradex/devel/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/libcpp_common.so -lboost_thread-mt -lboost_chrono-mt -lboost_date_time-mt -lboost_atomic-mt -lconsole_bridge /home/prajankya/work/toradex/devel/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/libroslib.so /home/prajankya/work/toradex/devel/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/librospack.so -lpython2.7 -lboost_filesystem-mt -lboost_program_options-mt -lboost_system-mt -ltinyxml -ltinyxml -lboost_filesystem-mt /home/prajankya/work/toradex/devel/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/libclass_loader.so -lPocoFoundation -ldl /home/prajankya/work/toradex/devel/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/librosconsole.so /home/prajankya/work/toradex/devel/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/librosconsole_log4cxx.so /home/prajankya/work/toradex/devel/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/librosconsole_backend_interface.so -llog4cxx -lboost_regex-mt /home/prajankya/work/toradex/devel/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/librostime.so /home/prajankya/work/toradex/devel/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/libcpp_common.so -lboost_thread-mt -lboost_chrono-mt -lboost_date_time-mt -lboost_atomic-mt -lconsole_bridge /home/prajankya/work/toradex/devel/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/libroslib.so /home/prajankya/work/toradex/devel/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/librospack.so -lpython2.7 -lboost_program_options-mt -ltinyxml -Wl,-rpath,/home/prajankya/work/toradex/devel/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib:| /home/prajankya/work/toradex/devel/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/libclass_loader.so: error: undefined reference to 'pthread_mutexattr_init'| /home/prajankya/work/toradex/devel/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/libclass_loader.so: error: undefined reference to 'pthread_mutexattr_settype'| /home/prajankya/work/toradex/devel/build/tmp-glibc/sysroots/apalis-t30/opt/ros/indigo/lib/libclass_loader.so: error: undefined reference to 'pthread_mutexattr_destroy'| ERROR: oe_runmake failed| collect2: error: ld returned 1 exit status
>
> This linker command clearly lacks `-lpthread` option - the library
containing the missing symbols. But I suspect this is a problem of Boost
which should have exported this option in its linker interface.

prajankya, 2017-05-05:
I clearly cant understand how to solve this issue and left it there as it is, for the time being, I am using robot state publisher running in other computer, and hence, my problem is solved, but it is not a solution(or its just not "resolved") and hence I think, others might also face it. So I left to you maintainers to either close this issue or not, depending on the method you are following.

Thank you for your  support @rojkov

mrlarsendk, 2017-06-01:
I don't know if this is a closed issue, but I encountered the same issue with a toradex vf61 board and the toradex oe-core image build setup. However, I worked around this issue by adding: TARGET_CC_ARCH += "-lpthread" to the .bb recipe file for pluginlib. I then had the same issue with two other recipies, which I fixed using the same method.

bulwahn, 2017-06-01:
Is this an issue that is due to a broken BSP layer or must we resolve this in the meta-ros layer?

mrlarsendk, 2017-06-01:
Hey Lukas,

I honestly don't know yet. I will have to test with a default poky setup to figure this out. Which I will be doing in the near future. 

Btw, it is through my job as development engineer that I am testing this. I am with a small drone manufacturing company based in Denmark, called sky-watch.

We will be happy to share any findings and suggestions.

Mvh Michael Larsen

bulwahn, 2017-06-01:
Sure. Go ahead, I am just wondering because I have not noticed this issue with any of the general qemu machines.

mrlarsendk, 2017-06-05:
Hi Lukas,

I was wondering if there are any plans to upgrade the ROS version to Kinetic og Lunar.

PS: Regarding the issue I wrote a reply to, I think it is indeed a BSP issue and thus I do not believe that you need to change anything.

Best regards,
Michael

bulwahn, 2017-06-05:
Concerning updating to kinetic, I already provided some information on https://github.com/bmwcarit/meta-ros/issues/485.
Currently, there is a branch called kinetic-experimental-v2-alpha2 on my personal fork. It is in acceptable quality: some recipes still fail and the branch is not continuously tested and not maintained, but you can start experimenting with it.
For the lunar release, we have not done anything.

bulwahn, 2017-06-27:
I am closing this issue because users resolved the issue differently and it is related to the specific BSP layer those users were using.

closed 2017-06-27 by bulwahn