ros/meta-ros #497: sophus: fails in do_compile on x86-64 architecture

bulwahn, 2017-04-29:
`bitbake sophus` fails during compiling with:
```
| /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/core2-64-oe-linux/sophus/0.9.1-r0/recipe-sysroot/usr/include/eigen3/Eigen/src/Core/Matrix.h:311:7:   required from 'Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>::Matrix(const Eigen::ReturnByValue<OtherDerived>&) [with OtherDerived = Eigen::MatrixExponentialReturnValue<Eigen::Matrix<float, 4, 4> >; _Scalar = float; int _Rows = 4; int _Cols = 4; int _Options = 0; int _MaxRows = 4; int _MaxCols = 4]'
| /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/core2-64-oe-linux/sophus/0.9.1-r0/sophus-0.9.1/sophus/tests.hpp:96:64:   required from 'bool Sophus::Tests<LieGroup>::expMapTest() [with LieGroup = Sophus::Sim3Group<float>]'
| /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/core2-64-oe-linux/sophus/0.9.1-r0/sophus-0.9.1/sophus/tests.hpp:211:24:   required from 'void Sophus::Tests<LieGroup>::runAllTests() [with LieGroup = Sophus::Sim3Group<float>]'
| /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/core2-64-oe-linux/sophus/0.9.1-r0/sophus-0.9.1/sophus/test_sim3.cpp:97:3:   required from 'void tests() [with Scalar = float]'
| /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/core2-64-oe-linux/sophus/0.9.1-r0/sophus-0.9.1/sophus/test_sim3.cpp:107:16:   required from here
| /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/core2-64-oe-linux/sophus/0.9.1-r0/recipe-sysroot/usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:56:30: error: ignoring attributes on template argument 'Eigen::internal::packet_traits<float>::type {aka __vector(4) float}' [-Werror=ignored-attributes]
| ERROR: oe_runmake failed
```

graugans, 2017-05-17:
I have an similar error when building for x86-64:

```
/home/chris/src/udoo-meta-freescale/build/tmp/sysroots/x86_64-linux/usr/bin/x86_64-fslc-linux/x86_64-fslc-linux-g++    -I/home/chris/src/udoo-meta-freescale/build/tmp/sysroots/udoox86/usr/include/eigen3  -m64 -march=corei7 -mtune=corei7 -mfpmath=sse -msse4.2  --sysroot=/home/chris/src/udoo-meta-freescale/build/tmp/sysroots/udoox86  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/chris/src/udoo-meta-freescale/build/tmp/work/corei7-64-fslc-linux/sophus/0.9.1-r0=/usr/src/debug/sophus/0.9.1-r0 -fdebug-prefix-map=/home/chris/src/udoo-meta-freescale/build/tmp/sysroots/x86_64-linux= -fdebug-prefix-map=/home/chris/src/udoo-meta-freescale/build/tmp/sysroots/udoox86=  -fvisibility-inlines-hidden  -m64 -march=corei7 -mtune=corei7 -mfpmath=sse -msse4.2  --sysroot=/home/chris/src/udoo-meta-freescale/build/tmp/sysroots/udoox86 -Wno-deprecated-declarations -Wno-misleading-indentation -O3 -DNDEBUG    -Wall -Werror -Wno-unused-variable                    -Wno-unused-but-set-variable -Wno-unknown-pragmas  -o CMakeFiles/test_rxso3.dir/sophus/test_rxso3.cpp.o -c /home/chris/src/udoo-meta-freescale/build/tmp/work/corei7-64-fslc-linux/sophus/0.9.1-r0/sophus-0.9.1/sophus/test_rxso3.cpp
In file included from /home/chris/src/udoo-meta-freescale/build/tmp/sysroots/udoox86/usr/include/eigen3/Eigen/Core:263:0,
                 from /home/chris/src/udoo-meta-freescale/build/tmp/sysroots/udoox86/usr/include/eigen3/unsupported/Eigen/MatrixFunctions:19,
                 from /home/chris/src/udoo-meta-freescale/build/tmp/work/corei7-64-fslc-linux/sophus/0.9.1-r0/sophus-0.9.1/sophus/test_se2.cpp:26:
/home/chris/src/udoo-meta-freescale/build/tmp/sysroots/udoox86/usr/include/eigen3/Eigen/src/Core/arch/SSE/PacketMath.h:29:39: error: ignoring attributes on template argument '__m128 {aka __vector(4) float}' [-Werror=ignored-attributes]
 template<> struct is_arithmetic<__m128>  { enum { value = true }; };
                                       ^                    ^
```
Looks like this is an warning what is turned to an error due to ``-Wall -Werror`` It seems there is an fix in the latest [libeigen beta](https://forum.kde.org/viewtopic.php?f=74&t=130479#p349199)

graugans, 2017-05-18:
I've fixed the issue with the following patch

```
diff --git a/recipes-extended/sophus/sophus_0.9.1.bb b/recipes-extended/sophus/sophus_0.9.1.bb
index 64cbc8b..406647b 100644
--- a/recipes-extended/sophus/sophus_0.9.1.bb
+++ b/recipes-extended/sophus/sophus_0.9.1.bb
@@ -14,4 +14,4 @@ S = "${WORKDIR}/sophus-${PV}"
 inherit cmake

 # CXXFLAGS are needed to compile eigen 3.3.1 headers properly
-CXXFLAGS += "-Wno-deprecated-declarations -Wno-misleading-indentation"
+CXXFLAGS += "-Wno-deprecated-declarations -Wno-misleading-indentation -Wno-ignored-attributes"
```

bulwahn, 2017-05-19:
👍 Great. Just open a pull request, and I will quickly test it and then merge it. By the way, if you add `(resolves #497)` in the commit message, the commit is immediately linked to this issue and the issue is closed once it has been merged.

commit 2b770491 (2017-07-06): Merge pull request #530 from bulwahn/resolve-issue-#497

sophus: compile on x86-64 arch (resolves #497)
commit 9d6db38f (2017-07-06): sophus: compile on x86-64 arch (resolves #497)

This commit implements Christian Ege's suggestion in the meta-ros
issue tracker to make gcc only warn on ignored template attributes.
closed 2017-07-06 by rojkov