ros/meta-ros #498: fcl:  non -dev/-dbg/nativesdk- package contains symlink .so

bulwahn, 2017-05-02:
On kinetic-experimental-v2-alpha2, `bitbake fcl` fails with:
```
ERROR: fcl-0.5.0-r0 do_package_qa: QA Issue: non -dev/-dbg/nativesdk- package contains symlink .so: fcl path '/work/i586-oe-linux/fcl/0.5.0-r0/packages-split/fcl/usr/lib/libfcl.so' [dev-so]
ERROR: fcl-0.5.0-r0 do_package_qa: QA run found fatal errors. Please consider fixing them.
ERROR: fcl-0.5.0-r0 do_package_qa: Function failed: do_package_qa
ERROR: Logfile of failure stored in: /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/fcl/0.5.0-r0/temp/log.do_package_qa.29534
ERROR: Task (/home/lukas/work/repositories/github.com/bmwcarit/meta-ros/recipes-extended/fcl/fcl_0.5.0.bb:do_package_qa) failed with exit code '1'
```

This is the used build configuration:
```
Build Configuration:
BB_VERSION        = "1.33.4"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "ubuntu-14.04"
TARGET_SYS        = "i586-oe-linux"
MACHINE           = "qemux86"
DISTRO            = "nodistro"
DISTRO_VERSION    = "nodistro.0"
TUNE_FEATURES     = "m32 i586"
TARGET_FPU        = ""
meta              = "master:123962018251dfb1d6ca5aa5c0d02534007de3ab"
meta-oe           
meta-multimedia   
meta-python       = "master:e7ad92decb7025d76e2e8b2bc15349d548c8ad14"
meta-ros          = "kinetic-experimental-v2-alpha2:aa4fa0890f5be298a6f3539c59683f85ee52cabc"
```

iamxmm, 2019-03-18:
I have the same error as you. Did you found the root cause of it?

herb-kuta-lge, 2019-11-14:
`fcl` successfully builds for **melodic** using `[thud-draft]`.  @bulwahn < Can I close this issue?

closed 2019-12-03 by herb-kuta-lge