ros/meta-ros #499: moveit-ros-perception:

bulwahn, 2017-05-02:
On kinetic-experimental-v2-alpha2, `bitbake moveit-ros-perception` fails with:
```
| -- Found OpenGL: /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/moveit-ros-perception/0.9.6-r0/recipe-sysroot/usr/lib/libGL.so
| CMake Error at /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/moveit-ros-perception/0.9.6-r0/recipe-sysroot-native/usr/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
|   Could NOT find GLEW (missing: GLEW_INCLUDE_DIR GLEW_LIBRARY)
| Call Stack (most recent call first):
|   /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/moveit-ros-perception/0.9.6-r0/recipe-sysroot-native/usr/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
|   /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/moveit-ros-perception/0.9.6-r0/recipe-sysroot-native/usr/share/cmake-3.7/Modules/FindGLEW.cmake:38 (find_package_handle_standard_args)
|   CMakeLists.txt:13 (find_package)
| 
| 
| -- Configuring incomplete, errors occurred!
| See also "/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/moveit-ros-perception/0.9.6-r0/build/CMakeFiles/CMakeOutput.log".
| See also "/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/moveit-ros-perception/0.9.6-r0/build/CMakeFiles/CMakeError.log".
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_configure (log file is located at /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/moveit-ros-perception/0.9.6-r0/temp/log.do_configure.29543)
ERROR: Task (/home/lukas/work/repositories/github.com/bmwcarit/meta-ros/recipes-ros/moveit/moveit-ros-perception_0.9.6.bb:do_configure) failed with exit code '1'
```

The distro features in local.conf are `DISTRO_FEATURES = "directfb opengl x11"`.
This is the used build configuration:
```
Build Configuration:
BB_VERSION        = "1.33.4"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "ubuntu-14.04"
TARGET_SYS        = "i586-oe-linux"
MACHINE           = "qemux86"
DISTRO            = "nodistro"
DISTRO_VERSION    = "nodistro.0"
TUNE_FEATURES     = "m32 i586"
TARGET_FPU        = ""
meta              = "master:123962018251dfb1d6ca5aa5c0d02534007de3ab"
meta-oe           
meta-multimedia   
meta-python       = "master:e7ad92decb7025d76e2e8b2bc15349d548c8ad14"
meta-ros          = "kinetic-experimental-v2-alpha2:aa4fa0890f5be298a6f3539c59683f85ee52cabc"
```

herb-kuta-lge, 2019-11-14:
NB. Requires `opengl` in appear in `DISTRO_FEATURES`.

herb-kuta-lge, 2020-02-07:
@bulwahn < Can this issue be closed?

closed 2020-02-21 by herb-kuta-lge