ros/meta-ros #500: collada-urdf: do_compile fails including eigen headers

bulwahn, 2017-05-02:
On kinetic-experimental-v2-alpha2, `bitbake collada-urdf` fails with:
```
| /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/collada-urdf/1.12.8-r0/recipe-sysroot-native/usr/bin/i586-oe-linux/i586-oe-linux-g++   -DASSIMP_EXPORT_API -DASSIMP_UNIFIED_HEADER_NAMES -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"collada_urdf\" -Dcollada_urdf_EXPORTS -I/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/collada-urdf/1.12.8-r0/robot_model-1.12.8/collada_urdf/include -I/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/collada-urdf/1.12.8-r0/recipe-sysroot/usr/include/assimp -I/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/collada-urdf/1.12.8-r0/recipe-sysroot/usr/include -I/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/collada-urdf/1.12.8-r0/recipe-sysroot/usr/include/collada-dom2.4/1.5 -I/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/collada-urdf/1.12.8-r0/recipe-sysroot/usr/include/collada-dom2.4 -I/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/collada-urdf/1.12.8-r0/recipe-sysroot/opt/ros/indigo/include -I/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/collada-urdf/1.12.8-r0/recipe-sysroot/opt/ros/indigo/share/xmlrpcpp/cmake/../../../include/xmlrpcpp  -m32 -march=i586  --sysroot=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/collada-urdf/1.12.8-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/collada-urdf/1.12.8-r0=/usr/src/debug/collada-urdf/1.12.8-r0 -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/collada-urdf/1.12.8-r0/recipe-sysroot-native= -fdebug-prefix-map=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/collada-urdf/1.12.8-r0/recipe-sysroot=  -fvisibility-inlines-hidden  -m32 -march=i586  --sysroot=/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/collada-urdf/1.12.8-r0/recipe-sysroot -fPIC   -std=c++11 -o CMakeFiles/collada_urdf.dir/src/collada_urdf.cpp.o -c /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/collada-urdf/1.12.8-r0/robot_model-1.12.8/collada_urdf/src/collada_urdf.cpp
| In file included from /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/collada-urdf/1.12.8-r0/recipe-sysroot/opt/ros/indigo/include/eigen_stl_containers/eigen_stl_containers.h:40:0,
|                  from /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/collada-urdf/1.12.8-r0/recipe-sysroot/opt/ros/indigo/include/geometric_shapes/mesh_operations.h:41,
|                  from /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/collada-urdf/1.12.8-r0/robot_model-1.12.8/collada_urdf/src/collada_urdf.cpp:89:
| /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/collada-urdf/1.12.8-r0/recipe-sysroot/opt/ros/indigo/include/eigen_stl_containers/eigen_stl_vector_container.h:40:22: fatal error: Eigen/Core: No such file or directory
|  #include <Eigen/Core>
|                       ^
| compilation terminated.
| ERROR: oe_runmake failed
| make[2]: *** [CMakeFiles/collada_urdf.dir/src/collada_urdf.cpp.o] Error 1
| make[2]: Leaving directory `/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/collada-urdf/1.12.8-r0/build'
| make[1]: *** [CMakeFiles/collada_urdf.dir/all] Error 2
| make[1]: Leaving directory `/home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/collada-urdf/1.12.8-r0/build'
| make: *** [all] Error 2
| ERROR: Function failed: do_compile (log file is located at /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/collada-urdf/1.12.8-r0/temp/log.do_compile.29748)
ERROR: Task (/home/lukas/work/repositories/github.com/bmwcarit/meta-ros/recipes-ros/robot-model/collada-urdf_1.12.8.bb:do_compile) failed with exit code '1'
```

This is the used build configuration:
```
Build Configuration:
BB_VERSION        = "1.33.4"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "ubuntu-14.04"
TARGET_SYS        = "i586-oe-linux"
MACHINE           = "qemux86"
DISTRO            = "nodistro"
DISTRO_VERSION    = "nodistro.0"
TUNE_FEATURES     = "m32 i586"
TARGET_FPU        = ""
meta              = "master:123962018251dfb1d6ca5aa5c0d02534007de3ab"
meta-oe           
meta-multimedia   
meta-python       = "master:e7ad92decb7025d76e2e8b2bc15349d548c8ad14"
meta-ros          = "kinetic-experimental-v2-alpha2:aa4fa0890f5be298a6f3539c59683f85ee52cabc"
```

bulwahn, 2017-10-25:
This issue can be actually resolved by updating the libeigen recipe in meta-oe (a current patch that is simply just laying around on my hardware, but I did not have time to properly review, test and provide to meta-oe yet).

herb-kuta-lge, 2019-11-14:
`collada-urdf` successfully builds for **melodic** using `[thud-draft]`. @bulwahn < Can I close this issue?

closed 2019-12-03 by herb-kuta-lge