ros/meta-ros #505: tf and bfl: do_compile fails with boost::serialization error with boost 1.64

bulwahn, 2017-05-13:
OpenEmbedded-Core updated boost to version 1.64 [1]. The two packages tf and bfl both include boost/numeric/ublas, which seems not to properly compile with boost version 1.64.

I already reported the issue with tf on the ros/geometry issue tracker at: https://github.com/ros/geometry/issues/146

Concerning the bfl issue, I did not report anything to the bfl project/maintainers yet.

`bitbake tf` fails during compiling with:
```
| In file included from /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf/1.11.8-r0/recipe-sysroot/usr/include/boost/numeric/ublas/vector.hpp:21:0,
|                  from /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf/1.11.8-r0/recipe-sysroot/usr/include/boost/numeric/ublas/matrix.hpp:18,
|                  from /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf/1.11.8-r0/geometry-1.11.8/tf/src/transform_listener.cpp:34:
| /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf/1.11.8-r0/recipe-sysroot/usr/include/boost/numeric/ublas/storage.hpp: In member function 'void boost::numeric::ublas::unbounded_array<T, ALLOC>::serialize(Archive&, unsigned int)':
| /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf/1.11.8-r0/recipe-sysroot/usr/include/boost/numeric/ublas/storage.hpp:299:18: error: 'make_array' is not a member of 'boost::serialization'
|              ar & serialization::make_array(data_, s);
|                   ^~~~~~~~~~~~~
| /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf/1.11.8-r0/recipe-sysroot/usr/include/boost/numeric/ublas/storage.hpp: In member function 'void boost::numeric::ublas::bounded_array<T, N, ALLOC>::serialize(Archive&, unsigned int)':
| /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/tf/1.11.8-r0/recipe-sysroot/usr/include/boost/numeric/ublas/storage.hpp:494:18: error: 'make_array' is not a member of 'boost::serialization'
|              ar & serialization::make_array(data_, s);
|                   ^~~~~~~~~~~~~
```

And similarly for bfl, a similar error occurs:
```
| /home/lukas/work/repositories/openembedded.org/openembedded-core/build/tmp-glibc/work/i586-oe-linux/bfl/0.7.0-6-r0/recipe-sysroot/usr/include/boost/numeric/ublas/storage.hpp:299:18: error: 'make_array' is not a member of 'boost::serialization'
|              ar & serialization::make_array(data_, s);
|                   ^~~~~~~~~~~~~
```

The failing build configuration is:
```
Build Configuration:
BB_VERSION        = "1.33.4"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "ubuntu-14.04"
TARGET_SYS        = "i586-oe-linux"
MACHINE           = "qemux86"
DISTRO            = "nodistro"
DISTRO_VERSION    = "nodistro.0"
TUNE_FEATURES     = "m32 i586"
TARGET_FPU        = ""
meta              = "master:c59fa3bd71b42410bf032846ee8fdb6e6eb1b95c"
meta-oe           
meta-multimedia   
meta-python       = "master:fc03b3d9382dd5b6a8fff5f84e4c86a1acc9a73a"
meta-ros          = "boost-update-to-1.64:118a1d072872cd1c7b013e9ff1545116a7703627"
```

[1] http://cgit.openembedded.org/openembedded-core/commit/?id=9ba71dc40dd782d192504b0827ca91f0fe180abe

rojkov, 2017-05-15:
This patch [1] fixes the issue. I'll submit a backport to OE-core.

[1] https://github.com/boostorg/serialization/commit/1d86261581230e2dc5d617a9b16287d326f3e229

rojkov, 2017-05-15:
Submitted https://patchwork.openembedded.org/patch/139856/

bulwahn, 2017-05-16:
The tf package could be easily addressed to work around the issue. Then, I also encountered another issue in the laser-geometry package.
However, it is much better to just get boost properly working again. I am currently testing the proposed boost patch with meta-ros.

bulwahn, 2017-05-16:
I tested the boost patch successfully with this build configuration:
```
Build Configuration:
BB_VERSION        = "1.33.4"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "ubuntu-14.04"
TARGET_SYS        = "i586-oe-linux"
MACHINE           = "qemux86"
DISTRO            = "nodistro"
DISTRO_VERSION    = "nodistro.0"
TUNE_FEATURES     = "m32 i586"
TARGET_FPU        = ""
meta              = "boost-1.64-fix:313410a3130272edf9c6d5155b27967e272ce16d"
meta-oe           
meta-multimedia   
meta-python       = "master:fc03b3d9382dd5b6a8fff5f84e4c86a1acc9a73a"
meta-ros          = "master:36eb1dd4f23f4a65827aea12258b4fbc7f751dc1"
```

bulwahn, 2017-06-01:
The commit that resolves this issue has been finally merged to openembedded-core, cf. http://cgit.openembedded.org/openembedded-core/commit/?id=e1da4cc99e57c596b2a354767cc1e261718dac52.

bulwahn, 2017-06-27:
Issue is resolved in the openembedded-core repository for quite some time; so we expect that users will not encounter the issue if they update to recent versions.

closed 2017-06-27 by bulwahn