ros/meta-ros #507: catkin_make for monolithic builds with multiple packages in yocto

fabianoboril, 2017-05-13:
Hi everyone,

when using catkin_make locally on the host it comes with the nice feature that you can run a monolithic build on a src-directory containing multiple catkin packages without the need to create a top-level CMakeList in the src-directory. catkin automatically finds all packages and creates a topological order to resolve dependencies (see: http://wiki.ros.org/catkin/commands/catkin_make)

Is there a way to use this feature also within yocto? I tried to do so, however the build process within yocto (I use "inherit catkin" in my recipe) complains about a missing CMakeList in the src-directory. I could only work around this issue, by creating the top-level CMakeList pointing to all packages manually.

I appreciate your help and support.

Best regards,
Fabian

bulwahn, 2017-05-15:
Dear Fabian,

Yocto is used to build distributions, and distributions are essentially composed of software packages. So, it seems reasonable to me to just write one recipe for each software package. The effort is quite little to provide one recipe for each package, and it provides the benefit to build images with a the fine-grained control on which package is in the image, and which one is not.

fabianoboril, 2017-05-15:
Hi,

I absolutely understand your point. However, in my case I have to use a monolithic build approach for certain reasons. I also understand, if it is not possible/interesting for you to support this catkin feature in yocto. For now, I found a way to generate the top-level CMakeLists manually.

Best regards,
Fabian

closed 2018-03-24 by bulwahn